srsran/srsRAN_4G #818: A question about srsRAN with ZMQ virtualized radio

elvinlife, 2022-02-28:
Hi thanks for checking this issue. Currently I'm deploying the srsRAN with ZMQ virtualized radio, and I notice that srsRAN can only support one UE. Can I ask what's the fundamental limitation that we cannot support multiple UE with virtualized radio in this context?

sonium-labs, 2022-03-01:
Hi elvinlife, 

I've been messing with the ZMQ radio stuff for a couple months, and may have some insight for you.

While I haven't tried multi-UE setups with ZMQ alone, and am unsure what the purpose might be for this restriction, if you set up the GNU Radio broker described at the bottom of [this](https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html) guide, from my experiments it appears as though you can indeed connect multiple srsUEs to a single eNB. While I haven't tested all portions of the LTE protocol through this system, it seems I/Q samples are indeed being sent from both UEs to the eNB, and back from the eNB to both UEs. If you get deeper into this, I'd love to hear back from you regarding anything you discover!

Here's what the flowchart might look like (the delays are unnecessary, just some simulated propagation delay for a project I'm working on).
[image]

Feel free to reach out if you have any additional questions.

elvinlife, 2022-03-03:
Hi Daniel, thanks for your reply and the provided approach.

I build a system with the same workflow shown in the plot (just without the delay module). I can see that some signal samples are exchanged between the two UEs and the eNB at the beginning. However, based on the logging, I think only one UE is finally attached to the EPC and the eNB, but another UE fails. I'm guessing that when the eNB broadcasts the signals to two UEs, the broadcasted information is actually for one UE (the one gets attached finally)?

I'm happy to discuss if you have answers or ideas about this issue. Thanks!

andrepuschmann, 2022-03-03:
The eNB is of course able to handle multiple UEs at the same time. The approach explained by @sonium-labs is correct and the GRC flowgraph should be working. We've tested similar things with many more UEs and it's working fine.

@elvinlife You need to post actual logs so other can help you with that.

I am going to close the issue and add the "tutorial" tag to it so other can find the information (and the flow-graph) later on. Please open a new issue (as it is likely something unrelated to ZMQ) with you logs. Thanks @sonium-labs for your help.

closed 2022-03-03 by andrepuschmann

sonium-labs, 2022-03-03:
Very interesting, excellent insight! Glad you were able to sort it out, and thanks for letting us know what it was. :)

toddeye, 2022-03-07:
@elvinlife I have been trying to set something similar up.  Would you mind attaching your working .grc file.   I have set it up similar to what @sonium-labs shows in the image, but I must be missing something.

elvinlife, 2022-03-08:
Hi @toddeye, I've attached the .grc file, remember to modify the file extension from .txt to .grc.
[2ue.txt](https://github.com/srsran/srsRAN/files/8202225/2ue.txt)

mariatsampazi, 2022-09-08:
Hi all, have you been able to open gnuradio-companion for the 5G SA ZMQ setup?
Although I see a successful attachment of the ue to the gnb (ue has an ip, no errors in the logs), while opening the gnuradio-companion I cannot see any signal being transmitted or any IQ sample
1. I have successfully set up the ZMQ E2E network as stated here: https://docs.srsran.com/en/hoverxref_test/app_notes/source/5g_sa_E2E/source/index.html
2. I followed the steps here: https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html#:~:text=netns%20delete%20ue1-,GNU%2DRadio%20Companion%20Integration,-GNU%2DRadio%20Companion so that I can see the transmission of IQ samples, but nothing is transmitted.
Thanks in advance!

Waseem0808, 2022-11-17:
> Hi elvinlife,
>
> I've been messing with the ZMQ radio stuff for a couple months, and may have some insight for you.
>
> While I haven't tried multi-UE setups with ZMQ alone, and am unsure what the [...]

Thanks for the detailed explanation. 
I want to create a 2tx/2rx port UE, but was unsuccessful. Is their a way to add multiple ports to UE/eNB? 
Thanks!

kaposnick, 2022-12-05:
Hi all, I've been facing this issue for a long time though I have managed to connect two UEs using the zeromq. One problem that was always coming was up when I wanted to re-initiate a scenario. I tracked down the code and noticed it was because of the .ctxt files they are creating and when they were attempting to [read them back](https://github.com/srsran/srsRAN/blob/254cc719a9a31f64ce0262f4ca6ab72b1803477d/srsue/src/stack/upper/nas.cc#L93).

```java  
if ((read_ctxt_file(&ctxt, &ctxt_base))) {
    usim->generate_nas_keys(
        ctxt.k_asme, ctxt_base.k_nas_enc, ctxt_base.k_nas_int, ctxt_base.cipher_algo, ctxt_base.integ_algo);
    logger.debug(ctxt_base.k_nas_enc, 32, "NAS encryption key - k_nas_enc");
    logger.debug(ctxt_base.k_nas_int, 32, "NAS integrity key - k_nas_int");
}
```
I commented this line and the (my) problem was solved.

wmwwimor, 2023-07-17:
What about srsRAN_Project and srsRAN_4G？
Can I build an End-to-end 5G SA network with mulitiple UEs using GNU ?

xixihahahha, 2024-09-03:
/

> Hi elvinlife,
>
> I've been messing with the ZMQ radio stuff for a couple months, and may have some insight for you.
>
> While I haven't tried multi-UE setups with ZMQ alone, and am unsure what the [...]

hello, the link in the above answer is not working. How can I find the tutorial? Thanks.