srsran/srsRAN_4G #819: srsRAN AMF-UE-NGAP-ID mismatch on Auth req/resp (Corescope)

richardjortega, 2022-02-28:
Was unable to find this issue within the mailing list.

## Issue Description ##
I am using [Corescope](https://github.com/srsran/corescope) which depends on srsRAN for handling the NGAP/NAS-NGS comms. I am currently experiencing a bug where the `id-AMF-UE-NGAP-ID` is not correctly being processed by srsRAN.

## Setup Details ##
- Corescope (a7fd328) and used example Corescope [Python test script](https://github.com/srsran/corescope/blob/master/examples/corescope-api-client/corescope-tester.py)
- Using Metaswitch 5G Packet Core (which has been tested against real 5G SA and NSA UEs) -- tested against 3GPP spec, can be substituted by any 5G core.

### corescope.conf

All settings here match what is within the 5G Core I am testing (Metaswitch 5G Packet Core)

```conf
[api]
bind_addr = "0.0.0.0"
bind_port = 8000

[gnb]
gnb_id  = 0x00
name    = "srsgnb"
cell_id = 0x01
tac     = 0x01
mcc     = 001
mnc     = 01

amf_addr      = 192.168.7.131
n1c_bind_addr = 192.168.1.140
gtp_bind_addr = 192.168.1.140
```

## Expected Behavior ##
I except srsRAN to return correctly store and return the `id-AMF-UE-NGAP-ID` given by the AMF/5GC. 

#### Abbreviated *expected* behavior

```txt
AMF/5GC --> Corescope/srsRAN	
NGAP/NAS-5GS DownlinkNASTransport, Authentication request

AMF-UE-NGAP-ID: 12920604972
RAN-UE-NGAP-ID: 1

Corescope/srsRAN --> AMF/5GC 
NGAP/NAS-5GS UplinkNASTransport, Authentication response

AMF-UE-NGAP-ID: 12920604972
RAN-UE-NGAP-ID: 1 
```

## Actual Behaviour ##
srsRAN returns an incorrect `id-AMF-UE-NGAP-ID` which mismatches when authentication handshake occurs over NGAP protocol (pcap below)

Steps:
1) `InitialUEMessage` is sent from Corescope/srsRAN to AMF/5GC to create a new UE.
2) AMF/5GC responds with a `DownlinkNASTransport, Authentication request` which contains a returned UID, in my case was: `AMF-UE-NGAP-ID: 12917860624`
3) Corescope/srsRAN responds to AMF/5GC with a `UplinkNASTransport, Authentication response` which contains a **wrong** UID, in my case was: `AMF-UE-NGAP-ID: 32958739`
4) AMF/5GC responds with `ErrorIndication` message which contains an error `radioNetwork: unknown-local-UE-NGAP-ID`.

#### Abbreviated *actual* behavior

```txt
AMF/5GC --> Corescope/srsRAN	
NGAP/NAS-5GS DownlinkNASTransport, Authentication request

AMF-UE-NGAP-ID: 12920604972
RAN-UE-NGAP-ID: 1

Corescope/srsRAN --> AMF/5GC 
NGAP/NAS-5GS UplinkNASTransport, Authentication response

AMF-UE-NGAP-ID: 35703087
RAN-UE-NGAP-ID: 1 
```

## Steps to reproduce the problem ##

1) Create new UE
2) Verify `id-RAN-UE-NGAP-ID` remains consistent across authentication req/resp of `DownlinkNASTransport` and `UplinkNASTransport` calls.

### Full Wireshark `pcap`:

```pcap
Frame 1: 134 bytes on wire (1072 bits), 134 bytes captured (1072 bits)
Internet Protocol Version 4, Src: 192.168.1.140, Dst: 192.168.7.131
Stream Control Transmission Protocol, Src Port: 36645 (36645), Dst Port: 38412 (38412)
NG Application Protocol (InitialUEMessage)
    NGAP-PDU: initiatingMessage (0)
        initiatingMessage
            procedureCode: id-InitialUEMessage (15)
            criticality: ignore (1)
            value
                InitialUEMessage
                    protocolIEs: 5 items
                        Item 0: id-RAN-UE-NGAP-ID
                            ProtocolIE-Field
                                id: id-RAN-UE-NGAP-ID (85)
                                criticality: reject (0)
                                value
                                    RAN-UE-NGAP-ID: 1
                        Item 1: id-NAS-PDU
                        Item 2: id-UserLocationInformation
                        Item 3: id-RRCEstablishmentCause
                        Item 4: id-UEContextRequest
                            ProtocolIE-Field
                                id: id-UEContextRequest (112)
                                criticality: ignore (1)
                                value
                                    UEContextRequest: requested (0)

Frame 2: 150 bytes on wire (1200 bits), 150 bytes captured (1200 bits)
Internet Protocol Version 4, Src: 192.168.7.131, Dst: 192.168.1.140
Stream Control Transmission Protocol, Src Port: 38412 (38412), Dst Port: 36645 (36645)
NG Application Protocol (DownlinkNASTransport)
    NGAP-PDU: initiatingMessage (0)
        initiatingMessage
            procedureCode: id-DownlinkNASTransport (4)
            criticality: ignore (1)
            value
                DownlinkNASTransport
                    protocolIEs: 3 items
                        Item 0: id-AMF-UE-NGAP-ID
                            ProtocolIE-Field
                                id: id-AMF-UE-NGAP-ID (10)
                                criticality: reject (0)
                                value
                                    AMF-UE-NGAP-ID: 12917860624
                        Item 1: id-RAN-UE-NGAP-ID
                            ProtocolIE-Field
                                id: id-RAN-UE-NGAP-ID (85)
                                criticality: reject (0)
                                value
                                    RAN-UE-NGAP-ID: 1
                        Item 2: id-NAS-PDU

Frame 3: 146 bytes on wire (1168 bits), 146 bytes captured (1168 bits)
Internet Protocol Version 4, Src: 192.168.1.140, Dst: 192.168.7.131
Stream Control Transmission Protocol, Src Port: 36645 (36645), Dst Port: 38412 (38412)
NG Application Protocol (UplinkNASTransport)
    NGAP-PDU: initiatingMessage (0)
        initiatingMessage
            procedureCode: id-UplinkNASTransport (46)
            criticality: ignore (1)
            value
                UplinkNASTransport
                    protocolIEs: 4 items
                        Item 0: id-AMF-UE-NGAP-ID
                            ProtocolIE-Field
                                id: id-AMF-UE-NGAP-ID (10)
                                criticality: reject (0)
                                value
                                    AMF-UE-NGAP-ID: 32958739
                        Item 1: id-RAN-UE-NGAP-ID
                            ProtocolIE-Field
                                id: id-RAN-UE-NGAP-ID (85)
                                criticality: reject (0)
                                value
                                    RAN-UE-NGAP-ID: 1
                        Item 2: id-NAS-PDU
                        Item 3: id-UserLocationInformation

Frame 4: 102 bytes on wire (816 bits), 102 bytes captured (816 bits)
Internet Protocol Version 4, Src: 192.168.7.131, Dst: 192.168.1.140
Stream Control Transmission Protocol, Src Port: 38412 (38412), Dst Port: 36645 (36645)
NG Application Protocol (ErrorIndication)
    NGAP-PDU: initiatingMessage (0)
        initiatingMessage
            procedureCode: id-ErrorIndication (9)
            criticality: reject (0)
            value
                ErrorIndication
                    protocolIEs: 2 items
                        Item 0: id-AMF-UE-NGAP-ID
                            ProtocolIE-Field
                                id: id-AMF-UE-NGAP-ID (10)
                                criticality: ignore (1)
                                value
                                    AMF-UE-NGAP-ID: 12917860627
                        Item 1: id-Cause
                            ProtocolIE-Field
                                id: id-Cause (15)
                                criticality: ignore (1)
                                value
                                    Cause: radioNetwork (0)
                                        radioNetwork: unknown-local-UE-NGAP-ID (14)
```

## Additional Information ##
Full pcap file available if bytes are required for debugging.

richardjortega, 2022-03-02:
Screenshots below show truncation of `AMF-UE-NGAP-ID`, the 5G spec [TS 138 413 - V15.3.0 - 5G](https://www.etsi.org/deliver/etsi_ts/138400_138499/138413/15.03.00_60/ts_138413v150300p.pdf) should be longer than an uint32 `9.3.3.1 AMF UE NGAP ID` which spec asks for `INTEGER (0..240 -1)`

[image]

[image]

andrepuschmann, 2022-03-03:
Thanks for the report @richardjortega . We'll look into it.

richardjortega, 2022-03-03:
Thanks @andrepuschmann , additionally (but not blocking) is that 'AMF-UE-NGAP-ID' is supposed to be required/mandatory based on spec, but based on the code I saw while I was debugging  it was set to optional.

andrepuschmann, 2022-03-15:
Hi @richardjortega,
sorry for the delay. Just to understand right: Is the issue you're seeing happening with the first UE you create/attach to the 5GC after starting CoreScope? Or is the first working fine but subsequent attempts expose the issue?

richardjortega, 2022-03-15:
@andrepuschmann This is on first UE registration attempt after starting Corescope. Let me know if you need further logs.

andrepuschmann, 2022-03-23:
Hey @richardjortega,
looks like there is indeed some truncation happening. We've not seen this before because the cores we've been testing with all seem to have allocated a much smaller AMF-UE-NGAP-ID. Anyway, could you give this oneliner a shot?
```
diff --git a/srsgnb/hdr/stack/ngap/ngap_ue_utils.h b/srsgnb/hdr/stack/ngap/ngap_ue_utils.h
index f0067ef50..a1a455919 100644
--- a/srsgnb/hdr/stack/ngap/ngap_ue_utils.h
+++ b/srsgnb/hdr/stack/ngap/ngap_ue_utils.h
@@ -24,7 +24,7 @@ struct ngap_ue_ctxt_t {

   uint16_t                   rnti           = SRSRAN_INVALID_RNTI;
   uint32_t                   ran_ue_ngap_id = invalid_gnb_id;
-  srsran::optional<uint32_t> amf_ue_ngap_id;
+  srsran::optional<uint64_t> amf_ue_ngap_id;
   uint32_t                   gnb_cc_idx     = 0;
   struct timeval             init_timestamp = {};
```

Thanks
Andre

andrepuschmann, 2022-03-28:
Hey @richardjortega,
just realized that you'd need to expand the type at a few more places to make it compile.
```
diff --git a/srsgnb/hdr/stack/ngap/ngap.h b/srsgnb/hdr/stack/ngap/ngap.h
index 6c8a2d12f..65700a5b3 100644
--- a/srsgnb/hdr/stack/ngap/ngap.h
+++ b/srsgnb/hdr/stack/ngap/ngap.h
@@ -154,7 +154,7 @@ private:

     ue*            find_ue_rnti(uint16_t rnti);
     ue*            find_ue_gnbid(uint32_t gnbid);
-    ue*            find_ue_amfid(uint32_t amfid);
+    ue*            find_ue_amfid(uint64_t amfid);
     ue*            add_user(value_type user);
     void           erase(ue* ue_ptr);
     iterator       begin() { return users.begin(); }
@@ -190,7 +190,7 @@ private:
     ngap* ngap_ptr = nullptr;
   };

-  ue* handle_ngapmsg_ue_id(uint32_t gnb_id, uint32_t amf_id);
+  ue* handle_ngapmsg_ue_id(uint32_t gnb_id, uint64_t amf_id);

   srsran::proc_t<ng_setup_proc_t> ngsetup_proc;

diff --git a/srsgnb/hdr/stack/ngap/ngap_ue_utils.h b/srsgnb/hdr/stack/ngap/ngap_ue_utils.h
index f0067ef50..a1a455919 100644
--- a/srsgnb/hdr/stack/ngap/ngap_ue_utils.h
+++ b/srsgnb/hdr/stack/ngap/ngap_ue_utils.h
@@ -24,7 +24,7 @@ struct ngap_ue_ctxt_t {

   uint16_t                   rnti           = SRSRAN_INVALID_RNTI;
   uint32_t                   ran_ue_ngap_id = invalid_gnb_id;
-  srsran::optional<uint32_t> amf_ue_ngap_id;
+  srsran::optional<uint64_t> amf_ue_ngap_id;
   uint32_t                   gnb_cc_idx     = 0;
   struct timeval             init_timestamp = {};

diff --git a/srsgnb/src/stack/ngap/ngap.cc b/srsgnb/src/stack/ngap/ngap.cc
index b6e15cb0d..423af12ab 100644
--- a/srsgnb/src/stack/ngap/ngap.cc
+++ b/srsgnb/src/stack/ngap/ngap.cc
@@ -302,7 +302,7 @@ ngap::ue* ngap::user_list::find_ue_gnbid(uint32_t gnbid)
   return (it != users.end()) ? it->second.get() : nullptr;
 }

-ngap::ue* ngap::user_list::find_ue_amfid(uint32_t amfid)
+ngap::ue* ngap::user_list::find_ue_amfid(uint64_t amfid)
 {
   auto it = std::find_if(users.begin(), users.end(), [amfid](const user_list::pair_type& v) {
     return v.second->ctxt.amf_ue_ngap_id == amfid;
@@ -758,7 +758,7 @@ bool ngap::sctp_send_ngap_pdu(const asn1::ngap::ngap_pdu_c& tx_pdu, uint32_t rnt
  * @param amf_id amf_ue_ngap_id value stored in NGAP message
  * @return pointer to user if it has been found
  */
-ngap::ue* ngap::handle_ngapmsg_ue_id(uint32_t gnb_id, uint32_t amf_id)
+ngap::ue* ngap::handle_ngapmsg_ue_id(uint32_t gnb_id, uint64_t amf_id)
 {
   ue*     user_ptr     = users.find_ue_gnbid(gnb_id);
   ue*     user_amf_ptr = nullptr;
```

Thanks
Andre

richardjortega, 2022-03-29:
Hey @andrepuschmann , yes when I started playing around I noticed it wouldn't compile without these header/code changes. I re-compiled and still had truncation.

My logs showed the truncation right after the wire binary message was parsed. 

How does srsRAN unpack the message? Or where is the test for `ngap::handle_dl_nas_transport` where I can give it a captured binary message for testing?

Here is a generic hex stream that contains a six byte AMF UE NGAP ID

```
001dd8ee562868d79a3c3b790800450200888d7b40003f842315c0a80783c0a8018c960ccfd52e26b14c23fd725c03000010b381326c0001a000000000000003005665ad8dcd000100000000000000044042000003000a0006800302490d4c0055000200010026002b2a7e0056000200002141f71d73ddbbdf44668429ff5e7034f32010699bfa58e4418000aa03881e6068c7270000

andrepuschmann, 2022-03-29:
Ah, the 21.10 release doesn't have the full NGAP (test) code, which will be part of the 22.04 release . I'll see if we can provide that so you can use it for testing.

richardjortega, 2022-03-30:
@andrepuschmann after digging deeper most of today, I believe it can be traced to the structs are created that are used by the pack/unpack methods. I will try again tomorrow, but I am looking here: [https://github.com/srsran/srsRAN/blob/5275f33360f1b3f1ee8d1c4d9ae951ac7c4ecd4e/lib/include/srsran/asn1/asn1_utils.h#L1109-L1116](https://github.com/srsran/srsRAN/blob/5275f33360f1b3f1ee8d1c4d9ae951ac7c4ecd4e/lib/include/srsran/asn1/asn1_utils.h#L1109-L1116) after evaluating all the different datatypes. Seems like its being added to a struct that uses unint32 for upperbound limits.

andrepuschmann, 2022-03-30:
Thanks for that. I'll reproduce here. Do you mind sharing the full PCAP you mentioned in the original post, please?

richardjortega, 2022-04-04:
@andrepuschmann sent pcap to your email on your GitHub page as I couldn't upload here. Please confirm you've received it and let me know you need any help.

I am still debugging, but it seems the "upperbound" data type is uint32 down through the unpack/pack methods. Having a hard time unraveling or if there's a better method to write that'll support this use case.

andrepuschmann, 2022-04-04:
> @andrepuschmann sent pcap to your email on your GitHub page as I couldn't upload here. Please confirm you've received it and let me know you need any help.
>
> I am still debugging, but it seems the [...]

Hey @richardjortega, I did receive the PCAP and also confirmed that via mail. I've also been able to reproduce the issue and we are working on a fix.

richardjortega, 2022-04-11:
@andrepuschmann Glad you were able to reproduce. 

Please let me know if we can support this effort. I'm trying to get another cpp dev on one of our teams to help shoulder surf with me.

frankist, 2022-04-11:
Hey @richardjortega,

I am looking into this now. So, from what I gathered the problem is not in the encoding of the `seq_of` lower and upper bounds as those bounds are generally lower than 32 bits.

I was checking a unit test for the hex:
```
00044042000003000a0006800303cf37d00055000200010026002b2a7e00560002000021bc8de561f5b4a7058fdbe23b4e21da4520105ab8d1db137680001b1a8d3c984c0106
```
and from what I gathered, there is a overflow while decoding the amf-ue-ngap-id. Below, is the diff with the correction:
```
diff --git a/lib/src/asn1/asn1_utils.cc b/lib/src/asn1/asn1_utils.cc
index 70f4e174d..3c021720a 100644
--- a/lib/src/asn1/asn1_utils.cc
+++ b/lib/src/asn1/asn1_utils.cc
@@ -186,7 +186,7 @@ SRSASN_CODE unpack_bits(T& val, Ptr& ptr, uint8_t& offset, const uint8_t*
 max_pt
       n_bits = 0;
     } else {
       auto mask = static_cast<uint8_t>((1u << (8u - offset)) - 1u);
-      val += ((uint32_t)((*ptr) & mask)) << (n_bits - 8 + offset);
+      val += static_cast<T>((*ptr) & mask) << (n_bits - 8 + offset);
       n_bits -= 8 - offset;
       offset = 0;
       ptr++;
```

richardjortega, 2022-04-11:
Here are the logs (with debug logs I added) after both updating the data types above and the changes @frankist mentioned, there seems to a PDU decrypt issue related to length

```
2022-04-11T23:09:54.883126 [COMN   ] [D] [    0] Successfully bound to address 192.168.1.140:2152
2022-04-11T23:09:54.883156 [COMN   ] [D] [    0] RxSockets: socket fd=5 has been registered.
2022-04-11T23:09:54.883187 [NGAP   ] [I] Proc "AMF Connection" - Starting new AMF connection.
2022-04-11T23:09:54.883191 [NGAP   ] [I] Connecting to AMF 192.168.7.131:38412
2022-04-11T23:09:54.883211 [COMN   ] [D] [    0] Setting RTO_INFO options on SCTP socket. Association 0, Initial RTO 3000, Minimum RTO 1000, Maximum RTO 6000
2022-04-11T23:09:54.883216 [COMN   ] [D] [    0] Setting SCTP_INITMSG options on SCTP socket. Max attempts 3, Max init attempts timeout 5000
2022-04-11T23:09:54.883225 [COMN   ] [D] [    0] Successfully bound to address 192.168.1.140:0
2022-04-11T23:09:54.883227 [NGAP   ] [I] SCTP socket opened. fd=6
2022-04-11T23:09:54.883989 [NGAP   ] [I] SCTP socket connected with AMF. fd=6
2022-04-11T23:09:54.884012 [COMN   ] [D] [    0] RxSockets: socket fd=6 has been registered.
2022-04-11T23:09:54.884014 [NGAP   ] [I] SCTP socket established with AMF
2022-04-11T23:09:54.884156 [NGAP   ] [I] Tx NGAP SDU, ngSetupRequest
2022-04-11T23:09:54.884186 [NGAP   ] [I] Proc "AMF Connection" - NGSetupRequest sent. Waiting for response...
2022-04-11T23:09:54.888362 [NGAP   ] [I] AMF name: 03.0002.001.amf.5gc.mnc001.mcc001.3gppnetwork.org
2022-04-11T23:09:54.888367 [NGAP   ] [I] Proc "AMF Connection" - NGSetup procedure completed successfully
2022-04-11T23:09:54.896113 [REST   ] [I] Server running on port 8000
2022-04-11T23:09:59.369622 [SCOPE  ] [I] Adding UE with IMSI 001019990001001
2022-04-11T23:09:59.369911 [SCOPE  ] [I] Succesfully init UE 0 with IMSI 001019990001001
2022-04-11T23:09:59.369921 [REST   ] [I] Adding: at gnb id 0 imsi=001019990001001 and ue id 0
2022-04-11T23:09:59.380127 [NAS 0  ] [I] Switching on
2022-04-11T23:09:59.380365 [USIM 0 ] [I] Read Home PLMN Id=00101
2022-04-11T23:09:59.380369 [NAS 0  ] [I] Requesting IMSI attach (IMSI=001019990001001)
2022-04-11T23:09:59.380387 [NAS 0  ] [I] Sending Registration Request
2022-04-11T23:09:59.380463 [RRC    ] [I] Added user with RNTI b
2022-04-11T23:09:59.380476 [RRC    ] [D] Sending Setup Request to NGAP. RNTI=0xb, cause: mo-Signalling, 23 Bytes
2022-04-11T23:09:59.380563 [NGAP   ] [I] Tx NGAP SDU, InitialUEMessage, rnti=0xb
2022-04-11T23:09:59.380605 [PDCP 0 ] [I] Set RNTI=0xb
2022-04-11T23:09:59.380607 [RRC 0  ] [I] Successful RRC connection setup assigned RNTI=0xb.
2022-04-11T23:09:59.392515 [NGAP   ] [I] DownlinkNASTransport (ngap.cc#handle_dl_nas_transport) - AMF UE NGAP ID - 12911042992
2022-04-11T23:09:59.392520 [NGAP   ] [I] handle_ngapmsg_ue_id - start
2022-04-11T23:09:59.392522 [NGAP   ] [I] handle_ngapmsg_ue_id (passed in amf_id) - AMF UE NGAP ID - 12911042992
2022-04-11T23:09:59.392526 [NGAP   ] [E] ngap.cc#find_ue_amfid | amfid: 12911042992
2022-04-11T23:09:59.392531 [RRC    ] [D] Incoming data from NGAP, writing to UE RRC. RNTI=0xb, 42 Bytes.
2022-04-11T23:09:59.392534 [NAS 0  ] [I] DL PDU (length 42)
    0000: 7e 00 56 00 02 00 00 21 be f5 f0 9d a7 8c 1f 4d
    0010: 70 99 23 fd 48 20 95 62 20 10 da 20 59 dc c6 e2
    0020: 80 00 3c a8 ed 5a 8e 23 52 81
2022-04-11T23:09:59.392542 [NAS 0  ] [I] Decrypted DL PDU (length 42)
    0000: 7e 00 56 00 02 00 00 21 be f5 f0 9d a7 8c 1f 4d
    0010: 70 99 23 fd 48 20 95 62 20 10 da 20 59 dc c6 e2
    0020: 80 00 3c a8 ed 5a 8e 23 52 81
2022-04-11T23:09:59.392549 [NAS 0  ] [I] Handling Registration Request
2022-04-11T23:09:59.392551 [NAS 0  ] [I] Authentication request RAND
    0000: be f5 f0 9d a7 8c 1f 4d 70 99 23 fd 48 20 95 62
2022-04-11T23:09:59.392553 [NAS 0  ] [I] Authentication request AUTN
    0000: da 20 59 dc c6 e2 80 00 3c a8 ed 5a 8e 23 52 81
2022-04-11T23:09:59.392560 [NAS 0  ] [I] Serving network name 5G:mnc001.mcc001.3gppnetwork.org
2022-04-11T23:09:59.392614 [NAS 0  ] [I] Generated k_amf:
    0000: ef 80 ce 68 7d 04 a0 30 81 3d c4 cd 07 98 bb 83
    0010: 84 90 79 75 2f bb 1b b8 f3 c1 24 68 13 7c 92 0e
2022-04-11T23:09:59.392615 [NAS 0  ] [I] Network authentication successful
2022-04-11T23:09:59.392623 [NAS 0  ] [I] Sending Authentication Response
2022-04-11T23:09:59.392625 [NGAP   ] [I] Received RRC SDU
2022-04-11T23:09:59.392638 [NGAP   ] [I] UplinkNASTransport (ngap_ue#send_ul_nas_transport) | AMF UE NGAP ID - 12911042992
2022-04-11T23:09:59.392650 [NGAP   ] [I] Tx NGAP SDU, UplinkNASTransport, rnti=0xb
2022-04-11T23:09:59.392675 [NAS 0  ] [I] Generated res_star (16):
    0000: cb f4 54 69 d0 35 9c 78 22 37 a2 4a cb 5a 71 56
2022-04-11T23:09:59.398581 [NGAP   ] [I] DownlinkNASTransport (ngap.cc#handle_dl_nas_transport) - AMF UE NGAP ID - 12911042992
2022-04-11T23:09:59.398587 [NGAP   ] [I] handle_ngapmsg_ue_id - start
2022-04-11T23:09:59.398589 [NGAP   ] [I] handle_ngapmsg_ue_id (passed in amf_id) - AMF UE NGAP ID - 12911042992
2022-04-11T23:09:59.398594 [RRC    ] [D] Incoming data from NGAP, writing to UE RRC. RNTI=0xb, 15 Bytes.
2022-04-11T23:09:59.398598 [NAS 0  ] [I] DL PDU (length 15)
    0000: 7e 03 88 db eb d2 00 7e 00 5d 02 00 02 e0 60
2022-04-11T23:09:59.398606 [NAS 0  ] [I] Decrypted DL PDU (length 15)
    0000: 7e 03 88 db eb d2 00 7e 00 5d 02 00 02 e0 60
2022-04-11T23:09:59.398612 [NAS 0  ] [I] Handling Security Mode Command
[... cut ...]
2022-04-11T23:10:17.455150 [RRC    ] [D] Incoming data from NGAP, writing to UE RRC. RNTI=0xb, 56 Bytes.
2022-04-11T23:10:17.455154 [NAS 0  ] [I] DL PDU (length 56)
    0000: 7e 02 c4 2f a0 5e 01 7e 00 42 01 01 77 00 0b f2
    0010: 00 f1 10 01 00 83 01 8e e1 b0 54 0d 02 00 f1 10
    0020: 00 00 01 00 00 02 00 00 03 15 02 01 01 11 02 10
    0030: 01 31 02 01 01 5e 01 05
2022-04-11T23:10:17.455167 [NAS 0  ] [I] Integrity check ok. Local: count=1, Received: count=1  [c4 2f a0 5e]
2022-04-11T23:10:17.455170 [NAS 0  ] [I] Decrypted DL PDU (length 56)
    0000: 7e 02 
```

richardjortega, 2022-04-12:
> ```
> 2022-04-11T23:10:11.455257 [NAS 0  ] [I] Decrypted DL PDU (length 56)
>     0000: 7e 02 c4 2f a0 5e 01 7e 00 42 01 01 77 00 0b f2
>     0010: 00 f1 10 01 00 83 01 8e e1 b0 54 0d 02 00 f1 10
>     0020: 00 00 01 00 00 02 00 00 03 15 02 01 01 11 02 10
>     0030: 01 31 02 01 01 5e 01 05
> 2022-04-11T23:10:11.455267 [ASN1   ] [E] [    0] Decoding Failed (5GS tracking area identity list): Length (13) does not equal expected length 7
> 2022-04-11T23:10:11.455274 [ASN1   ] [E] [    0] [/home/auser/corescope/srsRAN/lib/src/asn1/nas_5g_msg.cc][524] Decoding failure.
> 2022-04-11T23:10:11.455276 [ASN1   ] [E] [    0] [/home/auser/corescope/srsRAN/lib/src/asn1/nas_5g_msg.cc][3929] Decoding failure.
> 2022-04-11T23:10:11.455278 [ASN1   ] [E] [    0] [/home/auser/corescope/srsRAN/lib/src/asn1/nas_5g_msg.cc][3905] Decoding failure.
> 2022-04-11T23:10:11.455280 [NAS 0  ] [E] Unable to unpack complete NAS pdu
> ```

Was able to debug this by inspecting the NAS 5GS message. The error is based on srsRAN not supporting multiple TAC lists (I believe I can work around this - but 3GPP Release 16 describes this in Section 9.11.3.9).

[image]

andrepuschmann, 2022-04-14:
Thanks for the further digging @richardjortega. We can confirm the NAS unpacking issue. It's a separate one. Would you mind creating a new issue on Github with all the info and the hexdump so we can track it?

richardjortega, 2022-07-05:
@andrepuschmann  Closing issue as this was merged into Release 22.04 (f2dff0b)

closed 2022-07-05 by richardjortega