srsran/srsRAN_4G #820: Magma + srsRAN downlink not working.

ShubhamTatvamasi, 2022-03-01:
Hi all,

I have connected Magma with docker based srsRAN and tunnel has been created on srsRAN VM.
```
ubuntu@ubuntu:~/srsRAN-demo$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 46:09:53:16:2d:bc brd ff:ff:ff:ff:ff:ff
    inet 192.168.5.57/24 brd 192.168.5.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::4409:53ff:fe16:2dbc/64 scope link
       valid_lft forever preferred_lft forever
3: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default
    link/ether 02:42:86:65:fd:ed brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
5: tun_srsue: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 500
    link/none
    inet 192.168.128.13/24 scope global tun_srsue
       valid_lft forever preferred_lft forever
```

Upload test is working:
```
ubuntu@ubuntu:~/srsRAN-demo$ iperf3 -c 192.168.128.1
Connecting to host 192.168.128.1, port 5201
[  5] local 192.168.128.13 port 52666 connected to 192.168.128.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.25 MBytes  18.9 Mbits/sec    0    137 KBytes
[  5]   1.00-2.00   sec  2.30 MBytes  19.3 Mbits/sec    0    238 KBytes
[  5]   2.00-3.00   sec  2.36 MBytes  19.8 Mbits/sec    0    332 KBytes
[  5]   3.00-4.00   sec  1.43 MBytes  12.0 Mbits/sec   15    284 KBytes
[  5]   4.00-5.00   sec  2.49 MBytes  20.9 Mbits/sec    0    324 KBytes
[  5]   5.00-6.00   sec  2.42 MBytes  20.3 Mbits/sec    0    363 KBytes
[  5]   6.00-7.00   sec  1.62 MBytes  13.6 Mbits/sec    0    387 KBytes
[  5]   7.00-8.00   sec  2.42 MBytes  20.3 Mbits/sec    2    291 KBytes
[  5]   8.00-9.00   sec  1.62 MBytes  13.5 Mbits/sec    0    313 KBytes
[  5]   9.00-10.00  sec  2.42 MBytes  20.3 Mbits/sec    0    324 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  21.3 MBytes  17.9 Mbits/sec   17             sender
[  5]   0.00-10.15  sec  19.8 MBytes  16.4 Mbits/sec                  receiver

iperf Done.
```

But download test is not working:
```
ubuntu@ubuntu:~/srsRAN-demo$ iperf3 -c 192.168.128.1 -R
Connecting to host 192.168.128.1, port 5201
Reverse mode, remote host 192.168.128.1 is sending
[  5] local 192.168.128.13 port 52670 connected to 192.168.128.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.04  sec  76.4 KBytes  62.3 Kbits/sec    5             sender
[  5]   0.00-10.00  sec  0.00 Bytes  0.00 bits/sec                  receiver

iperf Done.
```

How can I resolve this?

Thanks

andrepuschmann, 2022-03-03:
You need to send logs for the eNB, otherwise nobody can help you.

ShubhamTatvamasi, 2022-03-03:
Thanks for reply @andrepuschmann, I have attached the logs below.

```
ubuntu@ubuntu:~$ docker ps
CONTAINER ID   IMAGE     COMMAND                  CREATED              STATUS              PORTS     NAMES
73d8e0cf8097   srsran    "srsue --usim.algo=m…"   About a minute ago   Up About a minute             srsue
e55534ed7aab   srsran    "srsenb --enb.mme_ad…"   About a minute ago   Up About a minute             srsenb
```
```
ubuntu@ubuntu:~$ docker logs srsue
Couldn't open , trying /root/.config/srsran/ue.conf
Couldn't open /root/.config/srsran/ue.conf either, trying /etc/srsran/ue.conf
Reading configuration file /etc/srsran/ue.conf...

Built in Release mode using 21.10.0.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Available RF device list: zmq
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.Closing stdin thread.

Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=50, tti=1941, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=0
Network attach successful. IP: 192.168.128.14
E
```
```
ubuntu@ubuntu:~$ docker logs srsenb
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Couldn't open /root/.config/srsran/enb.conf either, trying /etc/srsran/enb.conf
Reading configuration file /etc/srsran/enb.conf...
Failed to `mlockall`: {}
Built in Release mode using 21.10.0.

Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6
Available RF device list: zmq
CHx base_srate=23.04e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
Closing stdin thread.
RACH:  tti=1941, cc=0, preamble=50, offset=0, temp_crnti=0x46
User 0x46 connected
Disconnecting rnti=0x46.
```

andrepuschmann, 2022-07-21:
Hey, not sure if the issue still persists but in order to be able to help you should really send the eNB and UE logs. Not just the stdout. Anyway, I am closing the ticket for the moment. Please feel free to reopen if needed.

closed 2022-07-21 by andrepuschmann