srsran/srsRAN_4G #827: Carrier Aggregation - RX channel 2 out of range for configured RX frontends

neilrep, 2022-03-08:
## Issue Description ##
Srsenb running error (LookupError: IndexError: multi_usrp: RX channel 2 out of range for configured RX frontends)

## Setup Details ##
We're running an EPC --> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 20.10.1 , UHD 3.15.0

## Expected Behavior ##
No Error upon running srsenb

## Actual Behaviour ##
After configuring and changing parameters for Carrier Aggregation using SDR Hardware, (link: https://docs.srslte.com/en/rfsoc/app_notes/source/2ca/source/index.html) As I run the srsenb, we facing UHD RF error.

see attached file of my configuration from rr.conf and enb.conf, I also attached the picture of srsenb error. See screen grab below.

[image]

## Additional Information ##

[rr (1).txt](https://github.com/srsran/srsRAN/files/8202075/rr.1.txt)
[enb (1).txt](https://github.com/srsran/srsRAN/files/8202076/enb.1.txt)

NeilJohnR, 2022-05-04:
I just trying again CA and following the steps from https://docs.srslte.com/en/rfsoc/app_notes/source/2ca/source/index.html#introduction but I got the same error as you can see the picture above.

How's your testing for CA? have you any config changes? until now I don't get any solution

andrepuschmann, 2022-05-04:
As @WingPig99 said, you can't run CA with a B210 because you need carriers in different bands, which the B210 can't produce given the single LO shared between the channels. Closing the issue therefore.

closed 2022-05-04 by andrepuschmann