srsran/srsRAN_4G #828: srsUE Failed to attach in NSA mode,

arfanhayder, 2022-03-08:
## Issue Description ##
srsUE failed to attach with following error

Waiting PHY to initialize ... done!
Attaching UE...
[ERROR] [UHD RF] RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_0 specifies 1.92e+06, node 0/DDC_1 specifies 1.152e+07.
[ERROR] [UHD RF] RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_0 specifies 1.92e+06, node 0/DDC_1 specifies 1.152e+07.
...................................RF status: O=2, U=0, L=0

## Setup Details ##
Followed 
https://docs.srsran.com/en/latest/app_notes/source/5g_nsa_zmq/source/index.html

Machine-1:
Ubuntu 20.04
Running srsepc(NSA, version 21.10) amd srsenb (NSA, version 21.10)
UHD 3.15 installed
Connects to a USRP X310 with UHD 3.15

Machine-2:
Ubuntu 20.04
Running srsue (NSA version 21.10)
UHD 3.15 installed
Connects to a USRP N310 with UHD 3.15

## Expected Behavior ##
UE conects to eNB

## Actual Behaviour ##
Reading configuration file ue.conf...

Built in RelWithDebInfo mode using 21.10.0.

Opening 2 channels in RF device=uhd with args=type=x300,sampling_rate=11.52e6,lo_freq_offset_hz=11.52e6
Available RF device list: UHD 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,lo_freq_offset_hz=11.52e6,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1307 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1301 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
Waiting PHY to initialize ... done!
Attaching UE...
[ERROR] [UHD RF] RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_0 specifies 1.92e+06, node 0/DDC_1 specifies 1.152e+07.
[ERROR] [UHD RF] RuntimeError: Multiple sampling rates upstream of RX Terminator 0: RuntimeError: Node 0/DDC_0 specifies 1.92e+06, node 0/DDC_1 specifies 1.152e+07.
...................................RF status: O=2, U=0, L=0
.....................................................
................
..........
...........................................................
............................................................................................................................................................................
.................................
..........................................
...........................................................................
.................................................................................................................................................^CStopping ..
Couldn't stop after 5s. Forcing exit.
Killed

## Steps to reproduce the problem ##

[enb.conf.txt](https://github.com/srsran/srsRAN/files/8207613/enb.conf.txt)
[enb.log.txt](https://github.com/srsran/srsRAN/files/8207614/enb.log.txt)
[epc.conf.txt](https://github.com/srsran/srsRAN/files/8207615/epc.conf.txt)
[epc.log.txt](https://github.com/srsran/srsRAN/files/8207616/epc.log.txt)
[ue.conf.txt](https://github.com/srsran/srsRAN/files/8207618/ue.conf.txt)
[ue.log.txt](https://github.com/srsran/srsRAN/files/8207619/ue.log.txt)

[user_db.csv](https://github.com/srsran/srsRAN/files/8207655/user_db.csv)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2022-03-08:
Please check the `srate` param in the UE config and RF args. You find examples in the appnotes.

andrepuschmann, 2022-03-08:
Try this:
```
[rf]
srate = 23.04e6
device_name = uhd
device_args = type=x300,sampling_rate=23.04e6
```

arfanhayder, 2022-03-08:
Reading configuration file ue.conf...

Built in RelWithDebInfo mode using 21.10.0.

Opening 2 channels in RF device=uhd with args=type=x300,sampling_rate=23.04e6
Available RF device list: UHD 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1320 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1311 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
Waiting PHY to initialize ... done!
Attaching UE...
./build/srsran-V6Lskk/srsran-21.10/lib/src/phy/ue/ue_sync.c:772: Error receiving samples
/build/srsran-V6Lskk/srsran-21.10/lib/src/phy/ue/ue_cell_search.c:317: Error calling srsran_ue_sync_work()
/build/srsran-V6Lskk/srsran-21.10/lib/src/phy/ue/ue_cell_search.c:277: Error searching cell
ue--logs

[ue.log.txt](https://github.com/srsran/srsRAN/files/8208389/ue.log.txt)

andrepuschmann, 2022-03-08:
Maybe decrease both to 11.52e6, also increase MTU of your interface. Does it sustain streaming 2x11.52e6 samples?

bondadakumarsai, 2022-03-09:
Facing the same issue.

arfanhayder, 2022-03-09:
correction from the 
Setup Details 

Machine-2:
Ubuntu 20.04
Running srsue (NSA version 21.10)
UHD 3.15 installed
Connects to a USRP X310 with UHD 3.15 **(USRP X310)**

arfanhayder, 2022-03-10:
@andrepuschmann

arfanhayder, 2022-03-10:
After settings MTU = 9000, UE keep searching cell or there is problem with eNB side ?
[following output] 

sudo srsue ue.conf 
Reading configuration file ue.conf...

Built in RelWithDebInfo mode using 21.10.0.

Opening 2 channels in RF device=uhd with args=type=x300,sampling_rate=11.52e6,send_frame_size=8000,recv_frame_size=8000
Available RF device list: UHD 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,send_frame_size=8000,recv_frame_size=8000,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1304 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1313 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
Waiting PHY to initialize ... done!
Attaching UE...
........
.
.
...
.
RF status: O=2, U=0, L=0
.......
RF status: O=2, U=0, L=0
.
..
.
RF status: O=6, U=0, L=0
..
.
.
.
...RF status: O=4, U=0, L=0
...
.
RF status: O=4, U=0, L=0
...
........
.
.
.
.RF status: O=3, U=0, L=0
.....
..
RF status: O=4, U=0, L=0
.
...
....RF status: O=2, U=0, L=0
..
....
RF status: O=6, U=0, L=0
.....
.
.
.RF status: O=2, U=0, L=0
.......
[... 109 lines cut ...]
Type <t> to view trace
Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=1 nof_prb=52
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.504466 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.506639 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.512159 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.514332 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.516679 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.519025 >= 0. Starting new burst...
[ERROR] [X300] [192.168.10.2](http://192.168.10.2/): x300 fw communication failure #1
EnvironmentError: IOError: x300 fw poke32 - reply timed out
[ERROR] [X300] [192.168.10.2](http://192.168.10.2/): x300 fw communication failure #1
EnvironmentError: IOError: x300 fw poke32 - reply timed out
[ERROR] [X300] [192.168.10.2](http://192.168.10.2/): x300 fw communication failure #2
EnvironmentError: IOError: x300 fw poke32 - reply timed out
[ERROR] [X300] [192.168.10.2](http://192.168.10.2/): x300 fw communication failure #3
EnvironmentError: IOError: x300 fw poke32 - reply timed out
[ERROR] [UHD] An unexpected exception was caught in a task loop.The task loop will now exit, things may not work.EnvironmentError: IOError: [192.168.10.2](http://192.168.10.2/): x300 fw communication failure #3
EnvironmentError: IOError: x300 fw poke32 - reply timed out

[enb.log](https://github.com/srsran/srsRAN/files/8223824/enb.log)
[epc.log](https://github.com/srsran/srsRAN/files/8223828/epc.log)

 

[ue.log](https://github.com/srsran/srsRAN/files/8223774/ue.log)

andrepuschmann, 2022-03-10:
You have very high CFO, you need to sync both devices. Also the UE has many RF errors, they need to go away. Otherwise the UE will loose sync all the time.

arfanhayder, 2022-03-10:
frequency offset mentioned in the following notes, that is why i added it 
https://docs.srsran.com/en/latest/usermanuals/source/srsue/source/3_ue_trouble.html

After changing 
no errors in the console of enb 

but UE still doesn't conects 
UE log shows following error 
2022-03-10T16:15:52.272214 [RF     ] [I] Overflow
2022-03-10T16:15:52.325314 [RF     ] [I] Overflow
2022-03-10T16:15:52.378559 [RF     ] [I] Overflow
2022-03-10T16:15:52.431654 [RF     ] [I] Overflow
2022-03-10T16:15:52.484729 [RF     ] [I] Overflow
2022-03-10T16:15:52.537836 [RF     ] [I] Overflow
2022-03-10T16:15:52.590892 [RF     ] [I] Overflow
2022-03-10T16:15:52.643964 [RF     ] [I] Overflow
2022-03-10T16:15:52.697040 [RF     ] [I] Overflow
2022-03-10T16:15:52.750224 [RF     ] [I] Overflow
2022-03-10T16:15:52.803297 [RF     ] [I] Overflow

---------------------------------------------------------------------

do you think building from the source does make a difference, 
Since i was going with package installation

andrepuschmann, 2022-03-10:
I don't know, you never mentioned your system config. But try with source yes. Enable CPU performance governor. With that many overflows it won't work.

arfanhayder, 2022-03-11:
So i reinstalled, 
Srsenb, Srsepc, NSA  from source with UHD 3.15.0, 
System config:  intel nuc, 32 GB  
X310 

srsUE,  NSA from source with UHD3.15.0, 
Dell i7, 16GB ram, 
First i tested with b210, 
It Connects with enb, with following change
[rf]
device_args = auto

it connects,,,,  

however when i test srsUE with x310
updating the device args according to app notes 
it couldn't connects to cell keep searching

do you any tested srsUE
 [device_args] for  x310  

those in app notes not working for me

arfanhayder, 2022-03-15:
Exp. 1: Updated UHD version 4.2.0, build from source
Same System Config. as
 
Reading configuration file ue.conf...

Built in Release mode using commit 5275f3336 on branch master.

Opening 2 channels in RF device=uhd with args=type=x300,sampling_rate=11.52e6,lo_freq_offset_hz=11.52e6,send_frame_size=8000,recv_frame_size=8000,num_send_frames=64,num_recv_frames=64
Available RF device list: UHD 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.git-435-g220710b1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,lo_freq_offset_hz=11.52e6,send_frame_size=8000,recv_frame_size=8000,num_send_frames=64,num_recv_frames=64,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[WARNING] [RFNOC::GRAPH] One or more blocks timed out during flush!
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
Waiting PHY to initialize ... done!
Attaching UE...
.
Found Cell:  Mode=FDD, PCI=1, PRB=25, Ports=1, CP=Normal, CFO=2.1 KHz
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.729411 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.731584 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.736104 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.739277 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.742624 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.74597 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.748143 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.749143 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.752663 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.754837 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.757183 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.759356 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.766703 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.773049 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.775223 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.778396 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.782569 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.785916 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.789089 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.791262 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.795435 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.799609 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.803955 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.806129 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.808648 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.810822 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 0.812995 >= 0. Starting new burst...
RF status: O=40, U=0, L=16

However when try with b210 with device_args = auto, 
UE gets connected for some time 
UE b210 

Reading configuration file ue.conf...

Built in Release mode using commit 5275f3336 on branch master.

Opening 2 channels in RF device=uhd with args=default
Available RF device list: UHD 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.git-435-g220710b1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [B200] Loading firmware image: /usr/local/share/uhd/images/usrp_b200_fw.hex...
Opening USRP channels=2, args: type=b200,master_clock_rate=23.04e6
[... 191 lines cut ...]
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x1; eNB GTP-U Address 127.0.1.1
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 1010123456789
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Sending EMM Information
Received UE Context Release Request. MME-UE S1AP Id 3
There are active E-RABs, send release access bearers request
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Received UE Context Release Complete. MME-UE S1AP Id 3
UE Context Release Completed.

[logs]
[ue_x310.log](https://github.com/srsran/srsRAN/files/8254449/ue_x310.log)
[ue_b210.log](https://github.com/srsran/srsRAN/files/8254456/ue_b210.log)
[enb_x310.log](https://github.com/srsran/srsRAN/files/8254489/enb_x310.log)
[enb.conf.txt](https://github.com/srsran/srsRAN/files/8254543/enb.conf.txt)

[ue.conf.txt](https://github.com/srsran/srsRAN/files/8254514/ue.conf.txt)

Do you think system config. is the issue , i should try 32GB or higher on UE-end as well

EXP:2,  i have tried with UHD 3.14.1 as well but same error

kazim425, 2024-10-02:
Dear,

The issue still seems unresolved. I have tested using UHD 4.7 on usrp x300. With only 4G enabled in the ue.conf, it works perfectly fine (Even without the external clock at UE side ) I reach 20 Mbits/s speed using PRB=50. With only NR enabled, it generates error searching cell messaege. When using both EUTRA and NR enabled, it becomes unstable.