srsran/srsRAN_4G #829: [RRC Reestablishment]User 0x48 requesting RRC Reestablishment as 0x47. Cause: otherFailure

ysgaga, 2022-03-09:
## Issue Description ##
Hi,
    I use Router(4G) to connect to srsran. It is success to get ip, and it can ping 172.16.0.1.
   But it will be high ping delay every few seconds. I check epc.log and enb.log:
[image]
    When "User 0x48 requesting RRC Reestablishment as 0x47. Cause: otherFailure" happens, it may be high ping delay.

## Setup Details ##
I am new to this platform, the system is new too. I just change "n_prb = 25", add user to user_db.csv.
model name	: Intel(R) Core(TM) i5-4210U CPU @ 1.70GHz
srsran 4.15.0-47-lowlatency #50~16.04.1-Ubuntu SMP PREEMPT Fri Mar 15 17:23:09 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

enb.log
```
root@srsran:/srsRAN/srsenb# sudo srsenb enb.conf
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in Release mode using commit 5275f33 on branch heads/release_21_10.

Opening 1 channels in RF device=default with args=default
Available RF device list: UHD 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.15.0.0-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened
```
epc.log
`root@milesight-srsran:/srsRAN/srsepc# sudo srsepc epc.conf

Built in Release mode using commit 5275f33 on branch heads/release_21_10.

---  Software Radio Systems EPC  ---

Reading configuration file epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
`

## Expected Behavior ##
I hope it can keep on line

## Actual Behaviour ##
IN Setup Details

## Steps to reproduce the problem ##
sudo srsepc epc.conf
sudo srsenb enb.conf
AND connect to srsran. The system is new!

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

anilgurses, 2022-03-11:
Just try to change your cpu governor to performance with the command below, 
`for ((i=0;i<$(nproc);i++)); do sudo cpufreq-set -c $i -r -g performance; done`

If it doesn't work, your computer is not sufficient for srsran. Also, your CPU is old..

ysgaga, 2022-03-14:
> 

Thank you anilgurses, I try and it has improved, but not solved.
I will prepare a new PC. And I have another question,
what is the Minimum CPU Requirements for srsran？

ysgaga, 2022-03-14:
@anilgurses 

Thank you anilgurses, I try and it has improved, but not solved.
I will prepare a new PC. And I have another question,
what is the Minimum CPU Requirements for srsran？

matan1008, 2022-03-14:
I had the same issue, #748. The specs in the docs solved my problem.

andrepuschmann, 2022-03-14:
Closing the issue. It's related to #748 and others. 
Thanks

closed 2022-03-14 by andrepuschmann