srsran/srsRAN_4G #833: LimeSDR Calibration issue

seeklakhan, 2022-03-16:
Getting these errors can anyone guide me how to fix these errors, and reason for these errors i have already tried to increase gain but still no luck thanks

[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=0)
[ERROR] SetFrequencyCGEN(-nan MHz) failed
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=0)
[ERROR] SetFrequencyCGEN(92.16 MHz) failed
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
[ERROR] Tx Calibration: MCU error 2 (CGEN tune failed)
[ERROR] Rx calibration: MCU error 2 (CGEN tune failed)

jtang613, 2022-03-16:
Recommend asking the LimeSDR community over here: https://discourse.myriadrf.org/

EugeneBogush, 2022-05-20:
SoapySDRUtil --info
Lib Version: v0.8.1-2build1
API Version: v0.8.0
ABI Version: v0.8

By using this version the problem is resolved.

andrepuschmann, 2022-05-23:
Thanks for the hin @EugeneBogush . In any case, it's unlikely to be an srsRAN issue and therefore I am closing the ticket.

closed 2022-05-23 by andrepuschmann