srsran/srsRAN_4G #834: make test fails on latest git head https://github.com/srsran/srsRAN/commit/5275f33360f1b3f1ee8d1c4d9ae951ac7c4ecd4e

ZeroChaos-, 2022-03-20:
```
92% tests passed, 114 tests failed out of 1356

Label Time Summary:
chest    =  38.28 sec*proc (22 tests)
fec      =   1.40 sec*proc (11 tests)
ldpc     =   0.03 sec*proc (1 test)
lib      = 809.56 sec*proc (1005 tests)
lte      = 747.76 sec*proc (931 tests)
nr       = 253.83 sec*proc (219 tests)
pdcp     =   0.78 sec*proc (3 tests)
phch     = 296.34 sec*proc (485 tests)
phy      = 781.63 sec*proc (991 tests)
polar    =   0.00 sec*proc (1 test)
rlc      =  27.15 sec*proc (11 tests)
turbo    =   1.36 sec*proc (9 tests)

Total Test time (real) = 487.56 sec

The following tests FAILED:
	284 - pdsch_test_cdd_6 (Failed)
	285 - pdsch_test_cdd_12 (Failed)
	286 - pdsch_test_cdd_25 (Failed)
	287 - pdsch_test_cdd_50 (Failed)
	288 - pdsch_test_cdd_75 (Failed)
	289 - pdsch_test_cdd_100 (Failed)
	849 - phy_dl_test-p6-t3-E0-m0 (Failed)
	850 - phy_dl_test-p6-t3-E0-m7 (Failed)
	851 - phy_dl_test-p6-t3-E0-m14 (Failed)
	852 - phy_dl_test-p6-t3-E0-m21 (Failed)
	853 - phy_dl_test-p6-t3-E0-m28 (Failed)
	869 - phy_dl_test-p6-t3-E0-q-m0 (Failed)
	870 - phy_dl_test-p6-t3-E0-q-m7 (Failed)
	871 - phy_dl_test-p6-t3-E0-q-m14 (Failed)
	872 - phy_dl_test-p6-t3-E0-q-m21 (Failed)
	873 - phy_dl_test-p6-t3-E0-q-m27 (Failed)
	889 - phy_dl_test-p15-t3-E0-m0 (Failed)
	890 - phy_dl_test-p15-t3-E0-m7 (Failed)
	891 - phy_dl_test-p15-t3-E0-m14 (Failed)
	892 - phy_dl_test-p15-t3-E0-m21 (Failed)
	893 - phy_dl_test-p15-t3-E0-m28 (Failed)
	909 - phy_dl_test-p15-t3-E0-q-m0 (Failed)
	910 - phy_dl_test-p15-t3-E0-q-m7 (Failed)
	911 - phy_dl_test-p15-t3-E0-q-m14 (Failed)
	912 - phy_dl_test-p15-t3-E0-q-m21 (Failed)
	913 - phy_dl_test-p15-t3-E0-q-m26 (Failed)
	929 - phy_dl_test-p25-t3-E0-m0 (Failed)
	930 - phy_dl_test-p25-t3-E0-m7 (Failed)
	931 - phy_dl_test-p25-t3-E0-m14 (Failed)
	932 - phy_dl_test-p25-t3-E0-m21 (Failed)
	933 - phy_dl_test-p25-t3-E0-m28 (Failed)
	949 - phy_dl_test-p25-t3-E0-q-m0 (Failed)
	950 - phy_dl_test-p25-t3-E0-q-m7 (Failed)
	951 - phy_dl_test-p25-t3-E0-q-m14 (Failed)
	952 - phy_dl_test-p25-t3-E0-q-m21 (Failed)
	953 - phy_dl_test-p25-t3-E0-q-m27 (Failed)
	969 - phy_dl_test-p50-t3-E0-m0 (Failed)
	970 - phy_dl_test-p50-t3-E0-m7 (Failed)
	971 - phy_dl_test-p50-t3-E0-m14 (Failed)
	972 - phy_dl_test-p50-t3-E0-m21 (Failed)
	973 - phy_dl_test-p50-t3-E0-m28 (Failed)
	989 - phy_dl_test-p50-t3-E0-q-m0 (Failed)
	990 - phy_dl_test-p50-t3-E0-q-m7 (Failed)
	991 - phy_dl_test-p50-t3-E0-q-m14 (Failed)
	992 - phy_dl_test-p50-t3-E0-q-m21 (Failed)
	993 - phy_dl_test-p50-t3-E0-q-m27 (Failed)
	1009 - phy_dl_test-p75-t3-E0-m0 (Failed)
	1010 - phy_dl_test-p75-t3-E0-m7 (Failed)
	1011 - phy_dl_test-p75-t3-E0-m14 (Failed)
	1012 - phy_dl_test-p75-t3-E0-m21 (Failed)
	1013 - phy_dl_test-p75-t3-E0-m28 (Failed)
	1029 - phy_dl_test-p75-t3-E0-q-m0 (Failed)
	1030 - phy_dl_test-p75-t3-E0-q-m7 (Failed)
	1031 - phy_dl_test-p75-t3-E0-q-m14 (Failed)
	1032 - phy_dl_test-p75-t3-E0-q-m21 (Failed)
	1033 - phy_dl_test-p75-t3-E0-q-m27 (Failed)
	1049 - phy_dl_test-p100-t3-E0-m0 (Failed)
	1050 - phy_dl_test-p100-t3-E0-m7 (Failed)
	1051 - phy_dl_test-p100-t3-E0-m14 (Failed)
	1052 - phy_dl_test-p100-t3-E0-m21 (Failed)
	1053 - phy_dl_test-p100-t3-E0-m28 (Failed)
	1069 - phy_dl_test-p100-t3-E0-q-m0 (Failed)
	1070 - phy_dl_test-p100-t3-E0-q-m7 (Failed)
	1071 - phy_dl_test-p100-t3-E0-q-m14 (Failed)
	1072 - phy_dl_test-p100-t3-E0-q-m21 (Failed)
	1073 - phy_dl_test-p100-t3-E0-q-m27 (Failed)
	1087 - phy_dl_test-p6-t3-E1-m0 (Failed)
	1088 - phy_dl_test-p6-t3-E1-m7 (Failed)
	1089 - phy_dl_test-p6-t3-E1-m14 (Failed)
	1090 - phy_dl_test-p6-t3-E1-m21 (Failed)
	1103 - phy_dl_test-p6-t3-E1-q-m0 (Failed)
	1104 - phy_dl_test-p6-t3-E1-q-m7 (Failed)
	1105 - phy_dl_test-p6-t3-E1-q-m14 (Failed)
	1106 - phy_dl_test-p6-t3-E1-q-m21 (Failed)
	1119 - phy_dl_test-p15-t3-E1-m0 (Failed)
	1120 - phy_dl_test-p15-t3-E1-m7 (Failed)
	1121 - phy_dl_test-p15-t3-E1-m14 (Failed)
	1122 - phy_dl_test-p15-t3-E1-m21 (Failed)
	1135 - phy_dl_test-p15-t3-E1-q-m0 (Failed)
	1136 - phy_dl_test-p15-t3-E1-q-m7 (Failed)
	1137 - phy_dl_test-p15-t3-E1-q-m14 (Failed)
	1138 - phy_dl_test-p15-t3-E1-q-m21 (Failed)
	1151 - phy_dl_test-p25-t3-E1-m0 (Failed)
	1152 - phy_dl_test-p25-t3-E1-m7 (Failed)
	1153 - phy_dl_test-p25-t3-E1-m14 (Failed)
	1154 - phy_dl_test-p25-t3-E1-m21 (Failed)
	1167 - phy_dl_test-p25-t3-E1-q-m0 (Failed)
	1168 - phy_dl_test-p25-t3-E1-q-m7 (Failed)
	1169 - phy_dl_test-p25-t3-E1-q-m14 (Failed)
	1170 - phy_dl_test-p25-t3-E1-q-m21 (Failed)
	1183 - phy_dl_test-p50-t3-E1-m0 (Failed)
	1184 - phy_dl_test-p50-t3-E1-m7 (Failed)
	1185 - phy_dl_test-p50-t3-E1-m14 (Failed)
	1186 - phy_dl_test-p50-t3-E1-m21 (Failed)
	1199 - phy_dl_test-p50-t3-E1-q-m0 (Failed)
	1200 - phy_dl_test-p50-t3-E1-q-m7 (Failed)
	1201 - phy_dl_test-p50-t3-E1-q-m14 (Failed)
	1202 - phy_dl_test-p50-t3-E1-q-m21 (Failed)
	1215 - phy_dl_test-p75-t3-E1-m0 (Failed)
	1216 - phy_dl_test-p75-t3-E1-m7 (Failed)
	1217 - phy_dl_test-p75-t3-E1-m14 (Failed)
	1218 - phy_dl_test-p75-t3-E1-m21 (Failed)
	1231 - phy_dl_test-p75-t3-E1-q-m0 (Failed)
	1232 - phy_dl_test-p75-t3-E1-q-m7 (Failed)
	1233 - phy_dl_test-p75-t3-E1-q-m14 (Failed)
	1234 - phy_dl_test-p75-t3-E1-q-m21 (Failed)
	1247 - phy_dl_test-p100-t3-E1-m0 (Failed)
	1248 - phy_dl_test-p100-t3-E1-m7 (Failed)
	1249 - phy_dl_test-p100-t3-E1-m14 (Failed)
	1250 - phy_dl_test-p100-t3-E1-m21 (Failed)
	1263 - phy_dl_test-p100-t3-E1-q-m0 (Failed)
	1264 - phy_dl_test-p100-t3-E1-q-m7 (Failed)
	1265 - phy_dl_test-p100-t3-E1-q-m14 (Failed)
	1266 - phy_dl_test-p100-t3-E1-q-m21 (Failed)
Errors while running CTest
Output from these tests are in: /var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999_build/Testing/Temporary/LastTest.log
```

The "lasttest.log" it references is 619 Megabytes, I have compressed it to 83MB and it can be downloaded from https://pentoo.org/~zero/LastTest.log.xz

ZeroChaos-, 2022-03-20:
My configure/cmake step looks like this to get to this set of failures
```
 * Fetching https://github.com/srsran/srsRAN.git ...
git fetch https://github.com/srsran/srsRAN.git +HEAD:refs/git-r3/HEAD
git symbolic-ref refs/git-r3/net-wireless/srsran/0/__main__ refs/git-r3/HEAD
 * Checking out https://github.com/srsran/srsRAN.git to /var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999 ...
git checkout --quiet refs/git-r3/HEAD
GIT update -->
   repository:               https://github.com/srsran/srsRAN.git
   at the commit:            5275f33360f1b3f1ee8d1c4d9ae951ac7c4ecd4e
>>> Source unpacked in /var/tmp/portage/net-wireless/srsran-9999/work
>>> Preparing source in /var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999 ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999_build"
 * Hardcoded definition(s) removed in CMakeLists.txt:
 *     set(CMAKE_BUILD_TYPE Release)
 *  set(CMAKE_BUILD_TYPE ${CMAKE_BUILD_TYPE} CACHE STRING "")
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999 ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999_build"
cmake -C /var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_UHD=no -DENABLE_BLADERF=no -DENABLE_SOAPYSDR=no -DENABLE_ZEROMQ=no -DENABLE_HARDSIM=no -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999_build/gentoo_toolchain.cmake /var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999
loading initial cache file /var/tmp/portage/net-wireless/srsran-9999/work/srsran-9999_build/gentoo_common_config.cmake
CMake Deprecation Warning at CMakeLists.txt:33 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 11.2.1
-- The CXX compiler identification is GNU 11.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-5.16.4-pentoo
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/x86_64-pc-linux-gnu-g++
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/x86_64-pc-linux-gnu-pkg-config (found version "1.8.0") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.10
-- FFTW3F LIBRARIES: /usr/lib64/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: FFTW3F_STATIC_LIBRARY-NOTFOUND
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found FFTW3F: /usr/lib64/libfftw3f.so  
-- FFT_LIBRARIES: /usr/lib64/libfftw3f.so
-- Checking for module 'polarssl'
--   Package 'polarssl', required by 'virtual:world', not found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
[... 74 lines cut ...]
-- No post-build command defined
-- srsUE and srsENB builds disabled due to missing RF driver
-- Building with srsEPC
-- Found LibConfig++: /usr/lib64/libconfig++.so
-- static LibConfig++ path: LIBCONFIGPP_STATIC_LIBRARY-NOTFOUND
-- Found LibConfig: /usr/lib64/libconfig.so
-- static LibConfig path: LIBCONFIG_STATIC_LIBRARY-NOTFOUND
-- Checking for module 'sctp'
--   Package 'sctp', required by 'virtual:world', not found
-- SCTP LIBRARIES: /usr/lib64/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- <<< Gentoo configuration >>>
Build type      RelWithDebInfo
Install path    /usr
Compiler flags:
C               -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wno-unused-but-set-variable -march=native -Werror=incompatible-pointer-types -Wall -Wno-comment -Wno-write-strings -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -fno-strict-aliasing -D_GNU_SOURCE -ggdb -DBUILD_TYPE_RELWITHDEBINFO -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -mfma -DLV_HAVE_FMA -Ofast -funroll-loops -fvisibility=hidden -fdiagnostics-color=always --param large-function-growth=1600
C++             -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++11 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -ggdb -DBUILD_TYPE_RELWITHDEBINFO
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Module          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Shared          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0

-- Configuring done
```

ZeroChaos-, 2022-05-06:
22.04 results still have 114 failures, the numbers seem to have changed but the names seem to match up (for all the ones I checked manually)
```
The following tests FAILED:
	289 - pdsch_test_cdd_6 (Failed)
	290 - pdsch_test_cdd_12 (Failed)
	291 - pdsch_test_cdd_25 (Failed)
	292 - pdsch_test_cdd_50 (Failed)
	293 - pdsch_test_cdd_75 (Failed)
	294 - pdsch_test_cdd_100 (Failed)
	917 - phy_dl_test-p6-t3-E0-m0 (Failed)
	918 - phy_dl_test-p6-t3-E0-m7 (Failed)
	919 - phy_dl_test-p6-t3-E0-m14 (Failed)
	920 - phy_dl_test-p6-t3-E0-m21 (Failed)
	921 - phy_dl_test-p6-t3-E0-m28 (Failed)
	937 - phy_dl_test-p6-t3-E0-q-m0 (Failed)
	938 - phy_dl_test-p6-t3-E0-q-m7 (Failed)
	939 - phy_dl_test-p6-t3-E0-q-m14 (Failed)
	940 - phy_dl_test-p6-t3-E0-q-m21 (Failed)
	941 - phy_dl_test-p6-t3-E0-q-m27 (Failed)
	957 - phy_dl_test-p15-t3-E0-m0 (Failed)
	958 - phy_dl_test-p15-t3-E0-m7 (Failed)
	959 - phy_dl_test-p15-t3-E0-m14 (Failed)
	960 - phy_dl_test-p15-t3-E0-m21 (Failed)
	961 - phy_dl_test-p15-t3-E0-m28 (Failed)
	977 - phy_dl_test-p15-t3-E0-q-m0 (Failed)
	978 - phy_dl_test-p15-t3-E0-q-m7 (Failed)
	979 - phy_dl_test-p15-t3-E0-q-m14 (Failed)
	980 - phy_dl_test-p15-t3-E0-q-m21 (Failed)
	981 - phy_dl_test-p15-t3-E0-q-m26 (Failed)
	997 - phy_dl_test-p25-t3-E0-m0 (Failed)
	998 - phy_dl_test-p25-t3-E0-m7 (Failed)
	999 - phy_dl_test-p25-t3-E0-m14 (Failed)
	1000 - phy_dl_test-p25-t3-E0-m21 (Failed)
	1001 - phy_dl_test-p25-t3-E0-m28 (Failed)
	1017 - phy_dl_test-p25-t3-E0-q-m0 (Failed)
	1018 - phy_dl_test-p25-t3-E0-q-m7 (Failed)
	1019 - phy_dl_test-p25-t3-E0-q-m14 (Failed)
	1020 - phy_dl_test-p25-t3-E0-q-m21 (Failed)
	1021 - phy_dl_test-p25-t3-E0-q-m27 (Failed)
	1037 - phy_dl_test-p50-t3-E0-m0 (Failed)
	1038 - phy_dl_test-p50-t3-E0-m7 (Failed)
	1039 - phy_dl_test-p50-t3-E0-m14 (Failed)
	1040 - phy_dl_test-p50-t3-E0-m21 (Failed)
	1041 - phy_dl_test-p50-t3-E0-m28 (Failed)
	1057 - phy_dl_test-p50-t3-E0-q-m0 (Failed)
	1058 - phy_dl_test-p50-t3-E0-q-m7 (Failed)
	1059 - phy_dl_test-p50-t3-E0-q-m14 (Failed)
	1060 - phy_dl_test-p50-t3-E0-q-m21 (Failed)
	1061 - phy_dl_test-p50-t3-E0-q-m27 (Failed)
	1077 - phy_dl_test-p75-t3-E0-m0 (Failed)
	1078 - phy_dl_test-p75-t3-E0-m7 (Failed)
	1079 - phy_dl_test-p75-t3-E0-m14 (Failed)
	1080 - phy_dl_test-p75-t3-E0-m21 (Failed)
	1081 - phy_dl_test-p75-t3-E0-m28 (Failed)
	1097 - phy_dl_test-p75-t3-E0-q-m0 (Failed)
	1098 - phy_dl_test-p75-t3-E0-q-m7 (Failed)
	1099 - phy_dl_test-p75-t3-E0-q-m14 (Failed)
	1100 - phy_dl_test-p75-t3-E0-q-m21 (Failed)
	1101 - phy_dl_test-p75-t3-E0-q-m27 (Failed)
	1117 - phy_dl_test-p100-t3-E0-m0 (Failed)
	1118 - phy_dl_test-p100-t3-E0-m7 (Failed)
	1119 - phy_dl_test-p100-t3-E0-m14 (Failed)
	1120 - phy_dl_test-p100-t3-E0-m21 (Failed)
	1121 - phy_dl_test-p100-t3-E0-m28 (Failed)
	1137 - phy_dl_test-p100-t3-E0-q-m0 (Failed)
	1138 - phy_dl_test-p100-t3-E0-q-m7 (Failed)
	1139 - phy_dl_test-p100-t3-E0-q-m14 (Failed)
	1140 - phy_dl_test-p100-t3-E0-q-m21 (Failed)
	1141 - phy_dl_test-p100-t3-E0-q-m27 (Failed)
	1155 - phy_dl_test-p6-t3-E1-m0 (Failed)
	1156 - phy_dl_test-p6-t3-E1-m7 (Failed)
	1157 - phy_dl_test-p6-t3-E1-m14 (Failed)
	1158 - phy_dl_test-p6-t3-E1-m21 (Failed)
	1171 - phy_dl_test-p6-t3-E1-q-m0 (Failed)
	1172 - phy_dl_test-p6-t3-E1-q-m7 (Failed)
	1173 - phy_dl_test-p6-t3-E1-q-m14 (Failed)
	1174 - phy_dl_test-p6-t3-E1-q-m21 (Failed)
	1187 - phy_dl_test-p15-t3-E1-m0 (Failed)
	1188 - phy_dl_test-p15-t3-E1-m7 (Failed)
	1189 - phy_dl_test-p15-t3-E1-m14 (Failed)
	1190 - phy_dl_test-p15-t3-E1-m21 (Failed)
	1203 - phy_dl_test-p15-t3-E1-q-m0 (Failed)
	1204 - phy_dl_test-p15-t3-E1-q-m7 (Failed)
	1205 - phy_dl_test-p15-t3-E1-q-m14 (Failed)
	1206 - phy_dl_test-p15-t3-E1-q-m21 (Failed)
	1219 - phy_dl_test-p25-t3-E1-m0 (Failed)
	1220 - phy_dl_test-p25-t3-E1-m7 (Failed)
	1221 - phy_dl_test-p25-t3-E1-m14 (Failed)
	1222 - phy_dl_test-p25-t3-E1-m21 (Failed)
	1235 - phy_dl_test-p25-t3-E1-q-m0 (Failed)
	1236 - phy_dl_test-p25-t3-E1-q-m7 (Failed)
	1237 - phy_dl_test-p25-t3-E1-q-m14 (Failed)
	1238 - phy_dl_test-p25-t3-E1-q-m21 (Failed)
	1251 - phy_dl_test-p50-t3-E1-m0 (Failed)
	1252 - phy_dl_test-p50-t3-E1-m7 (Failed)
	1253 - phy_dl_test-p50-t3-E1-m14 (Failed)
	1254 - phy_dl_test-p50-t3-E1-m21 (Failed)
	1267 - phy_dl_test-p50-t3-E1-q-m0 (Failed)
	1268 - phy_dl_test-p50-t3-E1-q-m7 (Failed)
	1269 - phy_dl_test-p50-t3-E1-q-m14 (Failed)
	1270 - phy_dl_test-p50-t3-E1-q-m21 (Failed)
	1283 - phy_dl_test-p75-t3-E1-m0 (Failed)
	1284 - phy_dl_test-p75-t3-E1-m7 (Failed)
	1285 - phy_dl_test-p75-t3-E1-m14 (Failed)
	1286 - phy_dl_test-p75-t3-E1-m21 (Failed)
	1299 - phy_dl_test-p75-t3-E1-q-m0 (Failed)
	1300 - phy_dl_test-p75-t3-E1-q-m7 (Failed)
	1301 - phy_dl_test-p75-t3-E1-q-m14 (Failed)
	1302 - phy_dl_test-p75-t3-E1-q-m21 (Failed)
	1315 - phy_dl_test-p100-t3-E1-m0 (Failed)
	1316 - phy_dl_test-p100-t3-E1-m7 (Failed)
	1317 - phy_dl_test-p100-t3-E1-m14 (Failed)
	1318 - phy_dl_test-p100-t3-E1-m21 (Failed)
	1331 - phy_dl_test-p100-t3-E1-q-m0 (Failed)
	1332 - phy_dl_test-p100-t3-E1-q-m7 (Failed)
	1333 - phy_dl_test-p100-t3-E1-q-m14 (Failed)
	1334 - phy_dl_test-p100-t3-E1-q-m21 (Failed)
```

andrepuschmann, 2022-05-06:
We don't have a Ubuntu 22.04 in our automated CI yet but we've run manual build tests on it and it was compiling/testing fine. Also latest archlinux was fine. What CPU are you using? What does `cat /proc/cpuinfo` say?

robertfalkenberg, 2022-05-18:
Hi,
unfortunately, I cannot reproduce the reported errors.

To rule out that the issue comes from the packaging, could you please try to build+test without it, i.e. manual build?

```sh
mkdir build
cd build
cmake <path-to-local-srsRAN-repo>
make phy_dl_test -j`nproc`
ctest -j`nproc` -R phy_dl_test
```

ZeroChaos-, 2023-02-02:
Sorry I dropped the ball on troubleshooting this.  Picking it up now from current head 254cc719a9a31f64ce0262f4ca6ab72b1803477d
Current failures
```
88% tests passed, 181 tests failed out of 1546

Label Time Summary:
chest     =  37.48 sec*proc (22 tests)
fec       =   1.09 sec*proc (11 tests)
ldpc      =   0.02 sec*proc (1 test)
lib       = 954.79 sec*proc (1007 tests)
lte       = 963.50 sec*proc (942 tests)
nr        = 408.96 sec*proc (359 tests)
pdcp      =   0.80 sec*proc (3 tests)
phch      = 453.01 sec*proc (485 tests)
phy       = 992.04 sec*proc (1004 tests)
polar     =   0.01 sec*proc (1 test)
rlc       =  34.72 sec*proc (13 tests)
sched     =   0.38 sec*proc (7 tests)
srsenb    =  48.66 sec*proc (10 tests)
srsue     =  24.12 sec*proc (3 tests)
turbo     =   1.06 sec*proc (9 tests)

Total Test time (real) = 245.21 sec

The following tests FAILED:
	289 - pdsch_test_cdd_6 (Failed)
	290 - pdsch_test_cdd_12 (Failed)
	291 - pdsch_test_cdd_25 (Failed)
	292 - pdsch_test_cdd_50 (Failed)
	293 - pdsch_test_cdd_75 (Failed)
	294 - pdsch_test_cdd_100 (Failed)
	884 - srsran_asn1_rrc_mcch_test (Subprocess aborted)
	885 - srsran_asn1_rrc_meas_test (Subprocess aborted)
	920 - phy_dl_test-p6-t3-E0-m0 (Failed)
	921 - phy_dl_test-p6-t3-E0-m7 (Failed)
	922 - phy_dl_test-p6-t3-E0-m14 (Failed)
	923 - phy_dl_test-p6-t3-E0-m21 (Failed)
	924 - phy_dl_test-p6-t3-E0-m28 (Failed)
	940 - phy_dl_test-p6-t3-E0-q-m0 (Failed)
	941 - phy_dl_test-p6-t3-E0-q-m7 (Failed)
	942 - phy_dl_test-p6-t3-E0-q-m14 (Failed)
	943 - phy_dl_test-p6-t3-E0-q-m21 (Failed)
	944 - phy_dl_test-p6-t3-E0-q-m27 (Failed)
	960 - phy_dl_test-p15-t3-E0-m0 (Failed)
	961 - phy_dl_test-p15-t3-E0-m7 (Failed)
	962 - phy_dl_test-p15-t3-E0-m14 (Failed)
	963 - phy_dl_test-p15-t3-E0-m21 (Failed)
	964 - phy_dl_test-p15-t3-E0-m28 (Failed)
	980 - phy_dl_test-p15-t3-E0-q-m0 (Failed)
	981 - phy_dl_test-p15-t3-E0-q-m7 (Failed)
	982 - phy_dl_test-p15-t3-E0-q-m14 (Failed)
	983 - phy_dl_test-p15-t3-E0-q-m21 (Failed)
	984 - phy_dl_test-p15-t3-E0-q-m26 (Failed)
	1000 - phy_dl_test-p25-t3-E0-m0 (Failed)
	1001 - phy_dl_test-p25-t3-E0-m7 (Failed)
	1002 - phy_dl_test-p25-t3-E0-m14 (Failed)
	1003 - phy_dl_test-p25-t3-E0-m21 (Failed)
	1004 - phy_dl_test-p25-t3-E0-m28 (Failed)
	1020 - phy_dl_test-p25-t3-E0-q-m0 (Failed)
	1021 - phy_dl_test-p25-t3-E0-q-m7 (Failed)
	1022 - phy_dl_test-p25-t3-E0-q-m14 (Failed)
	1023 - phy_dl_test-p25-t3-E0-q-m21 (Failed)
	1024 - phy_dl_test-p25-t3-E0-q-m27 (Failed)
	1040 - phy_dl_test-p50-t3-E0-m0 (Failed)
	1041 - phy_dl_test-p50-t3-E0-m7 (Failed)
	1042 - phy_dl_test-p50-t3-E0-m14 (Failed)
	1043 - phy_dl_test-p50-t3-E0-m21 (Failed)
	1044 - phy_dl_test-p50-t3-E0-m28 (Failed)
	1060 - phy_dl_test-p50-t3-E0-q-m0 (Failed)
	1061 - phy_dl_test-p50-t3-E0-q-m7 (Failed)
[... 112 lines cut ...]
	1523 - nr_phy_test_20MHz_dummymac_6D+4U_dl_default (Subprocess aborted)
	1524 - nr_phy_test_20MHz_dummymac_6D+4U_dl_ts38101/5.2-1 (Subprocess aborted)
	1525 - nr_phy_test_20MHz_dummymac_6D+4U_ul_only (Subprocess aborted)
	1526 - nr_phy_test_20MHz_dummymac_6D+4U_bidir (Subprocess aborted)
	1527 - nr_phy_test_20MHz_dummymac_FR1.15-1_dl_default (Subprocess aborted)
	1528 - nr_phy_test_20MHz_dummymac_FR1.15-1_dl_ts38101/5.2-1 (Subprocess aborted)
	1529 - nr_phy_test_20MHz_dummymac_FR1.15-1_ul_only (Subprocess aborted)
	1530 - nr_phy_test_20MHz_dummymac_FR1.15-1_bidir (Subprocess aborted)
	1531 - nr_phy_test_20MHz_realmac_FDD_dl_default (Subprocess aborted)
	1532 - nr_phy_test_20MHz_realmac_FDD_dl_ts38101/5.2-1 (Subprocess aborted)
	1533 - nr_phy_test_20MHz_realmac_FDD_ul_only (Subprocess aborted)
	1534 - nr_phy_test_20MHz_realmac_FDD_bidir (Subprocess aborted)
	1535 - nr_phy_test_20MHz_realmac_6D+4U_dl_default (Subprocess aborted)
	1536 - nr_phy_test_20MHz_realmac_6D+4U_dl_ts38101/5.2-1 (Subprocess aborted)
	1537 - nr_phy_test_20MHz_realmac_6D+4U_ul_only (Subprocess aborted)
	1538 - nr_phy_test_20MHz_realmac_6D+4U_bidir (Subprocess aborted)
	1539 - nr_phy_test_20MHz_realmac_FR1.15-1_dl_default (Subprocess aborted)
	1540 - nr_phy_test_20MHz_realmac_FR1.15-1_dl_ts38101/5.2-1 (Subprocess aborted)
	1541 - nr_phy_test_20MHz_realmac_FR1.15-1_ul_only (Subprocess aborted)
	1542 - nr_phy_test_20MHz_realmac_FR1.15-1_bidir (Subprocess aborted)
	1543 - nr_phy_test_20MHz_prach_fdd (Subprocess aborted)
	1544 - nr_phy_test_20MHz_prach_tdd (Subprocess aborted)
	1545 - nr_phy_test_20MHz_sr (Subprocess aborted)
	1546 - nr_phy_test_20MHz_sr_harq (Subprocess aborted)
```

ZeroChaos-, 2023-02-02:
Building and running just the tests you suggested yields positive results
```
100% tests passed, 0 tests failed out of 432

Label Time Summary:
lib    = 390.67 sec*proc (432 tests)
lte    = 390.67 sec*proc (432 tests)
phy    = 390.67 sec*proc (432 tests)

Total Test time (real) =  33.11 sec
```
So something in my build setup is breaking it.  I'll test some obvious cmake flags and narrow it down.

ZeroChaos-, 2023-02-02:
> not sure if it helps but this is srsran-9999 build log on my side: https://gist.githubusercontent.com/richardredditch/6ea266365b049ee90cb019858d0c725f/raw/eada560a8af3b11a5e46dfcaf7c52bff050e4765/srsrun-9999%2520gentoo I will try to test this build with hardware setup in my lab and report back.

This bug is about testing, and I've got this.  I'll let you know if I give up but I'm working on a min test case now.  I can replicate the failures outside of portage already.

ZeroChaos-, 2023-02-02:
Okay, I've narrowed it down to why it failed and have a min test case.  It's caused by one of Gentoo's overrides which are typically there to prevent injection of undesired/unrequested cflags or similar.

```
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="" ..
make phy_dl_test -j`nproc`
ctest -j`nproc` -R phy_dl_test
```

The obvious difference is that CMakeCache.txt changes
```
-CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
+CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=
```

First I tested by changing my system cflags to `-O2 -g` and that didn't change anything, so I assume that `-DNDEBUG` needs to be injected somewhere, but adding it directly to main cflags doesn't make sense (or work).

I can't easily override this setting because it's not applied on the command line it is applied via a config file.  Any thoughts on how I might solve this?  I already tried overriding on the cmake command line with `-DCMAKE_C_FLAGS_RELWITHDEBINFO="-DNDEBUG"` but that didn't help either.

ZeroChaos-, 2023-02-02:
I super can't explain this, but here are some example test cases that WORK (tests pass):
```
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-O0" ..
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-O" ..
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-O1" ..
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-Os" ..
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-O2" ..
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-O3" ..
```
Compared to the examples which do not work (build works, tests fail)
```
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="" ..
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-pipe" ..
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS_RELWITHDEBINFO="-DI_DO_NOTHING" ..
```