srsran/srsRAN_4G #836: srsUE fails to attach to commercial base station with USRP-B210

advait-voltela, 2022-03-21:
## Issue Description ##
srsUE cannot connect to the commercial base station. 

## Setup Details ##

Ubuntu 20.04
USRP-B210 mini
Card reader and T-mobile sim card
running ./srsue/src/srsue --usim.mode=pcsc

[ueconf.txt](https://github.com/srsran/srsRAN/files/8317045/ueconf.txt)

## Expected Behavior ##
Attach to network and assigned an IP.

## Actual Behaviour ##
Fails to attach to the network 5/5 times. 
[uelogfiles](https://drive.google.com/file/d/1zLShEQ2ZlIo6Wo2wv9vLr2wxJzEA-ED4/view?usp=sharing)

andrepuschmann, 2022-03-21:
Hey, seems that the Random Access procedure is never completing because the UL signal is probably to weak and the eNB never receives Msg3. Note that you have a very low DL SNR of around ~6dB. The UL signal will be sent out at much lower power with the USRP than what the eNB uses for the DL. Long story short, from your location it will be difficult to connect to this eNB without proper PA. So move closer to the base station.
Next time you can upload logs in info mode first to keep the file size smaller.

Closing the issue, please reopen if needed.

closed 2022-03-21 by andrepuschmann

advait-voltela, 2022-03-23:
Hello @andrepuschmann, can you recommend a good PA for the USRP B210 mini to overcome this issue?