srsran/srsRAN_4G #838: pssch_ue example exits immediately with floating point exception

twardokus, 2022-03-23:
## Issue Description ##
Executing the PSSCH example application with `./pssch_ue -a clock=gpsdo` with UHD crashes immediately after setting the subframe length with a floating point exception.

## Setup Details ##
This is a fresh installation of srsRAN on Ubuntu 20.04.03 LTS. Issue occurs on two separate machines, one Dell and one Lenovo laptop, so not likely a hardware issue. I am using UHD 3.15.0 installed from the Ubuntu package repository (as per the srsRAN setup instructions) and have installed and built srsRAN using the official installation steps to build from source. My SDRs are USRP B210s with TXCO GPSDO modules installed. 

## Expected Behavior ##
Running the command `./pssch_ue -a clock=gpsdo` from within the `srsRAN\build\lib\examples` directory should launch the C-V2X receiver example and begin receiving samples.

## Actual Behaviour ##
The program crashes with a floating point exception. The full output is:
```
Opening RF device...
Available RF device list: UHD  zmq 
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO, Firmware Rev 0.929b
[WARNING] [GPS] update_cache: Malformed GPSDO string: EEPROM Write Failed!
[WARNING] [GPS] update_cache: Malformed GPSDO string: EEPROM Write Failed!
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting USRP time to 1623544208.000000s
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
RF device 'UHD' successfully opened
Set RX freq: 5920.000000 MHz
Set RX gain: 50.0 dB
Setting sampling rate 11.52 MHz
Using a SF len of 11520 samples
Floating point exception (core dumped)
```

## Steps to reproduce the problem ##
No special configuration. I installed the OS, then followed the srsRAN documentation to install dependencies and finally srsRAN. The first application I ran after completing the installation process was the above command to execute the `pssch_ue` application.

twardokus, 2022-03-27:
I ran the `cell_search`, `usrp_capture` and `usrp_capture_sync` examples using the exact configuration described above. All of them worked fine on both of my laptops. It seems to be a specific issue with the `pssch_ue` example.

Kurtissense, 2022-04-04:
I have also been encountering the same issue. My setup follows the instructions here https://www.rit.edu/wisplab/sites/rit.edu.wisplab/files/2021-08/Lab%204%20-%20C-V2X.pdf. let me know if there

scrappylabs, 2022-04-08:
Same issue here. I'm about to just start debugging it, I'll report any findings...

andrepuschmann, 2022-04-14:
Hey, I could reproduce the issue and it will be properly fixed in the upcoming release. For the time being apply the following patch and the example should work as expected.

```
diff --git a/lib/examples/pssch_ue.c b/lib/examples/pssch_ue.c
index d45ccb428..1c7334009 100644
--- a/lib/examples/pssch_ue.c
+++ b/lib/examples/pssch_ue.c
@@ -303,7 +303,7 @@ int main(int argc, char** argv)
   cf_t*    equalized_sf_buffer = srsran_vec_malloc(sizeof(cf_t) * sf_n_re);

   // RX
-  srsran_ofdm_t     fft[SRSRAN_MAX_PORTS];
+  srsran_ofdm_t     fft[SRSRAN_MAX_PORTS] = {};
   srsran_ofdm_cfg_t ofdm_cfg = {};
   ofdm_cfg.nof_prb           = cell_sl.nof_prb;
   ofdm_cfg.cp                = SRSRAN_CP_NORM;
diff --git a/lib/src/phy/ue/ue_sync.c b/lib/src/phy/ue/ue_sync.c
index e255e9e40..a47020d25 100644
--- a/lib/src/phy/ue/ue_sync.c
+++ b/lib/src/phy/ue/ue_sync.c
@@ -339,8 +339,8 @@ int srsran_ue_sync_set_cell(srsran_ue_sync_t* q, srsran_cell_t cell)
     q->cell     = cell;
     q->fft_size = srsran_symbol_sz(q->cell.nof_prb);
     q->sf_len   = SRSRAN_SF_LEN(q->fft_size);
-    srsran_sync_set_cp(&q->sfind, q->cell.cp);
-    srsran_sync_set_cp(&q->strack, q->cell.cp);
+    //srsran_sync_set_cp(&q->sfind, q->cell.cp);
+    //srsran_sync_set_cp(&q->strack, q->cell.cp);
     if (cell.id == 1000) {
       /* If the cell is unkown, we search PSS/SSS in 5 ms */
       q->nof_recv_sf = 5;
```

Thanks
Andre

Kurtissense, 2022-04-14:
I just installed it and the GUI comes up and seems to do what I expect. Thanks for that.

andrepuschmann, 2022-04-26:
Thanks for confirming. Closing the issue. Fix will be in next release.

closed 2022-04-26 by andrepuschmann