srsran/srsRAN_4G #839: A lot of RACH:  tti=8111, cc=0, preamble=20, offset=9, temp_crnti=0x58 Disconnecting rnti=0x58. messages after a commercial UE is connected.

fllay, 2022-04-02:
## Issue Description ##
[Describe the issue in detail]
When a COTS UE is connected to the eNB, there are a lot of messages of 
```
RACH:  tti=8111, cc=0, preamble=20, offset=9, temp_crnti=0x58
Disconnecting rnti=0x58.
```
For example.
```
RACH:  tti=8901, cc=0, preamble=10, offset=9, temp_crnti=0x54
User 0x54 requesting RRC Reestablishment as 0x46. Cause: otherFailure
Disconnecting rnti=0x46.
User 0x54 connected
lte   54    6   1    5   215k   28   14  33% |   5.5    1.0   36    1    55k   12    0   0%    0.0
lte   54    9   1   18   1.1M   77    7   8% |   5.4   -0.7   36    2   232k  122    0   0%    0.0
lte   54   15   1    8    21k   23   59  71% |   4.1   -1.1   36    1    25k   16    0   0%    0.0
lte   54   15   1   12     56    1   47  97% |   4.0    2.1   36    1   4.9k    3    0   0%    0.0
RACH:  tti=2431, cc=0, preamble=32, offset=9, temp_crnti=0x55
Disconnecting rnti=0x55.

          -----------------DL----------------|-------------------------UL-------------------------
rat rnti  cqi  ri  mcs  brate   ok  nok  (%) | pusch  pucch  phr  mcs  brate   ok  nok  (%)    bsr
lte   54  n/a   1   10      0    0   28 100% |   n/a   -2.1    0    0      0    0    0   0%    0.0
lte   54    7   1    0      0    0    3 100% |   n/a   -0.8    0    0      0    0    0   0%    0.0
RACH:  tti=4091, cc=0, preamble=36, offset=9, temp_crnti=0x56
Disconnecting rnti=0x56.
User 0x54 connected
RACH:  tti=4161, cc=0, preamble=46, offset=9, temp_crnti=0x57
Disconnecting rnti=0x57.
User 0x54 connected
lte   54   15   1   14   1.7k    3  104  97% |   6.9   -3.1   36    4   111k   35    2   5%    0.0
lte   54   15   1    0      0    0    8 100% |   5.5   -2.4   39    0   3.8k    3    0   0%    0.0
Disconnecting rnti=0x54.
RACH:  tti=8111, cc=0, preamble=20, offset=9, temp_crnti=0x58
Disconnecting rnti=0x58.
RACH:  tti=8191, cc=0, preamble=23, offset=9, temp_crnti=0x59
Disconnecting rnti=0x59.
RACH:  tti=8271, cc=0, preamble=27, offset=9, temp_crnti=0x5a
Disconnecting rnti=0x5a.
RACH:  tti=8351, cc=0, preamble=7, offset=9, temp_crnti=0x5b
Disconnecting rnti=0x5b.
RACH:  tti=8431, cc=0, preamble=15, offset=9, temp_crnti=0x5c
Disconnecting rnti=0x5c.
RACH:  tti=8511, cc=0, preamble=20, offset=9, temp_crnti=0x5d
Disconnecting rnti=0x5d.
RACH:  tti=8591, cc=0, preamble=20, offset=9, temp_crnti=0x5e
Disconnecting rnti=0x5e.
RACH:  tti=8671, cc=0, preamble=1, offset=9, temp_crnti=0x5f
Disconnecting rnti=0x5f.
RACH:  tti=8751, cc=0, preamble=49, offset=9, temp_crnti=0x60
Disconnecting rnti=0x60.
RACH:  tti=8831, cc=0, preamble=22, offset=9, temp_crnti=0x61
Disconnecting rnti=0x61.
RACH:  tti=9021, cc=0, preamble=14, offset=9, temp_crnti=0x62
RRCReestablishmentReject for rnti=0x62. Cause: no context available
lte   62  n/a   0    0   3.5k    1    0   0% |  18.8    n/a    0    0   2.2k    1    0   0%    0.0
```

The content of `enb.log` is 

```
2022-04-02T01:46:03.265841 [ENB    ] [I] Using binary srsenb with arguments: 
2022-04-02T01:46:03.732661 [ENB    ] [I] Built in Release mode using 21.10.0.
2022-04-02T01:46:03.810967 [COMN   ] [D] [    0] Setting RTO_INFO options on SCTP socket. Association 0, Initial RTO 3000, Minimum RTO 1000, Maximum RTO 6000
2022-04-02T01:46:03.810970 [COMN   ] [D] [    0] Setting SCTP_INITMSG options on SCTP socket. Max attempts 3, Max init attempts timeout 5000
2022-04-02T01:46:03.810985 [COMN   ] [D] [    0] Successfully bound to address 127.0.1.1:0
2022-04-02T01:46:03.811135 [COMN   ] [D] [    0] RxSockets: socket fd=6 has been registered.
2022-04-02T01:46:03.811227 [COMN   ] [D] [    0] Successfully bound to address 127.0.1.1:2152
2022-04-02T01:46:03.811233 [COMN   ] [D] [    0] RxSockets: socket fd=7 has been registered.
2022-04-02T01:47:55.115039 [STCK   ] [E] Bearers: No EPS bearer registered for rnti=0x47
[... 93 lines cut ...]
RACH:  tti=8271, cc=0, preamble=27, offset=9, temp_crnti=0x5a
Disconnecting rnti=0x5a.
RACH:  tti=8351, cc=0, preamble=7, offset=9, temp_crnti=0x5b
Disconnecting rnti=0x5b.
RACH:  tti=8431, cc=0, preamble=15, offset=9, temp_crnti=0x5c
Disconnecting rnti=0x5c.
RACH:  tti=8511, cc=0, preamble=20, offset=9, temp_crnti=0x5d
Disconnecting rnti=0x5d.
RACH:  tti=8591, cc=0, preamble=20, offset=9, temp_crnti=0x5e
Disconnecting rnti=0x5e.
RACH:  tti=8671, cc=0, preamble=1, offset=9, temp_crnti=0x5f
Disconnecting rnti=0x5f.
RACH:  tti=8751, cc=0, preamble=49, offset=9, temp_crnti=0x60
Disconnecting rnti=0x60.
RACH:  tti=8831, cc=0, preamble=22, offset=9, temp_crnti=0x61
Disconnecting rnti=0x61.
RACH:  tti=9021, cc=0, preamble=14, offset=9, temp_crnti=0x62
RRCReestablishmentReject for rnti=0x62. Cause: no context available
```

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2022-04-26:
Those RACH attempts usually happen if the phone loses sync to the eNB and tries to Reestablish. There seems to be an issue with the DL signal, either it's too weak or you have lates on the radio and the IQ stream has gaps that the phone interprets as sync losses. Try to improve the signal conditions.
I am closing the issue, please feel free to reopen if you see need.

fllay, 2022-04-29:
Thank you. It seems that DL power amplifier is a problem. I used a generic one and it seems that it drives the PA into non-linear region of the PA. I have reduced the TX power from SDR (BladeRF). It is a lot better now. I think I do need a signal analyzer.

andrepuschmann, 2022-04-29:
Thanks for confirmation.