srsran/srsRAN_4G #841: High srsenb traffic without IP traffic

avalori1, 2022-04-08:
## Issue Description ##

**SEE ATTACHED VIDEO!!**

192.168.18.22 is the enode host
192.168.18.122 is the UE (Ubuntu with USB modem)  
1.	The video has been started in a condition of “unexpected traffic”. The srsenb trace shows a single UE producing traffic at around 10Mbit/s, which is the maximum expected traffic since we have RB=25. The traffic monitor tools (nmon) on both the enode and the UE show virtually no traffic 
2.	Around time 00:10: sending the command at+cfun=0 to the modem, [from the modem instruction manual “sets the MT to minimum functionality (disable both transmit and receive RF circuits by deactivating both CS and PS services)”]. The UE disconnects from the enode (as expected) and the traffic previously shown on the enb trace stops
3.	Around 00:15: sending the command at+cfun=1 to the modem, [description from manual “sets the MT to full functionality, e.g. from airplane mode or minimum functionality”]. The condition goes back to the initial one, with the nmon tools not showing traffic but the srsenb trace showing close to 10Mbit/s
4.	Around 00:22: starting iperf client on the UE host, the traffic on the nmon tools shows something in the order of 1.5MB/s in the enode and out of the UE, the srsenb trace shows traffic at about 10Mbit/s (things “sort of match” converting bytes in Bits). When iperf stops, the traffic shown on nmon goes back to almost zero but the one shown on the srsenb trace not, actually going to a fixed 10MBits/s.
NOTES:
1.	We checked if there is RF being transmitted by the UE during the “unexpected traffic” and confirmed that indeed the RF is ON
2.	We tested the system with completely different UEs (some android phones) and have seen the same effect, therefore we tend to exclude the UEs as culprits
3.	We tested also with iperf run in UDP mode and using iperf2
4.	During acquisition there is a process (started by us) to check which UEs are connected, and this uses fping to check which UEs reply. It seems that stopping this process, at least sometime, the unexpected traffic stops (but only after several seconds). HOWEVER we saw the unexpected traffic restarting after a while even without the fping process in the background
Conclusion:
It seems that sometime the system ends in a situation where even little traffic (as much as a repeated ping) results in maximum bandwidth being used at a level much lower than IP. If the IP traffic stops completely for “enough time” the eNodeB disconnects the UE, but the bandwidth starts to be re-occupied completely if the IP traffic restarts (even very little traffic). 

## Setup Details ##
The phenomenon is difficult to reproduce reliably (unfortunately) but happened with different setups. In particular:
Different CPUs (i3 7th generation, i7 11th generation)
Different boards (B200 and LimeSDR mini)
Different UEs (UBlox USB modules, Android phones of different vendors and OS) 
Different OS (Ubuntu and Yocto)

## Expected Behavior ##
When the IP traffic stops (iperf ends) I expect the traffic to drop to zero (or at most some very low "background" traffic due to the different UE-eNodeB messages)

## Actual Behaviour ##
The traffic stays very high (even higher than during iperf, at or very close to the max possible throughput) with extremely low BLER for many seconds (if the network is not stimulated at all) and forever if there is a bit of IP traffic (just a ping is enough to keep it at max throughput). 

https://user-images.githubusercontent.com/73334180/162374303-6ae87088-f500-4671-8c87-000098bf3952.mp4

andrepuschmann, 2022-04-14:
The UE signals non-zero buffers and the eNB keeps allocating UL grants. That's how it works. I don't see an issue with that. I any case, check the logs of the eNB and look for the BSRs received. I am closing the issue, please reopen if needed

closed 2022-04-14 by andrepuschmann

avalori1, 2022-04-29:
@andrepuschmann,

we have been puzzled for a while after seeing this phenomena. As you mentioned, seems that the issue originates from the the module, transmitting a high BSR, and the eNodeB reacting to that by (correctly) assigning resources.
However, the module is a certified commercial module, and we were extremely surprised that such an issue would have passed certification.

To rule the module out, we did some tests with completely different UEs:

- laptops running Ubuntu with connected the uBlox Toby modules
- LG phones running Android

We saw the issue with both UEs.

Although I understand your point, it really seems to me that the issue is that the BSR gets "corrupted" somewhere after the UEs.

Is there a way to monitor the BSR as it passes through the transmission and decoding line? 

Cheers