srsran/srsRAN_4G #842: Synch/MAC Failure

NeilJohnR, 2022-04-12:
## Issue Description ##
Synch/MAC Failure errors upon camping UE's

## Setup Details ##
We're running an NextEPC v0.3.10--> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 21.10 ; UHD 3.15.0

## Expected Behavior ##
UE can camp automatically

## Actual Behaviour ##
- Realme
	- Samsung
Can't camp to the network (synch failure)
	- Vivo
	- Huawei

We've only using one EPC and all phones are provisioned, since the two UE's can camp automatically we expecting the other two UE's will camp also, but it can't. While observing the pcap traces, it says Synch failure or IMSI unknown in HSS. I just thinking because I provisioned all the sims in a same procedure then the two UE's can't camp.

I attached my pcap traces and config files of my enb.conf for your reference. If you have encountered this kind of issue we can work together to solve this. 

## Steps to reproduce the problem ##
Using Mongodb compass, by configuring in HSS, we are setting the SQN incrementing by 32. but it doesn't work.

## Additional Information ##
[Synch_Mac_Failure.zip](https://github.com/srsran/srsRAN/files/8469166/Synch_Mac_Failure.zip)

andrepuschmann, 2022-04-14:
Same as https://github.com/srsran/srsRAN/issues/843. Provide logs please.

NeilJohnR, 2022-04-18:
Again sorry for late response, I attached enb.log and mme.log of my testing as I run trough the process and I got the same error in logs. what should I do?

[ENB.LOG_SYNCH_FAILURE.txt](https://github.com/srsran/srsRAN/files/8504137/ENB.LOG_SYNCH_FAILURE.txt)
[MME.LOGS_SYNCH_FAILURE.txt](https://github.com/srsran/srsRAN/files/8504138/MME.LOGS_SYNCH_FAILURE.txt)

andrepuschmann, 2022-04-26:
IMHO it's the same like #843. The EPC is not responding to the UE request and is eventually removing the user again over S1AP. You need to find the cause for that in the EPC I believe. I am closing the issue, feel free to reopen if you see need.

closed 2022-04-26 by andrepuschmann

NeilJohnR, 2022-04-26:
only Huawei phone cannot camped to the network and the said logs was synch failure. I just tried swapping their sims from huawei phone to samsung phone and it successfully camp. I have 4 testing phones including Huawei phone, all can automatically camped except Huawei phone even though swapping their sims. Is there an issue regarding Huawei phones?