srsran/srsRAN_4G #843: RadioNetwork-cause=user-inactivity

NeilJohnR, 2022-04-12:
## Issue Description ##
RadioNetwork-cause=user-inactivity, UE's can't camp to the Network

## Setup Details ##
We're running an NextEPC v0.3.10--> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 21.10 ; UHD 3.15.0

## Expected Behavior ##
UE can camp automatically

## Actual Behaviour ##
I am using Vivo,Samsung,Realme UE phone brands and their sims are provisioned in our NextEPC for camp testing. But all can't camp to the network. I'm trying to swap their sims in case it will camp successful but it doesn't. Base on my pcap trace error causes RadioNetwork-cause=user-inactivity.

I attached my config file and pcap trace for your reference, incase you have idea or any suggestions to solve this issue please comment down. Thank you in advance.

## Additional Information ##
[RADIO_NETWORK_CAUSE_USER_INACTIVITY.zip](https://github.com/srsran/srsRAN/files/8469182/RADIO_NETWORK_CAUSE_USER_INACTIVITY.zip)

andrepuschmann, 2022-04-14:
Provide a full eNB log please.

NeilJohnR, 2022-04-18:
Sorry for late reply @andrepuschmann , We celebrating Holy Week in our country this past few days that's why I can't immediately response on you. By the way, I attached logs of my run through testing for phone camping. I still can't camp the UE's.

[ENB.LOGS_RADIO_NETWORK_CAUSE_INACTIVE.txt](https://github.com/srsran/srsRAN/files/8503709/ENB.LOGS_RADIO_NETWORK_CAUSE_INACTIVE.txt)
[MME.LOG_RADIO_NETWORK_CAUSE_INACTIVE.txt](https://github.com/srsran/srsRAN/files/8503747/MME.LOG_RADIO_NETWORK_CAUSE_INACTIVE.txt)

andrepuschmann, 2022-04-19:
The eNB receives the connection setup complete message and forwards this to the EPC. No response from the EPC from then on. The eNB behaves correctly and removes the user after some time. IMHO no issue here on the RAN side. Closing the issue.
```
2022-04-18T05:34:51.873709 [PHY1   ] [I] [ 7177] PUSCH: cc=0, rnti=0x4a, rb=(1,1), nof_re=144, tbs=26, mod=4, rv=0, crc=OK, avg_iter=2.0, snr=28.0 dB, epre=2.8 dBfs, ta=0.8 us, cfo=5.4 hz, t=19 us
2022-04-18T05:34:51.873722 [MAC    ] [I] [ 7177] SCHED: UL tx rnti=0x4a, cc=0, pid=1, dci=(0,3), prb=[1, 2), n_rtx=0, cfi=1, tbs=26, bsr=0 (26-26)
2022-04-18T05:34:51.873734 [PHY1   ] [I] [ 7177] PDCCH: cc=0, rnti=0x4a, f=0, cce= 3, L=0, riv=1, mcs=13, rv=0, ndi=0, f_h=n/a, cqi=no, tpc_pusch=1, dmrs_cs=0, tti_tx_dl=7181
2022-04-18T05:34:51.873736 [PHY1   ] [I] [ 7177] PHICH: rnti=0x4a, hi=1, I_lowest=1, n_dmrs=0, tti_tx_dl=7181
2022-04-18T05:34:51.873768 [MAC    ] [I] [ 7177] 0x4a UL SBSR: lcg=0 b=0 LCID=1 len=23
2022-04-18T05:34:51.873769 [RLC    ] [I] SRB1 Rx data PDU SN=1 (21 B)
    0000: 00 42 80 10 00 04 3e 00 08 04 c0 08 ba 02 07 82
    0010: 84 00 00 00 00
2022-04-18T05:34:51.873770 [RLC    ] [I] SRB1 Status packet requested through polling bit
2022-04-18T05:34:51.873770 [RLC    ] [I] SRB1 Rx SDU (81 B)
    0000: 00 20 10 90 2e 58 02 d7 26 04 0e 82 04 10 b2 a2
    0010: 92 22 22 22 22 32 0b e1 e1 81 80 3a 00 0a 04 27
2022-04-18T05:34:51.873772 [PDCP   ] [I] SRB1 Rx PDU SN=0 (81 B, integrity=none, encryption=none)
    0000: 00 20 10 90 2e 58 02 d7 26 04 0e 82 04 10 b2 a2
    0010: 92 22 22 22 22 32 0b e1 e1 81 80 3a 00 0a 04 27
2022-04-18T05:34:51.873774 [MAC    ] [I] [ 7177] SCHED: rnti=0x4a DL lcid=1 buffer_state=0,2
2022-04-18T05:34:51.874754 [MAC    ] [I] [ 7178] SCHED: DL tx rnti=0x4a, cc=0, pid=0, mask=0x1000, dci=(0, 0), n_rtx=0, cfi=1, tbs=7, buffer=7/0, tti_tx_dl=7182
2022-04-18T05:34:51.874757 [RLC    ] [I] SRB1 Tx status PDU - ACK_SN = 2, N_nack = 0
2022-04-18T05:34:51.874760 [MAC    ] [I] [ 7178] 0x4a DL LCID=1 len=2
2022-04-18T05:34:51.874771 [PHY0   ] [I] [ 7178] PDCCH: cc=0, rnti=0x4a, f=1A, cce= 0, L=0, riv=25, pid=0, mcs={1}, ndi={0}, tpc_pucch=1, tti_tx_dl=7182
2022-04-18T05:34:51.874784 [PHY0   ] [I] [ 7178] PDSCH: cc=0, rnti=0x4a, nof_prb=2, nof_re=288, tbs={7}, mod={2}, rv={0}, tx=div, nof_tb=1, nof_l=2, tti_tx_dl=7182
2022-04-18T05:34:51.874804 [RRC    ] [I] Rx SRB1 PDU, rnti=0x4a - rrcConnectionSetupComplete (76 B)
    0000: 20 10 90 2e 58 02 d7 26 04 0e 82 04 10 b2 a2 92
    0010: 22 22 22 22 32 0b e1 e1 81 80 3a 00 0a 04 27 a0
2022-04-18T05:34:51.874823 [RRC    ] [I] RRCConnectionSetupComplete transaction ID: 0
2022-04-18T05:34:51.874836 [S1AP   ] [I] Tx S1AP SDU, InitialUEMessage, rnti=0x4a
    0000: 00 0c 40 71 00 00 05 00 08 00 02 00 05 00 1a 00
    0010: 49 48 17 2c 01 6b 93 02 07 41 02 08 59 51 49 11
```

closed 2022-04-19 by andrepuschmann