srsran/srsRAN_4G #847: SPGW Allocated IP 0.0.0.0 to IMSI 001010123456789

ysgaga, 2022-04-13:
## Issue Description ##
[Describe the issue in detail]
Hi, I install srsran and try 5 UEs connect to it, it is good.
But when I try 6 UEs and more, some UE can not get ip.

## Log ##
Sending Create Session Request.
Creating Session Response -- IMSI: 1016789000013
Creating Session Response -- MME control TEID: 1359
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 1359
SPGW: Allocated User TEID 1359
SPGW: Allocate UE IP 0.0.0.0
Received Create Session Response
Create Session Response -- SPGW control TEID 1359
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 0.0.0.0 to IMSI 001016789000013
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x55b; eNB GTP-U Address 127.0.1.1
UL NAS: Detach Request
Detach request -- IMSI 001016789000006
UL NAS: Detach Request
Detach request -- IMSI 001016789000013
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DELETE_SESSION_REQUEST
UL NAS: Detach Request
Detach request -- IMSI 001016789000012
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x1eb0b6c2
Attach request -- eNB-UE S1AP Id: 1383
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 10
PDN Connectivity Request -- ESM Information Transfer requested: false
Attach Request -- Found previously attach UE.
Found UE context. IMSI: 001016789000004, old eNB UE S1ap Id 1371, old MME UE S1AP Id 2483
GUTI Attach request NAS integrity failed.
RE-starting authentication procedure.
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 001016789000004
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 001016789000004
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 1016789000004
Creating Session Response -- MME control TEID: 1360
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 1360
SPGW: Allocated User TEID 1360
SPGW: Allocate UE IP 0.0.0.0
Received Create Session Response
Create Session Response -- SPGW control TEID 1360
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 0.0.0.0 to IMSI 001016789000004
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x55c; eNB GTP-U Address 127.0.1.1
UL NAS: Detach Request
[... 524 lines cut ...]
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 001016789000013
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 1016789000013
Creating Session Response -- MME control TEID: 1372
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 1372
SPGW: Allocated User TEID 1372
SPGW: Allocate UE IP 0.0.0.0
Received Create Session Response
Create Session Response -- SPGW control TEID 1372
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 0.0.0.0 to IMSI 001016789000013
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x568; eNB GTP-U Address 127.0.1.1
UL NAS: Detach Request
Detach request -- IMSI 001016789000013
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DELETE_SESSION_REQUEST
UL NAS: Detach Request
Detach request -- IMSI 001016789000002

andrepuschmann, 2022-04-14:
Which version is this? I think this issue has been solved a while ago.

ysgaga, 2022-04-21:
> 
Hi,andrepuschmann,
My version is 21_10,it is newest.

andrepuschmann, 2022-04-26:
Please send complete eNB and EPC logs in info level. In general, however, it might be worth to try another EPC, like open5gs. The srsEPC is a light-weight testing EPC and was never intended for serious use.

mazimkhan, 2022-06-27:
For me this happens when UE IPs are exhausted 
[image]

matan1008, 2022-06-27:
Do you mind trying my branch https://github.com/srsran/srsRAN/pull/647?

andrepuschmann, 2022-06-30:
@mazimkhan As @matan1008 said we would be really grateful if you could test his branch with your exact scenario. If you can, could you also post the log entries (like above) when the rollover happens?