srsran/srsRAN_4G #852: 22.04 undefined reference to `typeinfo for srsenb::sched_nr_base_test_bench'

ZeroChaos-, 2022-05-06:
Config looks like this
```
cmake -C /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_UHD=yes -DENABLE_BLADERF=yes -DENABLE_SOAPYSDR=yes -DENABLE_ZEROMQ=yes -DENABLE_HARDSIM=yes -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/gentoo_toolchain.cmake /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04
loading initial cache file /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/gentoo_common_config.cmake
-- The C compiler identification is GNU 11.2.1
-- The CXX compiler identification is GNU 11.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-5.17.3-pentoo
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/x86_64-pc-linux-gnu-g++
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/x86_64-pc-linux-gnu-pkg-config (found version "1.8.0") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.10
-- FFTW3F LIBRARIES: /usr/lib64/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: FFTW3F_STATIC_LIBRARY-NOTFOUND
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found FFTW3F: /usr/lib64/libfftw3f.so  
-- FFT_LIBRARIES: /usr/lib64/libfftw3f.so
-- Checking for module 'polarssl'
--   Package 'polarssl', required by 'virtual:world', not found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find Polarssl (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   Package 'mbedtls', required by 'virtual:world', not found
-- MBEDTLS LIBRARIES: /usr/lib64/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: MBEDTLS_STATIC_LIBRARIES-NOTFOUND
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MbedTLS: /usr/lib64/libmbedcrypto.so  
-- Checking for module 'libpcsclite'
--   Found libpcsclite, version 1.9.0
-- Found PCSCLite: /usr/lib64/libpcsclite.so  
-- PCSC LIBRARIES: /usr/lib64/libpcsclite.so
-- PCSC INCLUDE DIRS: /usr/include/PCSC
-- Building with PCSC support.
-- UHD LIBRARIES /usr/lib64/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- Found UHD: /usr/lib64/libuhd.so  
-- Performing Test UHD_ENABLE_X300_FW_RESET
-- Performing Test UHD_ENABLE_X300_FW_RESET - Failed
-- Performing Test UHD_ENABLE_RFNOC
-- Performing Test UHD_ENABLE_RFNOC - Failed
-- Performing Test UHD_ENABLE_CUSTOM_RFNOC
-- Performing Test UHD_ENABLE_CUSTOM_RFNOC - Failed
-- SKIQ LIBRARIES SKIQ_LIBRARY-NOTFOUNDSKIQ_LIBRARY_GLIB-NOTFOUNDSKIQ_LIBRARY_USB-NOTFOUND/usr/lib/libtirpc.so.3
-- SKIQ INCLUDE DIRS SKIQ_INCLUDE_DIRS-NOTFOUND
-- Could NOT find SKIQ (missing: SKIQ_INCLUDE_DIRS) 
-- Checking for module 'libbladeRF'
--   Found libbladeRF, version 2.4.0-2021.03
-- Looking for bladerf_get_board_name in bladeRF
-- Looking for bladerf_get_board_name in bladeRF - found
[... 91 lines cut ...]
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Module          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Shared          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0

-- Configuring done
```
Build failure here:
```
[1023/1088] : && /usr/bin/x86_64-pc-linux-gnu-g++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++14 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -ggdb -DBUILD_TYPE_RELWITHDEBINFO -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o -o srsgnb/src/stack/mac/test/sched_nr_parallel_test  -Wl,-rpath,/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/srsgnb/src/stack/mac/test  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsgnb/src/stack/mac/test/libsched_nr_test_suite.so  lib/src/common/libsrsran_common.a  lib/src/asn1/librrc_nr_asn1.a  /usr/lib64/libboost_program_options.so.1.78.0  /usr/lib64/libboost_system.so.1.78.0  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsenb/src/stack/mac/common/libsrsenb_mac_common.a  lib/src/mac/libsrsran_mac.a  lib/src/asn1/librrc_nr_asn1.a  lib/src/asn1/libasn1_utils.a  lib/src/common/libsrsran_common.a  lib/src/phy/libsrsran_phy.a  -lpthread  -lm  /usr/lib64/libfftw3f.so  lib/src/support/libsupport.a  lib/src/srslog/libsrslog.a  /usr/lib64/libmbedcrypto.so  /usr/lib64/libdw.so && :
FAILED: srsgnb/src/stack/mac/test/sched_nr_parallel_test 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++14 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -ggdb -DBUILD_TYPE_RELWITHDEBINFO -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o -o srsgnb/src/stack/mac/test/sched_nr_parallel_test  -Wl,-rpath,/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/srsgnb/src/stack/mac/test  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsgnb/src/stack/mac/test/libsched_nr_test_suite.so  lib/src/common/libsrsran_common.a  lib/src/asn1/librrc_nr_asn1.a  /usr/lib64/libboost_program_options.so.1.78.0  /usr/lib64/libboost_system.so.1.78.0  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsenb/src/stack/mac/common/libsrsenb_mac_common.a  lib/src/mac/libsrsran_mac.a  lib/src/asn1/librrc_nr_asn1.a  lib/src/asn1/libasn1_utils.a  lib/src/common/libsrsran_common.a  lib/src/phy/libsrsran_phy.a  -lpthread  -lm  /usr/lib64/libfftw3f.so  lib/src/support/libsupport.a  lib/src/srslog/libsrslog.a  /usr/lib64/libmbedcrypto.so  /usr/lib64/libdw.so && :
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `srsenb::sched_nr_tester::sched_nr_base_test_bench(srsenb::sched_nr_interface::sched_args_t const&, std::vector<srsenb::sched_nr_cell_cfg_t, std::allocator<srsenb::sched_nr_cell_cfg_t> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:38: undefined reference to `srsenb::sched_nr_base_test_bench::sched_nr_base_test_bench(srsenb::sched_nr_interface::sched_args_t const&, std::vector<srsenb::sched_nr_cell_cfg_t, std::allocator<srsenb::sched_nr_cell_cfg_t> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:38: undefined reference to `srsenb::sched_nr_base_test_bench::~sched_nr_base_test_bench()'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `srsenb::run_sched_nr_test(unsigned int)':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:101: undefined reference to `srsenb::sched_nr_base_test_bench::user_cfg(unsigned short, srsenb::sched_nr_ue_cfg_t const&)'
[... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o:(.data.rel.ro._ZTIN6srsenb15sched_nr_testerE[_ZTIN6srsenb15sched_nr_testerE]+0x10): undefined reference to `typeinfo for srsenb::sched_nr_base_test_bench'
collect2: error: ld returned 1 exit status
```
Seems to get a bit further than https://github.com/srsran/srsRAN/issues/835 , but still doesn't build.  Is there a way to disable these tests that I'm missing?  How do others build this successfully?

robertfalkenberg, 2022-05-06:
Hi,

it seems there is an issue with a very recent GCC version. Your logs report
```
-- The C compiler identification is GNU 11.2.1
-- The CXX compiler identification is GNU 11.2.1
```

You could try to build with GCC <= 11.2.0 or Clang <= 13.0.1, which are the most recent versions that I have tested (on Archlinux).

ZeroChaos-, 2022-05-06:
Clang 13.0.1 is available in Gentoo and the failure looks pretty similar
```
[1022/1088] : && /usr/lib/llvm/13/bin/clang++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++14 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -ggdb -DBUILD_TYPE_RELWITHDEBINFO -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o -o srsgnb/src/stack/mac/test/sched_nr_parallel_test  -Wl,-rpath,/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/srsgnb/src/stack/mac/test  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsgnb/src/stack/mac/test/libsched_nr_test_suite.so  lib/src/common/libsrsran_common.a  lib/src/asn1/librrc_nr_asn1.a  /usr/lib64/libboost_program_options.so.1.78.0  /usr/lib64/libboost_system.so.1.78.0  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsenb/src/stack/mac/common/libsrsenb_mac_common.a  lib/src/mac/libsrsran_mac.a  lib/src/asn1/librrc_nr_asn1.a  lib/src/asn1/libasn1_utils.a  lib/src/common/libsrsran_common.a  lib/src/phy/libsrsran_phy.a  -lpthread  -lm  /usr/lib64/libfftw3f.so  lib/src/support/libsupport.a  lib/src/srslog/libsrslog.a  /usr/lib64/libmbedcrypto.so  /usr/lib64/libdw.so && :
FAILED: srsgnb/src/stack/mac/test/sched_nr_parallel_test 
: && /usr/lib/llvm/13/bin/clang++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++14 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -ggdb -DBUILD_TYPE_RELWITHDEBINFO -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o -o srsgnb/src/stack/mac/test/sched_nr_parallel_test  -Wl,-rpath,/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/srsgnb/src/stack/mac/test  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsgnb/src/stack/mac/test/libsched_nr_test_suite.so  lib/src/common/libsrsran_common.a  lib/src/asn1/librrc_nr_asn1.a  /usr/lib64/libboost_program_options.so.1.78.0  /usr/lib64/libboost_system.so.1.78.0  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsenb/src/stack/mac/common/libsrsenb_mac_common.a  lib/src/mac/libsrsran_mac.a  lib/src/asn1/librrc_nr_asn1.a  lib/src/asn1/libasn1_utils.a  lib/src/common/libsrsran_common.a  lib/src/phy/libsrsran_phy.a  -lpthread  -lm  /usr/lib64/libfftw3f.so  lib/src/support/libsupport.a  lib/src/srslog/libsrslog.a  /usr/lib64/libmbedcrypto.so  /usr/lib64/libdw.so && :
/usr/bin/x86_64-pc-linux-gnu-ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `sched_nr_base_test_bench':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:38: undefined reference to `srsenb::sched_nr_base_test_bench::sched_nr_base_test_bench(srsenb::sched_nr_interface::sched_args_t const&, std::vector<srsenb::sched_nr_cell_cfg_t, std::allocator<srsenb::sched_nr_cell_cfg_t> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)'
/usr/bin/x86_64-pc-linux-gnu-ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `srsenb::run_sched_nr_test(unsigned int)':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:101: undefined reference to `srsenb::sched_nr_base_test_bench::user_cfg(unsigned short, srsenb::sched_nr_ue_cfg_t const&)'
/usr/bin/x86_64-pc-linux-gnu-ld: /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:103: undefined reference to `srsenb::sched_nr_base_test_bench::run_slot(srsran::slot_point)'
/usr/bin/x86_64-pc-linux-gnu-ld: /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:106: undefined reference to `srsenb::sched_nr_base_test_bench::stop()'
/usr/bin/x86_64-pc-linux-gnu-ld: /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:114: undefined reference to `srsenb::sched_nr_base_test_bench::~sched_nr_base_test_bench()'
/usr/bin/x86_64-pc-linux-gnu-ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `sched_nr_base_test_bench':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:38: undefined reference to `srsenb::sched_nr_base_test_bench::~sched_nr_base_test_bench()'
/usr/bin/x86_64-pc-linux-gnu-ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `srsenb::run_sched_nr_test(unsigned int)':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:114: undefined reference to `srsenb::sched_nr_base_test_bench::~sched_nr_base_test_bench()'
/usr/bin/x86_64-pc-linux-gnu-ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `~sched_nr_tester':
[... cut ...]
/usr/bin/x86_64-pc-linux-gnu-ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o:(.data.rel.ro._ZTVN6srsenb15sched_nr_testerE[_ZTVN6srsenb15sched_nr_testerE]+0x10): undefined reference to `srsenb::sched_nr_base_test_bench::~sched_nr_base_test_bench()'
/usr/bin/x86_64-pc-linux-gnu-ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o:(.data.rel.ro._ZTIN6srsenb15sched_nr_testerE[_ZTIN6srsenb15sched_nr_testerE]+0x10): undefined reference to `typeinfo for srsenb::sched_nr_base_test_bench'
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
```

ZeroChaos-, 2022-05-06:
So I just tested, and this bug seems to only pop up with -Os.  If I compile with -O3 instead, I get https://github.com/srsran/srsRAN/issues/854

robertfalkenberg, 2022-05-07:
I can see some external configurations being loaded, e.g.
```
cmake -C /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_UHD=yes -DENABLE_BLADERF=yes -DENABLE_SOAPYSDR=yes -DENABLE_ZEROMQ=yes -DENABLE_HARDSIM=yes -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/gentoo_toolchain.cmake /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04
```
and a few extra CC/CXX flags, (on the first sight for example `-Os -march=native -mtune=native -pipe -frecord-gcc-switches`)

Could you please also try to build without them? https://docs.srsran.com/en/latest/general/source/1_installation.html

ZeroChaos-, 2022-05-09:
I removed my custom cflags from gentoo_rules.cmake and from the environment and rebuilt.  I get an extremely similar error still
```
[1022/1088] : && /usr/bin/x86_64-pc-linux-gnu-g++ -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++14 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -ggdb -DBUILD_TYPE_RELWITHDEBINFO -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o -o srsgnb/src/stack/mac/test/sched_nr_parallel_test  -Wl,-rpath,/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/srsgnb/src/stack/mac/test  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsgnb/src/stack/mac/test/libsched_nr_test_suite.so  lib/src/common/libsrsran_common.a  lib/src/asn1/librrc_nr_asn1.a  /usr/lib64/libboost_program_options.so.1.78.0  /usr/lib64/libboost_system.so.1.78.0  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsenb/src/stack/mac/common/libsrsenb_mac_common.a  lib/src/mac/libsrsran_mac.a  lib/src/asn1/librrc_nr_asn1.a  lib/src/asn1/libasn1_utils.a  lib/src/common/libsrsran_common.a  lib/src/phy/libsrsran_phy.a  -lpthread  -lm  /usr/lib64/libfftw3f.so  lib/src/support/libsupport.a  lib/src/srslog/libsrslog.a  /usr/lib64/libmbedcrypto.so  /usr/lib64/libdw.so  /usr/lib64/libsctp.so && :
FAILED: srsgnb/src/stack/mac/test/sched_nr_parallel_test 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++14 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -ggdb -DBUILD_TYPE_RELWITHDEBINFO -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o -o srsgnb/src/stack/mac/test/sched_nr_parallel_test  -Wl,-rpath,/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04_build/srsgnb/src/stack/mac/test  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsgnb/src/stack/mac/test/libsched_nr_test_suite.so  lib/src/common/libsrsran_common.a  lib/src/asn1/librrc_nr_asn1.a  /usr/lib64/libboost_program_options.so.1.78.0  /usr/lib64/libboost_system.so.1.78.0  srsgnb/src/stack/mac/libsrsgnb_mac.a  srsenb/src/stack/mac/common/libsrsenb_mac_common.a  lib/src/mac/libsrsran_mac.a  lib/src/asn1/librrc_nr_asn1.a  lib/src/asn1/libasn1_utils.a  lib/src/common/libsrsran_common.a  lib/src/phy/libsrsran_phy.a  -lpthread  -lm  /usr/lib64/libfftw3f.so  lib/src/support/libsupport.a  lib/src/srslog/libsrslog.a  /usr/lib64/libmbedcrypto.so  /usr/lib64/libdw.so  /usr/lib64/libsctp.so && :
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `srsenb::run_sched_nr_test(unsigned int)':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:101: undefined reference to `srsenb::sched_nr_base_test_bench::user_cfg(unsigned short, srsenb::sched_nr_ue_cfg_t const&)'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:103: undefined reference to `srsenb::sched_nr_base_test_bench::run_slot(srsran::slot_point)'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:106: undefined reference to `srsenb::sched_nr_base_test_bench::stop()'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `srsenb::sched_nr_tester::sched_nr_base_test_bench(srsenb::sched_nr_interface::sched_args_t const&, std::vector<srsenb::sched_nr_cell_cfg_t, std::allocator<srsenb::sched_nr_cell_cfg_t> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:38: undefined reference to `srsenb::sched_nr_base_test_bench::sched_nr_base_test_bench(srsenb::sched_nr_interface::sched_args_t const&, std::vector<srsenb::sched_nr_cell_cfg_t, std::allocator<srsenb::sched_nr_cell_cfg_t> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: /var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:38: undefined reference to `srsenb::sched_nr_base_test_bench::~sched_nr_base_test_bench()'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o: in function `srsenb::sched_nr_tester::~sched_nr_tester()':
/var/tmp/portage/net-wireless/srsran-22.04/work/srsran-22.04/srsgnb/src/stack/mac/test/sched_nr_parallel_test.cc:35: undefined reference to `srsenb::sched_nr_base_test_bench::~sched_nr_base_test_bench()'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.1/../../../../x86_64-pc-linux-gnu/bin/ld: srsgnb/src/stack/mac/test/CMakeFiles/sched_nr_parallel_test.dir/sched_nr_parallel_test.cc.o:(.data.rel.ro._ZTIN6srsenb15sched_nr_testerE[_ZTIN6srsenb15sched_nr_testerE]+0x10): undefined reference to `typeinfo for srsenb::sched_nr_base_test_bench'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

robertfalkenberg, 2022-05-10:
Thanks for the feedback.

I could reproduce your build issue by adding `BUILD_SHARED_LIBS=ON` as configured by the `gentoo_common_config.cmake` you posted before.

I added a patch to fix the build process, regardless of whether that option is set or not.
You can find the branch here: https://github.com/srsran/srsRAN/tree/fix_issue_852

Could you please confirm if this patch fixes the build process?

P.S. As an immediate workaround I suggest to set `BUILD_SHARED_LIBS` to `OFF` explicitly,
e.g. append `-DBUILD_SHARED_LIBS=OFF` to the cmake args.

ZeroChaos-, 2022-05-10:
Nice, confirmed that https://github.com/srsran/srsRAN/commit/2e1a7abfb313f874add9efa2cdd716dffd842782.patch fixes this.  I'm retesting my other bugs but I am fairly certain they are all interelated and just happened to hit a failure at a slightly different point.