srsran/srsRAN_4G #853: Support for metrics in external time series database

hashkeks, 2022-05-06:
Hello, 

first I'd like to thank for this great project and the good documentation. Unfortunately I couldn't find an answer to my following question in the documentations or e-mail list archives.

## Issue Description ##
Is there some kind of support for sending metrics data to an external time series database like InfluxDB to include the metrics in visualization tools? 
I see, that there is the possibility to write metrics to a csv file, so I'd need to make a tool to send that data to a TSDB. But if there happens to be a tool available or even a built in function implemented, I'd rather use these.

Thank you in advance for taking the time to read and answer.

All the best,
Cedric

FawalWasiela, 2022-05-08:
I am also interested in this. I tried to look for the variables I needed inside the code itself but I am not sure if simply logging these values will be viable.

andrepuschmann, 2022-05-09:
Hey, there is no such function built-in in srsRAN. But as you can image it would be rather easy to extend the CSV or Json metrics writer with that.
Hope that helps.
Closing the ticket.

closed 2022-05-09 by andrepuschmann