srsran/srsRAN_4G #858: Issues running srsENB for 5G NR SA using OPPO 5G A73

nomidaepapi, 2022-05-10:
**When running srsENB for 5G NR SA:**

Built in Release mode using commit f2dff0b7a on branch master.                                                                                                                                                                                  /root/srsRAN/srsenb/src/enb_cfg_parser.cc:1818: DL-ARFCN 173800 in band n5 not supported with coreset0 config.          /root/srsRAN/srsenb/src/enb_cfg_parser.cc:1819: Valid ARFCNs for band n5 are:                                           
Error deriving NR cell parameters                                                                                       
Error processing arguments.  

Hey. I followed all the steps according to the 5G documentation and everything is up and running but im facing issues attaching the UE with the spedified ARFCNs on the docummentation. None of the ARFCNs are able to be detected from my phone. Are there any other sets of ARFCNs i can try? Or do i have to change the phone to the one SRS uses which is the Nord phone. (my Country: UAE)

jtang613, 2022-05-11:
See https://github.com/srsran/srsRAN/blob/f2dff0b7a03e065060b48ea0a3c7a6a7890dfd46/srsenb/src/enb_cfg_parser.cc#L1834 for a list of DL_ARFCN's that are enabled.

nomidaepapi, 2022-05-12:
I have tried all of these ARFCNs and the phone does not respond to any of these. Could it be that the local OPPO phone i have does not support these frequencies?

hawk933, 2022-05-24:
> I have tried all of these ARFCNs and the phone does not respond to any of these. Could it be that the local OPPO phone i have does not support these frequencies?

Huawei p40 pro doesn't support me either

nomidaepapi, 2022-05-28:
> > I have tried all of these ARFCNs and the phone does not respond to any of these. Could it be that the local OPPO phone i have does not support these frequencies?
> 
> Huawei p40 pro doesn't support me either

Same. Ive tried my best with multiple phones and none of them are able to connect..

hawk933, 2022-05-28:
> 
> 
> 
> 
> > > I have tried all of these ARFCNs and the phone does not respond to any of these. Could it be that the local OPPO phone i have does not support these frequencies?
> 
> > 
> 
> > Huawei p40 pro doesn't support me either
> 
> 
> 
> Same. Ive tried my best with multiple phones and none of them are able to connect..
> 
> 

I have already tried five times, I have not seen a single one

mode51software, 2022-08-19:
I have an Oppo A54 5G that can see 375000 in band 3. I can see the PLMN name (for 001-01) n the scan but haven't managed to get it to register yet. It does appear to get RNTIs though.

I see the Oppo A53 5G doesn't have band 3 support. The latest release doesn't have that same hardcoded list of ARFCNs though.