srsran/srsRAN_4G #860: "Could not set TX rate to ..." with USRP N310

fzybot, 2022-05-12:
## Issue Description ##
The warning is showing up:
[WARNING] [MULTI_USRP] Could not set TX rate to 11.520 MHz. Actual rate is 11.171 MHz
[WARNING] [0/DUC#0] The requested interpolation is odd; the user should expect passband CIC rolloff.
Select an even interpolation to ensure that a halfband filter is enabled.
## Setup Details ##
USRP N310 and Samsung Galaxy S21+5G.

andrepuschmann, 2022-05-12:
You have to use the LTE rates option for the N310.

fzybot, 2022-05-12:
> [andrepuschmann](https://github.com/andrepuschmann)

Which parameter shoud i chenge or add?

andrepuschmann, 2022-05-12:
See [here](https://docs.srsran.com/en/latest/app_notes/source/5g_nsa_cots/source/index.html?highlight=n310#ettus-research-usrp-n310).

I am closing the issue, feel free to reopen if there are still issues.

closed 2022-05-12 by andrepuschmann