srsran/srsRAN_4G #861: make test produces a lot of segfaults

Rongjun-Chen, 2022-05-12:
## Issue Description ##
 I run make test and get 968 segfaults while building srsran 21.10.0.
[cmake_log.txt](https://github.com/srsran/srsRAN/files/8676741/cmake_log.txt)
[make_test_log.txt](https://github.com/srsran/srsRAN/files/8676750/make_test_log.txt)

## Setup Details ##
I compiled it in USRP E310.
CPU: ARM Cortex A9 667 MHz dual-core processor
GCC：arm-oe-linux-gnueabi

## Expected Behavior ##
all tests should pass

## Actual Behaviour ##
33% tests passed, 968 tests failed out of 1452

Label Time Summary:
chest     = 555.31 sec*proc (22 tests)
fec       =  10.59 sec*proc (11 tests)
ldpc      =   0.10 sec*proc (1 test)
lib       = 4244.66 sec*proc (987 tests)
lte       = 4004.25 sec*proc (924 tests)
nr        = 689.15 sec*proc (280 tests)
pdcp      =   5.11 sec*proc (3 tests)
phch      = 1353.86 sec*proc (467 tests)
phy       = 4350.22 sec*proc (986 tests)
polar     =   0.02 sec*proc (1 test)
rlc       =  32.50 sec*proc (11 tests)
srsenb    = 136.96 sec*proc (10 tests)
srsue     =   6.21 sec*proc (3 tests)
turbo     =  10.47 sec*proc (9 tests)

Total Test time (real) = 4903.56 sec

The following tests FAILED:
	  4 - chest_test_dl_cellid0 (SEGFAULT)
	  5 - chest_test_dl_cellid1 (SEGFAULT)
	  6 - chest_test_dl_cellid2 (SEGFAULT)
	  7 - chest_test_dl_cellid0_50prb (SEGFAULT)
	  8 - chest_test_dl_cellid1_50prb (SEGFAULT)
	  9 - chest_test_dl_cellid2_50prb (SEGFAULT)

## Steps to reproduce the problem ##
Install everything using the guide and run the make test command

## Additional Information ##

KaiQia, 2022-06-02:
These segfaults seem to occur at the physical layer, you can test if the required libraries are compiled correctly, or use GDB to debug these segfaults.

Rongjun-Chen, 2022-06-11:
Thank you!

I use GDB to debug the code , it is indeed a library causes errors. It works after I recompile the library.

xavierarteaga, 2023-04-27:
I understand that you solved the issue by recompiling the library. I close the issue.

closed 2023-04-27 by xavierarteaga