srsran/srsRAN_4G #862: 5G SA End-To-End with ZeroMQ sometimes fails after a while

TallGuy74, 2022-05-12:
## Issue Description ##
I have used [the 5G SA End-to-End appnote](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_E2E/source/index.html) to set up a 5G Core network + srsenb + srsue on a machine with an i9 12 core CPU (gen 12).
I am sometimes unable to use the network connection between UE and gNB, resulting in errors.

## Setup Details ##
* A Dell machine with an i9 12900K CPU
* srsRAN 22.04, compiled with UHD, BladeRF, SoapySDR and ZeroMQ RF frontends
* Open5GS configured according to the 5G SA E2E appnote
* Using the configuration files from the 5G SA E2E appnote
* `iperf3 -s` running in the normal network context
* `sudo ip netns exec ue1 iperf3 -c 10.45.0.1` running in the ue1 network context.

## Expected Behavior ##
The iperf3 client is able to communicate with the gNB endpoint running the iperf3 server.

## Actual Behaviour ##
If the connection has been idle for a while, I am sometimes no longer able to contact the gNB's IP address from the ue1 network context.

I see the following errors on the console:
```
RRC NR reconfiguration successful.                                                             [9/660]│Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)              [34/1976]
PDU Session Establishment successful. IP: 10.45.0.2                                                   │Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=0 nof_prb=52
RRC NR reconfiguration successful.                                                                    │RACH:  slot=171, cc=0, preamble=0, offset=0, temp_crnti=0x4601                                       
Scheduling request failed: releasing RRC connection...                                                │Disconnecting rnti=0x4601.                                                                           
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=7531                 │RACH:  slot=7531, cc=0, preamble=0, offset=0, temp_crnti=0x4602                                      
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=7691                 │Disconnecting rnti=0x4602.                                                                           
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=7851                 │RACH:  slot=7691, cc=0, preamble=0, offset=0, temp_crnti=0x4603                                      
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=8011                 │Disconnecting rnti=0x4603.                                                                           
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=8171                 │RACH:  slot=7851, cc=0, preamble=0, offset=0, temp_crnti=0x4604                                      
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=8331                 │Disconnecting rnti=0x4604.                                                                           
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=8491                 │RACH:  slot=8011, cc=0, preamble=0, offset=0, temp_crnti=0x4605
Scheduling request failed: releasing RRC connection...                                                │Disconnecting rnti=0x4605.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=171                  │RACH:  slot=8171, cc=0, preamble=0, offset=0, temp_crnti=0x4606
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=331                  │Disconnecting rnti=0x4606.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=491                  │RACH:  slot=8331, cc=0, preamble=0, offset=0, temp_crnti=0x4607
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=651                  │Disconnecting rnti=0x4607.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=811                  │RACH:  slot=8491, cc=0, preamble=0, offset=0, temp_crnti=0x4608
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=971                  │Disconnecting rnti=0x4608.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=1131                 │RACH:  slot=171, cc=0, preamble=0, offset=0, temp_crnti=0x4609
Scheduling request failed: releasing RRC connection...                                                │Disconnecting rnti=0x4609.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=4011                 │RACH:  slot=331, cc=0, preamble=0, offset=0, temp_crnti=0x460a
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=4171                 │Disconnecting rnti=0x460a.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=4331                 │RACH:  slot=491, cc=0, preamble=0, offset=0, temp_crnti=0x460b
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=4491                 │Disconnecting rnti=0x460b.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=4651                 │RACH:  slot=651, cc=0, preamble=0, offset=0, temp_crnti=0x460c
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=4811                 │Disconnecting rnti=0x460c.
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=4971                 │RACH:  slot=811, cc=0, preamble=0, offset=0, temp_crnti=0x460d
```

## Steps to reproduce the problem ##
Run the iperf3 client, wait a while, and try again.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

mariatsampazi, 2022-06-10:
Have you managed to figure it out?

andrepuschmann, 2022-06-10:
Hey,
the issue comes from the UE not handling RRC idle mode correctly. It's a limitation that renders the UE failing to connect again after it was put in RRC idle (after RRC Release). Currently the best way to circumvent this is to use a larger RRC inactivity timer in the eNB.