srsran/srsRAN_4G #866: 5g SA amarisoft using USRP N310

ayamoheddine, 2022-05-20:
## Issue Description ##

I’m trying to duplicate the scenario presented in the latest version of srsRAN using 5g SA configuration. I followed the guidelines mentioned in the latest release in the app notes. However, the USRP is not able to connect to the network and stuck in the UE attaching step.
 
Link to the latest version of srsRAN followed:
 
[https://docs.srsran.com/en/latest/app_notes/source/5g_sa_amari/source/index.html#g-sa-srsue](url)
 
## Setup Details ##

The setup used is:
N310 USRP as user equipment, connected through 10Gbit interface to ubuntu 18.04 PC. 
Amarisoft Classic Callbox as 5G core network.

The USRP and amarisoft are connected through a set of antennas. 

## Expected Behavior ##

Successful attachment of the UE to 5G network generated by the amarisoft callbox. 
## Actual Behaviour ##
Stuck in the UE attach phase with lots of RF underflows. 

## Steps to reproduce the problem ##

Changed the ue.conf file as mentioned in the app notes and as attached in ue.txt file and the execution is given within the screenshot. 
[ue.txt](https://github.com/srsran/srsRAN/files/8738446/ue.txt)
[image]

## Additional Information ##
The ue.log file can be found here. 

[usrp_dl_earfcn=2850_ue.txt](https://github.com/srsran/srsRAN/files/8738449/usrp_dl_earfcn.2850_ue.txt)

Your help is highly appreciated.

andrepuschmann, 2022-05-20:
We've not tested the N310 with the UE. But your sample frequency offset and `srate` params look odd to me.

ayamoheddine, 2022-05-20:
> We've not tested the N310 with the UE. But your sample frequency offset and `srate` params look odd to me.

The srate and sample frequency offset are those of N310 as can be seen from this table
[image]

andrepuschmann, 2022-05-23:
But you're trying to receive a 10 MHz wide signal. So sampling at 7.68Msps isn't enough. Try e.g. 15.36e6

ayamoheddine, 2022-05-23:
Nothing changed. Here is the ue.log file. 
[ue-15.35e6.txt](https://github.com/srsran/srsRAN/files/8752383/ue-15.35e6.txt)

TallGuy74, 2022-05-24:
Could you try dropping the rx_gain and tx_gain to a (much) lower number that is not 0? The CRC on the frames you receive is KO, and when I had that happen on my end (admittedly with 5G SA instead of NSA) dropping the gains helped a lot. I still see a lot of KO's, but I also see regular OK's with correlations large enough to be relevant.

Justin-Garey, 2024-10-04:
Did the USRP N310 end up being tested or confirmed working with the 5G UE?