srsran/srsRAN_4G #867: 5G NSA  with COST UE failed to generated gNB

coolcoollly, 2022-05-20:
## Issue Description ##
Run code with the tutorial :  https://docs.srsran.com/en/latest/app_notes/source/5g_nsa_cots/source/index.html
After running srsenb ,  only come out lte enb without gnb.
my phone can connect to lte enb to internet, but there only lte without nr.

## Setup Details ##
srsran 21.10 ,  ubuntu 18.04 , uhd v3.15 , usrp X310.

## Expected Behavior ##
When running srsenb, there will come out two nodes(enb, gnb).

## Actual Behaviour ##
There only came out one node (enb).

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
[user_db.csv](https://github.com/srsran/srsRAN/files/8740909/user_db.csv)
[enb.txt](https://github.com/srsran/srsRAN/files/8740936/enb.txt)
[epc.txt](https://github.com/srsran/srsRAN/files/8740937/epc.txt)
[rr.txt](https://github.com/srsran/srsRAN/files/8740938/rr.txt)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2022-05-23:
In NSA mode, both LTE and NR carrier need to be SISO. In your enb.conf, make sure to not use MIMO.
```
n_prb = 50
tm = 1
nof_ports = 1
```

coolcoollly, 2022-05-26:
> In NSA mode, both LTE and NR carrier need to be SISO. In your enb.conf, make sure to not use MIMO.
> 
> ```
> n_prb = 50
> tm = 1
> nof_ports = 1
> ```

But when i set nof_ports=1, srsenb only opens 1 channel, and only RFA of usrp lights.

coolcoollly, 2022-05-26:
[image]

coolcoollly, 2022-05-26:
When i set nof_ports=2, device_args like with the tutorial (device_args=type=x300,lo_freq_offset_hz=23.04e6,send_frame_size=8000,recv_frame_size=8000,num_send_frames=64,num_recv_frames=64,None=,master_clock_rate=184.32e6), there is another erroer.
[image]

coolcoollly, 2022-05-27:
thanks. Finally i find I run the original rr.conf instead of the one modified.

closed 2022-05-27 by coolcoollly