srsran/srsRAN_4G #870: 5G SA End-to-End with USRP B210's

TallGuy74, 2022-05-23:
## Issue Description ##
I am unable to get a connection if I use two B210's instead of the ZeroMQ setup.
I get a connection on RRC level, but after that things just stop.

## Setup Details ##
I use two Dell Inc. XPS 8950/09D2HH, with 32 GB of RAM and a 12900K CPU. Both of them are connected over USB3 to a USRP B210 with a cable connecting the RX and TX (with a 30 dB attenuation).
One of the Dell machines is running Open5GS, with the subscriber added.

## Expected Behavior ##
A valid 5G connection, where I can ping the gNB machine from the UE machine.

## Actual Behaviour ##
```
Inactive RF plugins:                                                                                      │Reading configuration file enb-uhd.conf...
Reading configuration file ue-uhd.conf...                                                                 │Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
                                                                                                          │Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
Built in RelWithDebInfo mode using 22.04.0.                                                               │
                                                                                                          │Built in RelWithDebInfo mode using 22.04.0.
Opening 1 channels in RF device=UHD with args=sampling_rate=23.04e6,lo_freq_offset_hz=23.04e6             │
Supported RF device list: UHD soapy bladeRF zmq file                                                      │Opening 1 channels in RF device=UHD with args=sampling_rate=23.04e6,lo_freq_offset_hz=23.04e6
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1                      │Supported RF device list: UHD soapy bladeRF zmq file
[INFO] [LOGGING] Fastpath logging disabled at runtime.                                                    │NG connection successful
Opening USRP channels=1, args: lo_freq_offset_hz=23.04e6,type=b200,master_clock_rate=23.04e6              │[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[INFO] [UHD RF] RF UHD Generic instance constructed                                                       │[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [B200] Detected Device: B210                                                                       │Opening USRP channels=1, args: lo_freq_offset_hz=23.04e6,type=b200,master_clock_rate=23.04e6
[INFO] [B200] Operating over USB 3.                                                                       │[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detecting internal GPSDO....                                                                │[INFO] [B200] Detected Device: B210
[INFO] [GPS] Found an internal GPSDO: GPSTCXO , Firmware Rev 0.929a                                       │[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...                                                                 │[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...                                                                 │[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...                                                        │[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed                                                               │[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...                                                        │[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed                                                               │[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz...                                                      │[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.                                                      │[INFO] [B200] Actually got clock rate 23.040000 MHz.
Waiting PHY to initialize ... done!                                                                       │
Attaching UE...                                                                                           │==== eNodeB started ===
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=5931                     │Type <t> to view trace
Random Access Complete.     c-rnti=0x4601, ta=1                                                           │Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=0 nof_prb=52
RRC Connected                                                                                             │RACH:  slot=5931, cc=0, preamble=0, offset=1, temp_crnti=0x4601
                                                                                                          │

```
After the RRC is connected, I get a lot of this in my log file; the PHY configuration is done, SR procedure is starting, and it keeps trying to retransmit (and failing because it doesn't have anything to transmit? not sure)

```
2022-05-23T09:32:47.037092 [MAC-NR ] [I] [ 5948] SR:    Starting procedure
2022-05-23T09:32:47.043136 [PHY1-NR] [I] [ 5954] PDCCH: cc=0, c-rnti=0x4601 dci=1_0 ss=common0 L=2 cce=0 f_alloc=0x5f t_alloc=0x0 vrb_to_prb_map=0 mcs=28 ndi=1 rv=2 harq_id=0 dai=0 pucch_tpc=1 pucch_res=0 harq_feedback=3 
2022-05-23T09:32:47.043137 [PHY1-NR] [I] [ 5954] PDCCH: coreset=0, 
2022-05-23T09:32:47.043149 [MAC-NR ] [I] [ 5954] DL 0: Received duplicate. Discarding and retransmitting ACK (n_retx=1)
2022-05-23T09:32:47.043149 [PHY1-NR] [I] [ 5954] Decoding not required. Skipping PDSCH. ack_tti_tx=5958
2022-05-23T09:32:47.043176 [PHY1-NR] [I] [ 5954] PUCCH: cc=0, f=1 prb=51:1 symb=0:14 cs=0 occ=0 rnti=0x4601 ack=1 , tti_tx=5958
2022-05-23T09:32:47.044183 [PHY0-NR] [I] [ 5955] PDCCH: cc=0, c-rnti=0x4601 dci=0_0 ss=common0 L=2 cce=0 f_alloc=0x67 t_alloc=0x0 hop=n mcs=28 ndi=0 rv=2 harq_id=0 tpc=1 
2022-05-23T09:32:47.044192 [MAC    ] [I] [ 5955] UL 0:  Retx=1, rv=2, tbs=4737
2022-05-23T09:32:47.044425 [PHY0-NR] [I] [ 5955] PUSCH: cc=0 pid=0 c-rnti=0x4601 prb=(0,51) symb=(0,13) CW0: mod=64QAM tbs=4737 R=0.924 rv=2  t=185 us cfo=-16 tti_tx=5959
2022-05-23T09:32:47.052215 [PHY0-NR] [I] [ 5963] PDCCH: cc=0, c-rnti=0x4601 dci=0_0 ss=common0 L=2 cce=0 f_alloc=0x67 t_alloc=0x0 hop=n mcs=28 ndi=0 rv=3 harq_id=0 tpc=1 
2022-05-23T09:32:47.052221 [MAC    ] [I] [ 5963] UL 0:  Retx=2, rv=3, tbs=4737
2022-05-23T09:32:47.052444 [PHY0-NR] [I] [ 5963] PUSCH: cc=0 pid=0 c-rnti=0x4601 prb=(0,51) symb=(0,13) CW0: mod=64QAM tbs=4737 R=0.924 rv=3  t=176 us cfo=-15 tti_tx=5967
2022-05-23T09:32:47.060125 [PHY0-NR] [I] [ 5971] PDCCH: cc=0, c-rnti=0x4601 dci=0_0 ss=common0 L=2 cce=0 f_alloc=0x67 t_alloc=0x0 hop=n mcs=28 ndi=0 rv=1 harq_id=0 tpc=1 
2022-05-23T09:32:47.060131 [MAC    ] [I] [ 5971] UL 0:  Retx=3, rv=1, tbs=4737
2022-05-23T09:32:47.060353 [PHY0-NR] [I] [ 5971] PUSCH: cc=0 pid=0 c-rnti=0x4601 prb=(0,51) symb=(0,13) CW0: mod=64QAM tbs=4737 R=0.924 rv=1  t=179 us cfo=-14 tti_tx=5975
2022-05-23T09:32:47.068089 [PHY0-NR] [I] [ 5979] PDCCH: cc=0, c-rnti=0x4601 dci=0_0 ss=common0 L=2 cce=0 f_alloc=0x67 t_alloc=0x0 hop=n mcs=28 ndi=0 rv=0 harq_id=0 tpc=1 
2022-05-23T09:32:47.068095 [MAC    ] [I] [ 5979] UL 0:  Retx=4, rv=0, tbs=4737
2022-05-23T09:32:47.068320 [PHY0-NR] [I] [ 5979] PUSCH: cc=0 pid=0 c-rnti=0x4601 prb=(0,51) symb=(0,13) CW0: mod=64QAM tbs=4737 R=0.924 rv=0  t=181 us cfo=-12 tti_tx=5983
2022-05-23T09:32:47.076104 [PHY0-NR] [I] [ 5987] PDCCH: cc=0, c-rnti=0x4601 dci=0_0 ss=common0 L=2 cce=0 f_alloc=0x67 t_alloc=0x0 hop=n mcs=28 ndi=1 rv=0 harq_id=0 tpc=1 
2022-05-23T09:32:47.076109 [MAC    ] [I] [ 5987] UL 0:  Maximum number of ReTX reached (5). Discarding TB.
2022-05-23T09:32:47.076113 [RLC-NR ] [I] SRB1: Started t-PollRetransmit. POLL_SN=0
2022-05-23T09:32:47.076115 [RLC-NR ] [I] SRB1: [Data PDU, P=1, SI=full, SN_SIZE=12 bits, SN=0, SO=0]
2022-05-23T09:32:47.076116 [RLC-NR ] [I] SRB1: No data available to be sent
[... 12 lines cut ...]
2022-05-23T09:32:47.100417 [PHY0-NR] [I] [ 6011] PUSCH: cc=0 pid=0 c-rnti=0x4601 prb=(0,51) symb=(0,13) CW0: mod=64QAM tbs=4737 R=0.924 rv=1  t=177 us cfo=-7 tti_tx=6015
[... cut ...]
When starting, I first start the gNB using the command `sudo UHD_IMAGES_DIR=/usr/share/uhd/images srsenb` on the first PC. This gets a connection to the 5G Core network, and can communicate with the B210.
After that, I start the UE using the command `sudo UHD_IMAGES_DIR=/usr/share/uhd/images srsue` on the second PC. This can communicate with the B210, and looks like it can communicate with the gNB as well, up to a point. The UE is attaching, and it gets a connection on the RRC level (see the output in the previous section).

## Additional Information ##
I have the log files available if needed (they're rather big).

andrepuschmann, 2022-06-03:
Hey, can you share the logs files in info mode please until this happens? It's likely a hickup in the last messages that cause the UE do be in state without recovery. Note that the recovery and retransmission of NAS signaling, etc. isn't implemented yet

loktiing, 2022-06-13:
Hi, I am working on 5G SA End-to-End with USRP B205-mini's. Using the same enb.conf, rr.conf and ue.conf. However, there is no response on both sides. Here is the display by srsenb:

Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=uhd with args=type=b200,serial=3180ECB,clock=internal,sampling_rate=23.04e6,lo_freq_offset_hz=23.04e6,None
Supported RF device list: UHD file
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.0.0.HEAD-0-g90ce6062
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,serial=3180ECB,lo_freq_offset_hz=23.04e6,None=,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
NG connection successful
[INFO] [B200] Detected Device: B205mini
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=0 nof_prb=52

------------------------------------------------------------------------------------------------------
Here is the display by srsue:
Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=uhd with args=type=b200,serial=1710009,clock=internal,sampling_rate=23.04e6,lo_freq_offset_hz=23.04e6,None
Supported RF device list: UHD zmq file
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,serial=1710009,lo_freq_offset_hz=23.04e6,None=,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Loading firmware image: /usr/share/uhd/images/usrp_b200_fw.hex...
[INFO] [B200] Detected Device: B205mini
[INFO] [B200] Loading FPGA image: /usr/share/uhd/images/usrp_b205mini_fpga.bin...
[INFO] [B200] Operating over USB 2.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Waiting PHY to initialize ... done!
Attaching UE...
RF status: O=42, U=0, L=0
RF status: O=46, U=0, L=0
RF status: O=46, U=0, L=0

cat /tmp/srsue.log produces:
2022-06-13T07:50:32.242173 [UE     ] [I] Using binary srsue with arguments: 
2022-06-13T07:50:32.311232 [UE     ] [I] Built in Release mode using commit f2dff0b7a on branch master.
2022-06-13T07:51:40.649195 [NAS5G  ] [I] Switching on
2022-06-13T07:51:40.649205 [NAS5G  ] [D] Changed to mm5g state: DEREGISTERED, with substate PLMN-SEARCH
2022-06-13T07:51:40.657369 [NAS5G  ] [I] Requesting IMSI attach (IMSI=001010000000003)
2022-06-13T07:51:40.657380 [NAS5G  ] [D] Starting T3410. Timeout in 15000 ms.
2022-06-13T07:51:40.657382 [NAS5G  ] [I] Sending Registration Request
2022-06-13T07:51:40.657383 [NAS5G  ] [D] Initiating RRC NR Connection
2022-06-13T07:51:40.657406 [NAS5G  ] [D] Changed to mm5g state: REGISTERED-INITIATED
2022-06-13T07:51:41.183555 [RRC-NR ] [W] Could not finish setup request. Deallocating dedicatedInfoNAS PDU

What am I missing?

TallGuy74, 2022-06-13:
I since have this working; I use a 30 dB attenuation on the cable connection, and I had to use a tx_gain of 75 and rx_gain of 35 for the enb.conf:

```
[rf]
tx_gain = 75
rx_gain = 35
```

The same for the ue.conf. Your mileage may vary, it probably depends on your connections. You will also have to create a record for your SIM in the 5G Core Network (I used open5gs for this).

TallGuy74, 2022-06-14:
Using the latest agpl_next branch makes things a lot more resilient; Currently running 2x2 MIMO (using B210's with a direct connection from RF1 to RF1 and RF2 to RF2). I get 147 Mbps DL and 48.5 Mbps UL with PRB=100.

TallGuy74, 2022-06-14:
@loktiing Could you check that your CPU is running at full performance? I have to run `echo performance | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor` to get all my cores running near max performance.

loktiing, 2022-06-16:
Thanks @TallGuy74, I tried follow your suggestions, both sides still do not get any response. 

If I run srsenb and then srsue, /tmp/srsue.log produces:
```
2022-06-16T06:46:42.489228 [UE     ] [I] Using binary srsue with arguments: 
2022-06-16T06:46:42.559283 [UE     ] [I] Built in Release mode using commit f2dff0b7a on branch master.
2022-06-16T06:46:43.751262 [NAS5G  ] [I] Switching on
2022-06-16T06:46:43.751272 [NAS5G  ] [D] Changed to mm5g state: DEREGISTERED, with substate PLMN-SEARCH
2022-06-16T06:46:43.760040 [NAS5G  ] [I] Requesting IMSI attach (IMSI=001010000000003)
2022-06-16T06:46:43.760052 [NAS5G  ] [D] Starting T3410. Timeout in 15000 ms.
2022-06-16T06:46:43.760053 [NAS5G  ] [I] Sending Registration Request
2022-06-16T06:46:43.760054 [NAS5G  ] [D] Initiating RRC NR Connection
2022-06-16T06:46:43.760072 [NAS5G  ] [D] Changed to mm5g state: REGISTERED-INITIATED
2022-06-16T06:46:44.054785 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.056411 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.057526 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.058609 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.059709 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.060993 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.061901 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.062968 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.064139 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.065222 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.066328 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.067621 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.068525 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.069724 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.070908 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.071900 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.073218 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.074326 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.075358 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.076515 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.077786 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.078991 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.079912 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.081177 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.082398 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.083350 [PHY-SA ] [E] [ 4600] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.084769 [PHY0-NR] [E] [ 4600] PBCH-MIB: NR SFN (464) does not match current SFN (460)
2022-06-16T06:46:44.151815 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.152882 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.154109 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.155266 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.156260 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.157616 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.158387 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.159593 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.160843 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.161857 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.163042 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.164174 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.165219 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.166363 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.167629 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.168708 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.169887 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.171069 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.172271 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
2022-06-16T06:46:44.173267 [PHY-SA ] [E] [ 4610] SYNC: detected out-of-sync... skipping slot ...
[... cut ...]
2022-06-16T06:52:05.088096 [NAS5G  ] [I] Requesting IMSI attach (IMSI=001010000000003)
2022-06-16T06:52:05.088107 [NAS5G  ] [D] Starting T3410. Timeout in 15000 ms.
2022-06-16T06:52:05.088109 [NAS5G  ] [I] Sending Registration Request
2022-06-16T06:52:05.088110 [NAS5G  ] [D] Initiating RRC NR Connection
2022-06-16T06:52:05.088128 [NAS5G  ] [D] Changed to mm5g state: REGISTERED-INITIATED
2022-06-16T06:52:05.612221 [RRC-NR ] [W] Could not finish setup request. Deallocating dedicatedInfoNAS PDU
```

UE cannot seem to read ENB TX signal properly. Is it some synchronization issue?

loktiing, 2022-06-16:
The setup works if I used ZMQ Virtual Radios on both node. But not for USRP B205-mini's

TallGuy74, 2022-06-29:
I'm not sure if the B205 mini's use a different radio component (mostly the RF parts, with perhaps differing output power?).
I am running a similar setup with two USRP X300's, and haven't found the right gains for it yet.

TallGuy74, 2022-07-26:
@mariatsampazi 30 dB attenuation on the line with a tx_gain of 6 and an rx_gain of 0 worked with Amarisoft, and I think will work for srsue to srsenb as well.

I currently use 2 x 30 dB attenuation and tx_gain of 20 and rx_gain of 13.

TallGuy74, 2022-07-27:
Also, it looks like the clocks need to be really stable to make things work correctly. In my setup, if X300-1 is connected to the eNB, and X300-2 connected to the UE, things don't work correctly (connection is slow and unstable). If I switch X300-1 to UE, things work (30.5 Mbps DL and 18.9 Mbps UL for 5G SA SISO vs 9 Mbps DL and 18.9 Mbps UL with the non-working setup).

Using a GPSDO or an octaclock should help with that though.

mariatsampazi, 2022-09-14:
Hi @TallGuy74 , thanks for your answer. In the configuration files, as soon as you made it work with usrps b210, the only change to the ones you shared was the tx and rx gains, right?

nadiayoza, 2022-10-03:
Hi @mariatsampazi , @TallGuy74  I am also trying to connect two B210s through a 5G link. I can ping between gNB and UE, but the link only lasts for a few seconds/minutes. Does the same happen to you? I am using tx_gain=80, rx_gain=40 and a fixed attenuator of 30 dB connected to the TX port of the B210s. I am running Ubuntu 20.04 LTS and UHD 3.15-LTS on both.

mariatsampazi, 2022-10-04:
Yes, it has happened to me as well

To combat if I followed the steps mentioned here:
 https://www.youtube.com/watch?v=kRc-dR73uao&ab_channel=NeoKarugamo
 
 If the steps mentioned in the previous step don't help you can also try this:
For instance to deal with the overflows that I faced I used the numactl like this: `numactl --cpunodebind=netdev:usrp0 --membind=netdev:usrp0 srsenb`

nadiayoza, 2022-10-06:
Thanks for your response, @mariatsampazi I used the command you suggested and followed the steps mentioned on the video, including using an external clock reference, and the link is way more stable, especially with the lower MCS they use (with MCS=10, the connection lasts over one hour). Are you also using a lower MCS? With the maximum MCS, I'm getting only ~26 Mbps in DL and 12 Mbps in UL. How stable is the link in your case?  
Thanks!

mariatsampazi, 2022-10-07:
Hi @nadiayoza 

After following the steps mentioned above the link becomes more stable (I also use an MCS of 10). I have noticed that when the connection hangs the error is: `Scheduling request failed: releasing RRC connection...` Maybe you get this error?
I am investigating this and I will get back to you. Here is a link where they have addressed this, if it helps: https://github.com/srsran/srsRAN/issues/805

nadiayoza, 2022-10-07:
Hi @mariatsampazi 
Yes, I get these errors: RF status: O=0, U=5, L=0 
And then that message "Scheduling request failed: releasing RRC connection...". 
I am now running iperf using the --zerocopy option as suggested on https://github.com/srsran/srsRAN/issues/924 and, just in case, using a higher RRC inactivity timer (20000) https://github.com/srsran/srsRAN/issues/862 
With those new parameters and using the default MCS, the 5G SA connection lasted for over one hour and the DL and UL throughput were 29 Mbps and 17 Mbps, respectively, so at least it's getting better. I will keep checking!
Also, have you tried it with X300 or any other SDR?

mariatsampazi, 2022-10-07:
Yes, the RRC inactivity timer should be increased in the enb side.
Yes, I have tried both with B210 and X310 (I use the latter now).

nadiayoza, 2022-10-07:
Good to know! We currently don't have X310, but only N310. I will see if we can use it. 
Do you get a stable link and the expected throughput using X310?

whitecloudy, 2022-10-11:
I used MCS value 0 for UL and DL and now it works really well with USRP X300. I would recommend to use lowest MCS value if you only want to test the connection procedure.

nadiayoza, 2022-10-13:
Thanks, @whitecloudy Using B210 SDRs and a fixed MCS of 10, the connection lasts for almost 3 hours. That is significantly more stable than with the maximum MCS of 28. It's good to know that this works well with USRP X300, but unfortunately we don't have one. If you know of any other way to improve the stability and throughput using the B210, let me know.
Thanks!

Bhuvaneshnetcon, 2024-01-02:
Hi,
Here I am trying to connect open5gs core and UHD ettus usrp b210 sdr by srsran with my real UE.
Am facing this issue **Late: 108; Underflow: 39; Overflow: 10** when I run gnb config file . How can I rectify it. Kindly help me.
Even I receive **real time failure in RF ,FAPI, low-phy** in gnb.log using srsran and time differs between srsran installed machine and gnb.log file in srsran.

herlesupreeth, 2024-01-02:
@Bhuvaneshnetcon If you are seeing high number of Lates either your PC is not strong enough or the CPU governor is not put in performance mode.

Bhuvaneshnetcon, 2024-01-03:
Hi @herlesupreeth 
Thanks alot , but I already set cpu governor in performance mode. Then my PC have 8 GB memory ,intel i3 8th generation.
Even am facing the same issue.

herlesupreeth, 2024-01-03:
On Intel based machines I usually do the following steps to get maximum performance (resulting in less Lates and underflows)

```
Install cpufrequtils:

sudo apt-get install cpufrequtils

Then edit the following file (if it doesn't exist, create it):

sudo nano /etc/default/cpufrequtils

And add the following line to it:

GOVERNOR="performance"

Save and exit.

Now you need to disable ondemand daemon, otherwise after you reboot the settings will be overwritten.

sudo update-rc.d ondemand disable

sudo /etc/init.d/cpufrequtils restart

Power management
Remove all power management features in the BIOS (sleep states, in particularC-states) and CPU frequency scaling (Intel SpeedStep). In some cases, you can
also do this with cpufreqtool. Also, disable hyperthreading in BIOS and make sure its turned off in Linux

1. check this using this command: watch grep \"cpu MHz\" /proc/cpuinfo
2. you must disable p-state and c-state in linux so you need to add intel_pstate=disable to the Linux boot options,
i.e GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_pstate=disable"  in /etc/default/grub, then perform update-grub. You may optionally add the following as well "processor.max_cstate=1 intel_idle.max_cstate=0 idle=poll"

Append "blacklist intel_powerclamp" to the end of /etc/modprobe.d/blacklist.conf, to blacklist the intel_powerclamp" module. If the file does not exist, create one, and add the line into it.
You also need to disable hyperthreading, CPU frequency control, C-States, P-States and any other power management from BIOS as well.

Install i7z utility to check the cpu

sudo apt-get install i7z
sudo i7z

The CPU should not change its frequency by more than 1-2 hertz and should not be any C-state other than C0

```

Bhuvaneshnetcon, 2024-01-08:
Hi @herlesupreeth 
If I do the above steps, it throws error like there is no ondemand service.inspite of these I have one doubt is there any gpsdo or octoclock required for UHD usrp ettus b210 to achive 5g SA. Because there is no  gpsdo or octoclock in my environmrnt. I have also attached my gnb configuration file for your reference.
[gnb_rf_b200_tdd_n78_20mhz.docx](https://github.com/srsran/srsRAN_4G/files/13858737/gnb_rf_b200_tdd_n78_20mhz.docx)

herlesupreeth, 2024-01-08:
Notice in your logs above (screenshot) that USRP B210 is using USB2 rather than USB3. Please check the port to which you have connected the USB cable of B210.

Bhuvaneshnetcon, 2024-01-08:
Hi @herlesupreeth 
When I cange the USB2 port to USB3 , I got connectivity to UE but for very few sec only. Kindly help how to make it stable.

Bhuvaneshnetcon, 2024-01-10:
Hi @herlesupreeth 
**Is there gpsdo or octoclock is importanat to use with usrp B210?** Because my radio usrp B210 by srsran-LTE connected with open5gs 5gc successfully and UE get internet. When I try my radio usrp B210 by srsran-5g connected with open5gs 5gc, UE does not get internet(but 10.45.0.X ip assigned by UPF). could I know gpsdo or octoclock is responsible for internet ?

Bhuvaneshnetcon, 2024-01-10:
Hi
When I add clock=default , sync=default in gnb configuration , my UE get internet for minimal period of time.
ru_sdr:
  device_driver: uhd                                              
  device_args: type=b200,clock=default,sync=default,num_recv_frames=64,num_send_frames=64  
**What should i do to get persistent internet connection for UE?**

herlesupreeth, 2024-01-15:
Please post here the gNB logs at info log level in order to debug the issue

Bhuvaneshnetcon, 2024-01-16:
I have attached the gnb logs.

2024-01-08T10:29:38.478536 [FAPI    ] [W] [   281.8] Real-time failure in FAPI: Received late UL_TTI.request from slot 281.7
2024-01-08T10:29:38.580200 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.581078 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.581953 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.582819 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.583694 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.584570 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.585320 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.586193 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.587076 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.587694 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.588319 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.588571 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.588586 [RF      ] [W] Real-time failure in RF: overflow
2024-01-08T10:29:38.589070 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.589693 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.590445 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.591442 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.592318 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.593201 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.593206 [Low-PHY#0] [W] Real-time failure in low-phy: PRACH request late for sector 0, slot 281.19 and start symbol 0.
2024-01-08T10:29:38.594075 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.594951 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.595821 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.596693 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.597567 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.598443 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599067 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599568 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599956 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599957 [RF      ] [W] Real-time failure in RF: overflow
2024-01-08T10:29:38.600443 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.601354 [FAPI    ] [W] [   283.9] Real-time failure in FAPI: Received late UL_TTI.request from slot 283.7
2024-01-08T10:29:38.601367 [FAPI    ] [W] [  283.10] Real-time failure in FAPI: Received late UL_TTI.request from slot 283.8
2024-01-08T10:29:38.601375 [FAPI    ] [W] [  283.10] Real-time failure in FAPI: Received late UL_TTI.request from slot 283.9
2024-01-08T10:29:38.601445 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.601543 [FAPI    ] [W] [  283.14] Real-time failure in FAPI: Received late DL_TTI.request from slot 283.13
2024-01-08T10:29:38.601819 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.602318 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.602694 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.602819 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.603450 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.603827 [Low-PHY#0] [W] Real-time failure in low-phy: PRACH request late for sector 0, slot 282.19 and start symbol 0.
2024-01-08T10:29:38.604192 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.605075 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.605445 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.605946 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.606320 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.606448 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.607069 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.607818 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.608206 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.609067 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.610318 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.610697 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.610703 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.610942 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.610948 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.611324 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.611455 [RF      ] [W] Real-time failure in RF: overflow
2024-01-08T10:29:38.611819 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.612198 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.612616 [FAPI    ] [W] [   284.7] Real-time failure in FAPI: Received late DL_TTI.request from slot 284.6
2024-01-08T10:29:38.612692 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.612780 [FAPI    ] [W] [  284.11] Real-time failure in FAPI: Received late DL_TTI.request from slot 284.10
[... cut ...]
2024-01-08T10:29:38.612916 [SCHED   ] [W] [  284.17] cell=0, slot=284.12: Discarding error indication. Cause: Scheduler results associated with the slot of the error indication have already been erased
2024-01-08T10:29:38.612917 [SCHED   ] [W] [  284.17] cell=0, slot=284.13: Discarding error indication. Cause: Scheduler results associated with the slot of the error indication have already been erased
2024-01-08T10:29:38.613075 [RF      ] [W] Real-time failure in RF: late
2024-01-08T10:29:38.613451 [Low-PHY#0] [W] Real-time failure in low-phy: PRACH request late for sector 0, slot 283.19 and start symbol 0.

Bhuvaneshnetcon, 2024-01-17:
2024-01-08T10:29:38.478536 [FAPI    ] [W] [   281.8] Real-time failure in FAPI: Received late UL_TTI.request from slot 281.7
2024-01-08T10:29:38.580200 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.581078 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.581953 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.582819 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.583694 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.584570 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.585320 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.586193 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.587076 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.587694 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.588319 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.588571 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.588586 [RF      ] [W] Real-time failure in RF: overflow
2024-01-08T10:29:38.589070 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.589693 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.590445 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.591442 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.592318 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.593201 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.593206 [Low-PHY#0] [W] Real-time failure in low-phy: PRACH request late for sector 0, slot 281.19 and start symbol 0.
2024-01-08T10:29:38.594075 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.594951 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.595821 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.596693 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.597567 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.598443 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599067 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599568 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599956 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.599957 [RF      ] [W] Real-time failure in RF: overflow
2024-01-08T10:29:38.600443 [RF      ] [W] Real-time failure in RF: underflow
2024-01-08T10:29:38.601354 [FAPI    ] [W] [   283.9] Real-time failure in FAPI: Received late UL_TTI.request from slot 283.7
2024-01-08T10:29:38.601367 [FAPI    ] [W] [  283.10] Real-time failure in FAPI: Received late UL_TTI.request from slot 283.8
2024-01-08T10:29:38.601375 [FAPI    ] [W] [  283.10] Real-time failure in FAPI: Received late UL_TTI.request from slot 283.9
2024-01-08T10:29:38.601445 [RF      ] [W] Real-time failure in RF: late

herlesupreeth, 2024-01-17:
I would suggest benchmark your system connected to USRP B210 for the BW you are testing - https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio#Benchmarking_your_system

There should no dropped/overflows. If there are then I would suggest using more powerful machine to run gNB

xiongruoting, 2024-05-28:
Still not sloving the problem? I also got USRP B210 srsUE overflow problem..
RF status: O=21, U=0, L=0

meong3000, 2025-03-07:
> The setup works if I used ZMQ Virtual Radios on both node. But not for USRP B205-mini's

Hi @loktiing , did u solved the tx & rx gain configuration for the over-the-air setup? I'm also using b205-mini and have the same problem (stuck on RF overflows). Thanks in advance