srsran/srsRAN_4G #876: 5G NSA COST UE(Honor X10) failed to register network

coolcoollly, 2022-05-27:
## Issue Description ##
I run 5G NAS COST with the turorial(https://docs.srsran.com/en/latest/app_notes/source/5g_nsa_cots/source/index.html) .
And I run srsepc, srsenb successfully, but my phone(Honor X10) failed to connect the enb and epc. 
I can search my network operator(46070), but when I register in, my phone failed to register.

## Setup Details ##
PHOHE: Honor X10; USRP X310;
Ubuntu 18.04; UHD 3.15; SRSRAN 21.04

## Expected Behavior ##
Having run enb and epc, my phone can search my network operator  register in.(the same as I do when I run 4G COST UE)

## Actual Behaviour ##
Having run enb and epc, I can search my network operator. But when I register in, my phone failed to register.

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
[user_db.csv](https://github.com/srsran/srsRAN/files/8787674/user_db.csv)
[enb.txt](https://github.com/srsran/srsRAN/files/8787700/enb.txt)
[epc.txt](https://github.com/srsran/srsRAN/files/8787701/epc.txt)
[rr.txt](https://github.com/srsran/srsRAN/files/8787702/rr.txt)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
These are my srsepc and srsenb windows.
[image]
[image]

coolcoollly, 2022-06-04:
this is my enb.log
[enb.log](https://github.com/srsran/srsRAN/files/8837666/enb.log)

iman432, 2022-09-09:
Based on eNB logs, there are many disconnecting so connection between UE and eNB isn't stable, I have this problem when host resources isn't enough,  increase memory and add CPU(in case of VM) then see result.