srsran/srsRAN_4G #881: I tried installing srsRAN with ZMQ and I got Errors while running CTest

Avatarrunt, 2022-06-02:
## Issue Description ##
I tried installing srsRAN with ZMQ and I got Errors while running CTest

Errors while running CTest Output from these tests are in: /home/runt/srsRAN/build/Testing/Temporary/LastTest.log Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely. make: *** [Makefile:91: test] Error 8

## Setup Details ##
installing ZeroMQ and build srsRAN. On Ubuntu

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
99% tests passed, 6 tests failed out of 1545

Label Time Summary:
chest     =  33.16 sec*proc (22 tests)
fec       =   0.39 sec*proc (11 tests)
ldpc      =   0.01 sec*proc (1 test)
lib       = 204.61 sec*proc (1007 tests)
lte       = 168.01 sec*proc (942 tests)
nr        = 189.46 sec*proc (359 tests)
pdcp      =   0.43 sec*proc (3 tests)
phch      =  38.34 sec*proc (485 tests)
phy       = 179.64 sec*proc (1004 tests)
polar     =   0.00 sec*proc (1 test)
rlc       =  31.51 sec*proc (13 tests)
sched     =   0.83 sec*proc (7 tests)
srsenb    =   6.19 sec*proc (10 tests)
srsue     =   0.78 sec*proc (3 tests)
turbo     =   0.38 sec*proc (9 tests)

Total Test time (real) = 437.60 sec

The following tests FAILED:
	876 - benchmark_radio_multi_rf (Failed)
	1444 - rrc_meas_test (Failed)
	1445 - rrc_phy_ctrl_test (Failed)
	1446 - rrc_cell_test (Failed)
	1447 - rrc_rlf_report_test (Failed)
	1448 - usim_test (Failed)
Errors while running CTest
Output from these tests are in: /home/runt/srsRAN/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
make: *** [Makefile:91: test] Error 8

## Steps to reproduce the problem ##
# Install prerequisites
sudo apt-get -y install libzmq3-dev build-essential cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev libtool autoconf automake
# Install libzmq
git clone https://github.com/zeromq/libzmq.git
cd libzmq
./autogen.sh
./configure
make
sudo make install
sudo ldconfig
# Install czmq
git clone https://github.com/zeromq/czmq.git
cd czmq
./autogen.sh
./configure
make
sudo make install
sudo ldconfig
# Compile srsRAN
git clone https://github.com/srsRAN/srsRAN.git
cd srsRAN
mkdir build
cd build
cmake ../
[... 204 lines cut ...]
cpu family	: 6
model		: 140
model name	: 11th Gen Intel(R) Core(TM) i3-1125G4 @ 2.00GHz
stepping	: 1
microcode	: 0x88
cpu MHz		: 2000.000
cache size	: 8192 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 27
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 invpcid_single cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a avx512f avx512dq rdseed adx smap avx512ifma clflushopt clwb intel_pt avx512cd sha_ni avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req avx512vbmi umip pku ospke avx512_vbmi2 gfni vaes vpclmulqdq avx512_vnni avx512_bitalg avx512_vpopcntdq rdpid movdiri movdir64b fsrm avx512_vp2intersect md_clear flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple pml ept_mode_based_exec tsc_scaling
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs
bogomips	: 3993.60
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

andrepuschmann, 2022-06-03:
Hey, but what are the test errors? You don't provide the logs for the failed tests.

Avatarrunt, 2022-06-03:
I don't don't know how to do it

robertfalkenberg, 2022-06-14:
[image]

```
cd build
ctest --rerun-failed --output-on-failure
```

andrivet, 2022-07-26:
I get one of the errors mentioned (876 - benchmark_radio_multi_rf (Failed)) by following the same steps than described previously in this issue with the latest version of srsRAN. The details of the error are:

```
$ ctest --rerun-failed --output-on-failure
Test project /home/pi/SDR/srsRAN/BUILD
    Start 858: benchmark_radio_multi_rf
1/1 Test #858: benchmark_radio_multi_rf .........***Failed    1.28 sec
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins:
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2000,rx_port=tcp://localhost:2000
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2000
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2001
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2001
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2002,rx_port=tcp://localhost:2002
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2002
CH0 tx_port=tcp://*:2002
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2003,rx_port=tcp://localhost:2003
Error: allocating rx buffer
Error: initiating ringbuffer
[zmq] Error: Neither Tx port nor Rx port specified.
Error: Calling radio_multi constructor
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2003
CH0 tx_port=tcp://*:2003
Tearing down...
Failed!

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   1.41 sec

The following tests FAILED:
	858 - benchmark_radio_multi_rf (Failed)
Errors while running CTest
```

If it can help. I'm ignoring this error as it is the only one in my case and I have no issue with srsRAN.

robertfalkenberg, 2022-07-28:
The messages
```
Error: allocating rx buffer
Error: initiating ringbuffer
```
indicate failed memory buffer allocation.

Please ensure you are not running out of memory.

Avatarrunt, 2022-08-14:
> The messages
> 
> ```
> Error: allocating rx buffer
> Error: initiating ringbuffer
> ```
> 
> indicate failed memory buffer allocation.
> 
> Please ensure you are not running out of memory.

I have 8GB, do I need more

robertfalkenberg, 2022-08-14:
Then could you please share the output of following shell commands?

```sh
ulimit -a
```
and
```sh
free -h
```

robertfalkenberg, 2022-09-15:
The `benchmark_radio` test (as well as srsUE and srsENB) attempt to keep all current and future memory pages in RAM via `mlockall(MCL_CURRENT | MCL_FUTURE)` to prevent swapping if possible.
However, the maximum size of the locked memory (`RLIMIT_MEMLOCK`) that an unprivileged process is allowed to occupy can be restricted.
You can check the limit (`max locked memory`) with `ulimit -a`, which is
* 65536kB on Ubuntu Server 20.04 LTS (RPi)
* 115860kB on Ubuntu Server 22.04 LTS (RPi)

The call of `mlockall` is done during application startup after allocating the majority of internal buffers, but not the (large) buffers that are associated with the particular RF device (in this case ZMQ).
If `mlockall` fails because the already reserved memory cannot be locked, the application will continue without locking (at the risk of performance degradation).
In this case you will see an error message "mlockall" (for benchmark_radio) or "Failed to 'mlockall': {}" (for srsUE and srsENB).

However, if `mlockall` is successful, future memory allocations may fail with `ENOMEM` if this would exceed the `RLIMIT_MEMLOCK`.
That seems to be exactly the case here.

If there is sufficient RAM on the system, the limit can be waived entirely by adding the following line to `/etc/security/limits.conf`:
```
*               -       memlock         unlimited
```

Otherwise, lower the limit to a smaller value, e.g. 65536, so that `mlockall` fails to continue without locking:

```
*               -       memlock         65536

```

andrivet, 2022-09-16:
I will perform more tests and let you know the results (with detailed information)

andrivet, 2022-10-15:
I can confirm that the failure was indeed due to `max locked memory` like suggested by  @robertfalkenberg.

Here is my test with latest version of srsRAN and no config change:

```
$ ctest --rerun-failed --output-on-failure
Test project /home/pi/SDR/srsRAN/BUILD
    Start 857: benchmark_radio_multi_rf
1/1 Test #857: benchmark_radio_multi_rf .........***Failed    1.25 sec
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Instantiating objects and allocating memory...
Initialising instances...
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2000,rx_port=tcp://localhost:2000
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2000
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2001
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2001
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2002,rx_port=tcp://localhost:2002
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2002
CH0 tx_port=tcp://*:2002
Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2003,rx_port=tcp://localhost:2003
Error: allocating rx buffer
Error: initiating ringbuffer
[zmq] Error: Neither Tx port nor Rx port specified.
Error: Calling radio_multi constructor
Supported RF device list: UHD zmq file
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2003
CH0 tx_port=tcp://*:2003
Tearing down...
Failed!

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   1.39 sec

The following tests FAILED:
        857 - benchmark_radio_multi_rf (Failed)
Errors while running CTest
```

In this config, `max locked memory` is limited to less than 1 GiB:

```
 ulimit -a
real-time non-blocking time  (microseconds, -R) unlimited
core file size              (blocks, -c) 0
data seg size               (kbytes, -d) unlimited
scheduling priority                 (-e) 0
file size                   (blocks, -f) unlimited
pending signals                     (-i) 30154
max locked memory           (kbytes, -l) 1006368
max memory size             (kbytes, -m) unlimited
open files                          (-n) 1024
pipe size                (512 bytes, -p) 8
POSIX message queues         (bytes, -q) 819200
real-time priority                  (-r) 0
stack size                  (kbytes, -s) 8192
cpu time                   (seconds, -t) unlimited
max user processes                  (-u) 30154
virtual memory              (kbytes, -v) unlimited
file locks                          (-x) unlimited
```

After setting `max locked memory` to unlimited, the same test is now successful:

```
$ ctest --rerun-failed --output-on-failure
Test project /home/pi/SDR/srsRAN/BUILD
    Start 857: benchmark_radio_multi_rf
1/1 Test #857: benchmark_radio_multi_rf .........   Passed   12.58 sec

100% tests passed, 0 tests failed out of 1

Total Test time (real) =  13.00 sec
```

```
$ ulimit -l
unlimited
```

I see that the [documentation was already updated](https://docs.srsran.com/en/latest/app_notes/source/pi4/source/index.html) so I think that this issue can be closed.

robertfalkenberg, 2022-10-17:
Happy to hear that. Thanks for the detailed feedback.

closed 2022-10-17 by robertfalkenberg