srsran/srsRAN_4G #885: COTS UE not finding the LTE network

Usman-Josh, 2022-06-06:
I recently set-up an LTE network  to connect to COTS UE, I changed the MCC and MNC values in both the EPC and ENB config files 
 in accordance to the SIM card being used in the UE ,and also the APN name in the EPC file and set the APN name on the COTS UE the same as the set name on the EPC config file, I started the EPC and ENB, but the two user equipment with SIM cards of the same mobile network that I have tried have not been able to at least detect the network, could the issue be because of the authentication parameters like IMSI, K, and other unknown values that I have not been able to find, that is required in the user_db.csv file? 
and how do I get these values because it seems like recent versions of operating systems of phones have blocked the viewing of IMSI of SIM cards being used? 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config fi#322

andrepuschmann, 2022-06-06:
What SIM cards are you using? If you are trying to use cards from a commercial network then that is not going to work because you don't get the keys from the card.

Usman-Josh, 2022-06-09:
> http://shop.sysmocom.de/products/sysmoISIM-SJA2.

Thank you, would a writable programmable sim card be used too?

Usman-Josh, 2022-06-09:
> http://shop.sysmocom.de/products/sysmoISIM-SJA2.

Thank you, could a writable programmable sim card be used too?

fllay, 2022-06-10:
I used this one.  
https://www.aliexpress.com/item/32900944064.html?spm=a2g0s.9042311.0.0.57ad4c4dEW1tZ9

you have to ask for software from the seller. 

Both sysmocom and Chinese sims work fine, but I prefer sysmocom since I don't have to worry about how to generate those keys.

Usman-Josh, 2022-06-20:
> I used this one. https://www.aliexpress.com/item/32900944064.html?spm=a2g0s.9042311.0.0.57ad4c4dEW1tZ9
> 
> you have to ask for software from the seller.
with a sim card reader would I be able to get the authentication values from a commercial sim card?

Usman-Josh, 2022-06-20:
what if you want to build a web app for the whole BTS process where instead of writing the commands on a terminal, you click (buttons) on the web app and your enode B and EPC start running, also values like IMSI,MCC,MNC can be changed on the web app interface, is there such?

robertfalkenberg, 2022-06-22:
> with a sim card reader would I be able to get the authentication values from a commercial sim card?
Probably no. Typically, those values are write-only (for a reason).

robertfalkenberg, 2022-06-22:
> what if you want to build a web app for the whole BTS process where instead of writing the commands on a terminal, you click (buttons) on the web app and your enode B and EPC start running, also values like IMSI,MCC,MNC can be changed on the web app interface, is there such?

A web-based configuration interface is currently not planned.

Usman-Josh, 2022-06-23:
> A web-based configuration interface is currently not planned.

But is it possible

Usman-Josh, 2022-06-23:
> But is it possible

?

brendan-mcauliffe, 2022-07-26:
In theory, this is possible if you are willing to create it yourself, although this is not something we can provide information on. 

For future support with issues that are not specific to the code itself, please use the mailing list. You can sign up for it [here](https://lists.srsran.com/mailman/listinfo/srsran-users). 

Closing this issue.

closed 2022-07-26 by brendan-mcauliffe