srsran/srsRAN_4G #886: 5G COTS UE Success Registration but No Connection

peanutButter782, 2022-06-06:
## Issue Description ##
I am attempting to establish a 5G SA connection between a Samsung S21 and srsRAN, using Open5gs as the core network.  The UE registers successfully, and viewed thru wireshark, appears to setup the PDU session.  However, the UE never receives an IP address, and continually displays the No Signal icon instead of the 5G icon.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
Ubuntu 18.04
srsRAN 22.04 (also tried the "agpl_next" branch per similar issue #882, same outcome) 
Open5gs core network
Same issue using either B210 or X300 USRP
UHD 4.2
srsRAN and open5gs on the same machine, such that everything is communicating on the loopback addresses

## Expected Behavior ##
I expected the COTS UE to connect successfully to srsRAN, and get assigned an IP address.  (This UE has done this successfully with srsRAN in NSA mode.)  I also expected the UE to have internet access.

## Actual Behaviour ##
The UE registers successfully with srsRAN/Open5gs, but no IP address is assigned.  The UE continually displays the No Signal icon instead of the 5G icon (with some bars).

## Steps to reproduce the problem ##
Attaching the configuration and log files. (renamed *.conf and *.yaml to *.txt)
[amf.txt](https://github.com/srsran/srsRAN/files/8845734/amf.txt)
[enb.log](https://github.com/srsran/srsRAN/files/8845735/enb.log)
[enb.txt](https://github.com/srsran/srsRAN/files/8845736/enb.txt)
[rr.txt](https://github.com/srsran/srsRAN/files/8845737/rr.txt)

Attaching the pcap.
[s21_00101.pcapng.txt](https://github.com/srsran/srsRAN/files/8845741/s21_00101.pcapng.txt)

Attaching the trace from the eNB.
[trace.txt](https://github.com/srsran/srsRAN/files/8845749/trace.txt)

## Additional Information ##
Any advice is muchly appreciated.

andrepuschmann, 2022-06-15:
From the eNB logs it seems the core isn't responding after sending the 
```
2022-06-06T15:18:43.343607 [NGAP   ] [I] Tx NGAP SDU, PDUSessionResourceSetupResponse, rnti=0x4601
```
I am not sure why this is the case though

andrepuschmann, 2022-06-15:
I confirm this is the same issue as #882 . Have a look at the logs in the core, something is there that the core doesn't like.

coolcoollly, 2022-06-15:
> 

Hey I have the same problem. I am very suspicious that my phone(Honor X10 ) have the problem to connect to the gnb, so I want to buy a phone used in the the tutorial (https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html). Do you have more information about Oneplus Nord 5G? Is Oneplus Nord 5G like the picture?
[image]

iman432, 2022-10-04:
> Hey I have the same problem. I am very suspicious that my phone(Honor X10 ) have the problem to connect to the gnb, so I want to buy a phone used in the the tutorial [...]

Hi,

Did you solve this issue? i have this issue with Samsung S21 and Huawei Mate 40 pro , Registration and PDU establishment is successful but no user plane data, just UE sends service request.

yzucyh, 2023-06-19:
Hi @iman432 , 
have you do any solution later? If you fixed the problem, could you please share the way you deal with the problem?
Thank's

iman432, 2023-06-19:
> Hi @iman432 , have you do any solution later? If you fixed the problem, could you please share the way you deal with the problem? Thank's

At new version of srsRAN project, this issue is solved.

https://github.com/srsran/srsran_project

closed 2023-06-19 by ismagom