srsran/srsRAN_4G #888: Segmentation fault with large UE capability messages

davwheat, 2022-06-08:
## Issue Description ##
Hey! I'm trying to customise srsRAN to send custom UE capability requests to determine the carrier aggregation capabilities of various LTE/NR devices.

I've managed to do so, to the point where COTS UEs are reliably interpreting and responding to the requests, but at some point afterwards, srsRAN encounters a segmentation fault, which I assume is due to the large capability message from the UE.

Where could I go about fixing this? Presumably by increasing the size of some buffer somewhere within the source of srsRAN, but I'm not even sure where to look.

If you'd like to see the changes I have made: https://github.com/srsran/srsRAN/compare/master...davwheat:srsRAN-22.04

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

- bladeRF micro 2.0 xA5
- srsRAN eNB + Open5GS core
- Ubuntu 20.04

## Error

```
Stack trace (most recent call last) in thread 38647:
#9    Object "", at 0xffffffffffffffff, in 
#8    Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 95, in __clone [0x7f48e77ae132]
#7    Source "/build/glibc-SzIz7B/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7f48e7fe6608]
#6    Object "/usr/local/bin/srsenb", at 0x564694e12eb4, in srsran::thread::thread_function_entry(void*)
#5    Object "/usr/local/bin/srsenb", at 0x564694e60d8c, in srsenb::enb_stack_lte::run_thread()
#4    Object "/usr/local/bin/srsenb", at 0x564694f16579, in srsenb::rrc::tti_clock()
#3    Object "/usr/local/bin/srsenb", at 0x564694f0fc8f, in srsenb::rrc::rem_user(unsigned short)
#2    Object "/usr/local/bin/srsenb", at 0x564694eae7a3, in srsenb::mac::ue_rem(unsigned short)
#1    Object "/usr/local/bin/srsenb", at 0x564694ec4374, in srsenb::sched::ue_rem(unsigned short)
#0    Object "/usr/local/bin/srsenb", at 0x564694ecb6ef, in std::_Deque_base<srsran::dl_sch_lcid, srsran::cached_alloc<srsran::dl_sch_lcid> >::~_Deque_base()
Segmentation fault (Address not mapped to object [0xc00000010])
Segmentation fault
```

andrepuschmann, 2022-06-14:
Hey,
hard to say. The segfault seems in the `srsenb::sched::ue_rem` though, not when handling the UE caps. Can you share the workload and under what situation it happens. Right from the start? After a while?

davwheat, 2022-06-17:
Sorry about the delay in getting back to you.

I've recently swapped out my RF-customised Lubuntu distro with a stock Ubuntu 20.04 install and haven't noticed any issues since, but I've also made some code changes.

At the moment, I'm instead trying to work out why srsRAN seems to not be sending signalling reliably, but that's in my attempts to rewrite the native bladeRF driver to work with 2x2 MIMO since the SoapySDR has a number of issues when attempting to run the bladeRF as reported in #504 and #552.

closed 2022-06-17 by davwheat