srsran/srsRAN_4G #890: 5G SA COTS UE srsRAN crashed: terminate called after throwing an instance of 'std::out_of_range'

cristina-posada, 2022-06-13:
## Issue Description ##
Following the 5G SA COTS UE [Application Note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html), after running srsENB with `sudo ./srsenb enb.conf` the UE successfully connects and gets an assigned IP and access to the internet. But after a few seconds or a few minutes the srsENB stops due to the following error: **terminate called after throwing an instance of 'std::out_of_range'
 what():  map::at**

## Setup Details ##
- Operation System: Ubuntu 20.04
- srsRAN master
- RF: USRP B210 (UHD driver)
_____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   serial: 30DCED4
|   |   name: MyB210
|   |   product: 2
|   |   revision: 4
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: gps_gpgga, gps_gprmc, gps_time, gps_locked, gps_servo, ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 1
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX2
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: B
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B210 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
[... 101 lines cut ...]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa3f7) [0x7fa3867393f7]
	/home/cristina/srsRAN/build/lib/src/phy/rf/libsrsran_rf_uhd.so(+0x19c58) [0x7fa3861d7c58]
	/home/cristina/srsRAN/build/lib/src/phy/rf/libsrsran_rf_uhd.so(+0x44e62) [0x7fa386202e62]
	/usr/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0xce) [0x7fa386379fde]
	/home/cristina/srsRAN/build/lib/src/phy/rf/libsrsran_rf_uhd.so(+0x117b7) [0x7fa3861cf7b7]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---
[image]

[image]

## Steps to reproduce the problem ##
To reproduce this issue follow the guide https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html
Execute srsenb `sudo ./srsenb enb.conf` try to connect the UE to the open5GS net. And copy the files:
- srsRAN Config files: enb.conf, rr.conf, sib.conf
- Open5GS config files: amf.yaml, upf.yaml
[amf.txt](https://github.com
[upf.txt](https://github.com/srsran/srsRAN/files/8891089/upf.txt)
/srsran/srsRAN/files/8891087/amf.txt)
[enb.txt](https://github.com/srsran/srsRAN/files/8891091/enb.txt)
[sib.txt](https://github.com/srsran/srsRAN/files/8891901/sib.txt)
[rr.txt](https://github.com/srsran/srsRAN/files/8891903/rr.txt)

## Logs ##
[enb.log](https://github.com/srsran/srsRAN/files/8891978/enb.log)

andrepuschmann, 2022-06-15:
Hey @cristina-posada, thanks for the report. The issue has already been identified. Can you give https://github.com/srsran/srsRAN/tree/agpl_next a try and see if this fixes the issue for you?
Thanks

cristina-posada, 2022-06-21:
> Hey @cristina-posada, thanks for the report. The issue has already been identified. Can you give https://github.com/srsran/srsRAN/tree/agpl_next a try and see if this fixes the issue for you? Thanks

Hi @andrepuschmann Thanks for the reply! I have tried that branch and there is no crash but the UE disconnects from the network after a few seconds/minutes. 

The ping from PC to UE IP stop having a reply and the UE stops having acces to the internet.

srsenb output:
[image]

ping to the UE:
[image]

kaposnick, 2022-06-23:
> Hey @cristina-posada, thanks for the report. The issue has already been identified. Can you give https://github.com/srsran/srsRAN/tree/agpl_next a try and see if this fixes the issue for you? Thanks

I encounter the same problem, as mentioned, both in release_22_04 ([f2dff0b7a](https://github.com/srsran/srsRAN/commit/f2dff0b7a03e065060b48ea0a3c7a6a7890dfd46)), but also in ([6a3b925](https://github.com/srsran/srsRAN/commit/6a3b9257e33948977bedab71ce75e6746d843de2)).

The scenario is as following:
srsUE connectes to srsENB, pings for some period. Then, it waits till the RRC Connection is released. Then tries to ping again but without error:
`Scheduling request failed: releasing RRC connection... `

**srsENB output**
[image]

**srsUE logs**
[image]

(I upload the logs, for further analysis)
I cite, the part of the logs where the BS starts the UEContextReleaseRequest towards the AMF. After a while, it removes the user rnti.
**srsENB logs**
```
2022-06-23T20:53:42.281838 [NGAP   ] [I] Tx NGAP SDU, UEContextReleaseRequest, rnti=0x4601
2022-06-23T20:53:42.286126 [NGAP   ] [D] Rx - UEContextReleaseCommand (21 B)
2022-06-23T20:53:42.286132 [NGAP   ] [I] Started UE Context Release
2022-06-23T20:53:42.286159 [RRC-NR ] [D] rnti=0x4601, SRB1 - Tx DL-DCCH.rrcRelease (2 B)
2022-06-23T20:53:42.286208 [NGAP   ] [I] Tx NGAP SDU, UEContextReleaseComplete, rnti=0x4601
2022-06-23T20:53:42.360173 [RRC-NR ] [D] Activity registered for rnti=0x4601 (timeout_value=30000ms)
2022-06-23T20:53:42.360440 [RRC-NR ] [I] Removed user rnti=0x4601
2022-06-23T20:53:42.374530 [MAC-NR ] [W] [  786] SCHED: "ul_crc_info" called for unknown rnti=0x4601,cc=0.
2022-06-23T20:53:42.375018 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:42.375019 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:42.375020 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:42.375020 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:42.375021 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:42.375022 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:42.375022 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:42.375023 [MAC-NR ] [W] [  788] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.670478 [RRC-NR ] [D] Setting timer for Msg3 reception for rnti=0x4602 to 100ms
2022-06-23T20:53:49.670482 [RRC-NR ] [D] Activity registered for rnti=0x4602 (timeout_value=100ms)
2022-06-23T20:53:49.670510 [RRC-NR ] [I] Added new user rnti=0x4602
2022-06-23T20:53:49.683692 [RRC-NR ] [I] rnti=0x4602 received MAC CRNTI CE: 0x4601, but old context is unavailable
2022-06-23T20:53:49.683757 [RRC-NR ] [I] Removed user rnti=0x4602
2022-06-23T20:53:49.689057 [MAC-NR ] [W] [ 6903] SCHED: "ul_sr_info" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689059 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689060 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689061 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689062 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689063 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689064 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689064 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.689065 [MAC-NR ] [W] [ 6903] SCHED: "ul_bsr" called for unknown rnti=0x4601.
2022-06-23T20:53:49.861440 [RRC-NR ] [D] Setting timer for Msg3 reception for rnti=0x4603 to 100ms
2022-06-23T20:53:49.861443 [RRC-NR ] [D] Activity registered for rnti=0x4603 (timeout_value=100ms)
2022-06-23T20:53:49.861469 [RRC-NR ] [I] Added new user rnti=0x4603
2022-06-23T20:53:49.980723 [RRC-NR ] [I] Activity timer for rnti=0x4603 expired after 100 ms
2022-06-23T20:53:49.980808 [RRC-NR ] [I] Rem
[logs.zip](https://github.com/srsran/srsRAN/files/8971029/logs.zip)
oved user rnti=0x4603
2022-06-23T20:53:50.055405 [RRC-NR ] [D] Setting timer for Msg3 reception for rnti=0x4604 to 100ms
2022-06-23T20:53:50.055408 [RRC-NR ] [D] Activity registered for rnti=0x4604 (timeout_value=100ms)
2022-06-23T20:53:50.055432 [RRC-NR ] [I] Added new user rnti=0x4604
```

**srsUE logs**
```
2022-06-23T20:53:42.299458 [RRC-NR ] [D] SRB1 - Rx rrcRelease (2 B)
2022-06-23T20:53:42.299471 [RRC-NR ] [I] RRC Release not handled yet
2022-06-23T20:53:49.756286 [MAC-NR ] [W] [ 6960] Contention Resolution Timer expired. Stopping PDCCH Search and going to Response Error (TODO)
2022-06-23T20:53:49.885897 [MAC-NR ] [W] [ 7069] RAR Timer expired. RA response not received within the response window
2022-06-23T20:53:50.077887 [MAC-NR ] [W] [ 7229] RAR Timer expired. RA response not received within the response window
2022-06-23T20:53:50.269741 [MAC-NR ] [W] [ 7389] RAR Timer expired. RA response not received within the response window
2022-06-23T20:53:50.457234 [MAC-NR ] [W] [ 7549] RAR Timer expired. RA response not received within the response window
2022-06-23T20:53:50.647215 [MAC-NR ] [W] [ 7709] RAR Timer expired. RA response not received within the response window
[... 7 lines cut ...]
[... cut ...]
2022-06-23T20:54:03.240994 [MAC-NR ] [W] [ 7869] RAR Timer expired. RA response not received within the response window
2022-06-23T20:54:03.240996 [MAC-NR ] [W] [ 7869] Maximum number of transmissions reached (7)
2022-06-23T20:54:06.949633 [NAS    ] [I] Switching off
2022-06-23T20:54:06.949638 [NAS    ] [D] Received request to detach in state NULL
2022-06-23T20:54:06.949639 [NAS    ] [D] Clearing EPS bearer context
```

(As a temporary solution, increasing the activity timer does fixe the issue)

cristina-posada, 2022-06-27:
Hello! 
I set a larger RRC inactivity timer in the enb.conf and even though the connection now lasts 5 or 8 minutes, it still fails after that time. I would like to keep the connection stable for longer.

`RACH:  slot=8971, cc=0, preamble=6, offset=3, temp_crnti=0x4602
User 0x4602 requesting RRC Reestablishment as 0x4601. Cause: otherFailure
Disconnecting rnti=0x4601.
RACH:  slot=9131, cc=0, preamble=6, offset=3, temp_crnti=0x4603
Disconnecting rnti=0x4603.
RACH:  slot=9611, cc=0, preamble=4, offset=3, temp_crnti=0x4604
User 0x4604 requesting RRC Reestablishment as 0x4602. Cause: reconfigurationFailure
Disconnecting rnti=0x4602.
RACH:  slot=9771, cc=0, preamble=5, offset=3, temp_crnti=0x4605
Disconnecting rnti=0x4605.
RACH:  slot=9931, cc=0, preamble=6, offset=3, temp_crnti=0x4606
Disconnecting rnti=0x4606.
RACH:  slot=10091, cc=0, preamble=3, offset=3, temp_crnti=0x4607
Disconnecting rnti=0x4607.
RACH:  slot=11, cc=0, preamble=2, offset=3, temp_crnti=0x4608
Disconnecting rnti=0x4608.
RACH:  slot=171, cc=0, preamble=4, offset=3, temp_crnti=0x4609
Disconnecting rnti=0x4609.
RACH:  slot=331, cc=0, preamble=3, offset=3, temp_crnti=0x460a
Disconnecting rnti=0x460a.
RACH:  slot=491, cc=0, preamble=7, offset=3, temp_crnti=0x460b
Disconnecting rnti=0x460b.
RACH:  slot=651, cc=0, preamble=7, offset=3, temp_crnti=0x460c
Disconnecting rnti=0x460c.
RACH:  slot=1131, cc=0, preamble=7, offset=3, temp_crnti=0x460d
User 0x460d requesting RRC Reestablishment as 0x4604. Cause: otherFailure
Disconnecting rnti=0x4604.
RACH:  slot=1291, cc=0, preamble=2, offset=3, temp_crnti=0x460e
Disconnecting rnti=0x460e.
RACH:  slot=1451, cc=0, preamble=3, offset=3, temp_crnti=0x460f
Disconnecting rnti=0x460f.
RACH:  slot=1611, cc=0, preamble=1, offset=3, temp_crnti=0x4610
Disconnecting rnti=0x4610.
RACH:  slot=1771, cc=0, preamble=3, offset=3, temp_crnti=0x4611
Disconnecting rnti=0x4611.
RACH:  slot=1931, cc=0, preamble=1, offset=3, temp_crnti=0x4612
Disconnecting rnti=0x4612.
RACH:  slot=2091, cc=0, preamble=7, offset=3, temp_crnti=0x4613
Disconnecting rnti=0x4613.
RACH:  slot=2251, cc=0, preamble=3, offset=3, temp_crnti=0x4614
Disconnecting rnti=0x4614.
RACH:  slot=2411, cc=0, preamble=7, offset=3, temp_crnti=0x4615
Disconnecting rnti=0x4615.
RACH:  slot=2891, cc=0, preamble=5, offset=3, temp_crnti=0x4616
User 0x4616 requesting RRC Reestablishment as 0x460d. Cause: otherFailure
Disconnecting rnti=0x460d.
RACH:  slot=3051, cc=0, preamble=0, offset=3, temp_crnti=0x4617
Disconnecting rnti=0x4617.
RACH:  slot=3211, cc=0, preamble=6, offset=3, temp_crnti=0x4618
Disconnecting rnti=0x4618.
RACH:  slot=3371, cc=0, preamble=4, offset=5, temp_crnti=0x4619
Disconnecting rnti=0x4619.
RACH:  slot=3531, cc=0, preamble=5, offset=3, temp_crnti=0x461a
Disconnecting rnti=0x461a.
RACH:  slot=3691, cc=0, preamble=0, offset=3, temp_crnti=0x461b
Disconnecting rnti=0x461b.
RACH:  slot=3851, cc=0, preamble=0, offset=3, temp_crnti=0x461c
Disconnecting rnti=0x461c.
RACH:  slot=4011, cc=0, preamble=6, offset=3, temp_crnti=0x461d
Disconnecting rnti=0x461d.
RACH:  slot=4171, cc=0, preamble=1, offset=3, temp_crnti=0x461e
Disconnecting rnti=0x461e.
RACH:  slot=4651, cc=0, preamble=0, offset=3, temp_crnti=0x461f
User 0x461f requesting RRC Reestablishment as 0x4616. Cause: otherFailure
Disconnecting rnti=0x4616.
RACH:  slot=4811, cc=0, preamble=0, offset=3, temp_crnti=0x4620
Disconnecting rnti=0x4620.
RACH:  slot=4971, cc=0, preamble=4, offset=3, temp_crnti=0x4621
Disconnecting rnti=0x4621.
RACH:  slot=5131, cc=0, preamble=4, offset=3, temp_crnti=0x4622
Disconnecting rnti=0x4622.
RACH:  slot=5291, cc=0, preamble=0, offset=5, temp_crnti=0x4623
Disconnecting rnti=0x4623.
RACH:  slot=5451, cc=0, preamble=3, offset=5, temp_crnti=0x4624
Disconnecting rnti=0x4624.
RACH:  slot=5611, cc=0, preamble=0, offset=5, temp_crnti=0x4625
Disconnecting rnti=0x4625.
RACH:  slot=5771, cc=0, preamble=2, offset=5, temp_crnti=0x4626
Disconnecting rnti=0x4626.
RACH:  slot=5931, cc=0, preamble=3, offset=3, temp_crnti=0x4627
Disconnecting rnti=0x4627.
`
Built in Release mode using 22.04.0. (srsRAN-agpl_next)

cristina-posada, 2022-06-28:
> Hello! I set a larger RRC inactivity timer (1073741823) in the enb.conf and even though the connection now lasts 8 or 10 minutes, it still fails after that time. I would like to keep the connection stable for longer.
> 
> Built in Release mode using 22.04.0. (srsRAN-agpl_next)
> 
> ==== eNodeB started === Type to view trace Setting manual TX/RX offset to 100 samples Setting frequency: DL=2655.0 Mhz, UL=2535.0 MHz for cc_idx=0 nof_prb=52 [INFO] [UHD RF] Tx while waiting for EOB, timed out... 232.918 >= 230.124. Starting new burst... RACH: slot=9131, cc=0, preamble=1, offset=3, temp_crnti=0x4601 RACH: slot=8971, cc=0, preamble=6, offset=3, temp_crnti=0x4602 User 0x4602 requesting RRC Reestablishment as 0x4601. Cause: otherFailure Disconnecting rnti=0x4601. RACH: slot=9131, cc=0, preamble=6, offset=3, temp_crnti=0x4603 Disconnecting rnti=0x4603. RACH: slot=9611, cc=0, preamble=4, offset=3, temp_crnti=0x4604 **User 0x4604 requesting RRC Reestablishment as 0x4602. Cause: reconfigurationFailure** Disconnecting rnti=0x4602. RACH: slot=9771, cc=0, preamble=5, offset=3, temp_crnti=0x4605 Disconnecting rnti=0x4605. RACH: slot=9931, cc=0, preamble=6, offset=3, temp_crnti=0x4606 Disconnecting rnti=0x4606. RACH: slot=10091, cc=0, preamble=3, offset=3, temp_crnti=0x4607 Disconnecting rnti=0x4607. RACH: slot=11, cc=0, preamble=2, offset=3, temp_crnti=0x4608 Disconnecting rnti=0x4608. RACH: slot=171, cc=0, preamble=4, offset=3, temp_crnti=0x4609 Disconnecting rnti=0x4609. RACH: slot=331, cc=0, preamble=3, offset=3, temp_crnti=0x460a Disconnecting rnti=0x460a. RACH: slot=491, cc=0, preamble=7, offset=3, temp_crnti=0x460b Disconnecting rnti=0x460b. RACH: slot=651, cc=0, preamble=7, offset=3, temp_crnti=0x460c Disconnecting rnti=0x460c. RACH: slot=1131, cc=0, preamble=7, offset=3, temp_crnti=0x460d User 0x460d requesting RRC Reestablishment as 0x4604. Cause: otherFailure Disconnecting rnti=0x4604. RACH: slot=1291, cc=0, preamble=2, offset=3, temp_crnti=0x460e Disconnecting rnti=0x460e. RACH: slot=1451, cc=0, preamble=3, offset=3, temp_crnti=0x460f Disconnecting rnti=0x460f. RACH: slot=1611, cc=0, preamble=1, offset=3, temp_crnti=0x4610 Disconnecting rnti=0x4610. RACH: slot=1771, cc=0, preamble=3, offset=3, temp_crnti=0x4611 Disconnecting rnti=0x4611. RACH: slot=1931, cc=0, preamble=1, offset=3, temp_crnti=0x4612 Disconnecting rnti=0x4612. RACH: slot=2091, cc=0, preamble=7, offset=3, temp_crnti=0x4613 Disconnecting rnti=0x4613. RACH: slot=2251, cc=0, preamble=3, offset=3, temp_crnti=0x4614 Disconnecting rnti=0x4614. RACH: slot=2411, cc=0, preamble=7, offset=3, temp_crnti=0x4615 Disconnecting rnti=0x4615. RACH: slot=2891, cc=0, preamble=5, offset=3, temp_crnti=0x4616 User 0x4616 requesting RRC Reestablishment as 0x460d. Cause: otherFailure Disconnecting rnti=0x460d. RACH: slot=3051, cc=0, preamble=0, offset=3, temp_crnti=0x4617 Disconnecting rnti=0x4617. RACH: slot=3211, cc=0, preamble=6, offset=3, temp_crnti=0x4618 Disconnecting rnti=0x4618. RACH: slot=3371, cc=0, preamble=4, offset=5, temp_crnti=0x4619 Disconnecting rnti=0x4619. RACH: slot=3531, cc=0, preamble=5, offset=3, temp_crnti=0x461a Disconnecting rnti=0x461a. RACH: slot=3691, cc=0, preamble=0, offset=3, temp_crnti=0x461b Disconnecting rnti=0x461b. RACH: slot=3851, cc=0, preamble=0, offset=3, temp_crnti=0x461c Disconnecting rnti=0x461c. RACH: slot=4011, cc=0, preamble=6, offset=3, temp_crnti=0x461d Disconnecting rnti=0x461d. RACH: slot=4171, cc=0, preamble=1, offset=3, temp_crnti=0x461e Disconnecting rnti=0x461e. RACH: slot=4651, cc=0, preamble=0, offset=3, temp_crnti=0x461f User 0x461f requesting RRC Reestablishment as 0x4616. Cause: otherFailure Disconnecting rnti=0x4616. RACH: slot=4811, cc=0, preamble=0, offset=3, temp_crnti=0x4620 Disconnecting rnti=0x4620. RACH: slot=4971, cc=0, preamble=4, offset=3, temp_crnti=0x4621 Disconnecting rnti=0x4621. RACH: slot=5131, cc=0, preamble=4, offset=3, temp_crnti=0x4622 Disconnecting rnti=0x4622. RACH: slot=5291, cc=0, preamble=0, offset=5, temp_crnti=0x4623 Disconnecting rnti=0x4623. RACH: slot=5451, cc=0, preamble=3, offset=5, temp_crnti=0x4624 Disconnecting rnti=0x4624. RACH: slot=5611, cc=0, preamble=0, offset=5, temp_crnti=0x4625 Disconnecting rnti=0x4625. RACH: slot=5771, cc=0, preamble=2, offset=5, temp_crnti=0x4626 Disconnecting rnti=0x4626. RACH: slot=5931, cc=0, preamble=3, offset=3, temp_crnti=0x4627 Disconnecting rnti=0x4627. ...

If I activate in the UE the airplane mode and deactivate it again, the UE reconnects without any problems with the network and stays connected for a few minutes, until it disconnects again and so on.

cristina-posada, 2022-06-28:
If I activate in the UE the airplane mode and deactivate it again, the UE reconnects without any problems with the network and stays connected for a few minutes, until it disconnects again and so on.

closed 2022-06-30 by cristina-posada