srsran/srsRAN_4G #892: srsue cannot reconnect after disconnection

mshshokouhi, 2022-06-14:
Hi
I'm researching on lte. I want to use zmq virtual radio to simulate a ue connection.
## Issue Description ##
I disconnect srsue from srsenb by Ctrl+C after connection was established. then I try to reconnect srsue to srsenb but the connection does not establish. 
## Setup Details ##
Using last version of srsran (tag: release_22_04) on Ubuntu 20.04.3 LTS 
First I ran the srsepc:
```
sudo srsepc
```
Then srsenb:
```
srsenb --rf.device_name=zmq --rf.device_args="fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6"
```
and at the end srsue:
```
sudo srsue --rf.device_name=zmq --rf.device_args="tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6" --gw.netns=ue1
```
Terminal output:
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit 2fb99fe69 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.2 KHz
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=42, tti=181, ra-rnti=0x2
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=0
Network attach successful. IP: 172.16.0.3
Software Radio Systems RAN (srsRAN) 14/6/2022 13:19:54 TZ:0
^CStopping ..
Received RRC Connection Release (releaseCause: other)
RRC IDLE
Couldn't stop after 5s. Forcing exit.
Saving NAS PCAP file (DLT=148) to  
Killed
```
the connection without any problem establishes but after disconnection by Ctrl+C at the srsue terminal srsue can not reconnect to srsenb by same command:
```
sudo srsue --rf.device_name=zmq --rf.device_args="tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6" --gw.netns=ue1
```
Terminal output:
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Couldn't open , trying /root/.config/srsran/ue.conf
Reading configuration file /root/.config/srsran/ue.conf...

Built in Release mode using commit 2fb99fe69 on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Waiting PHY to initialize ... done!
Attaching UE...
```
## Expected Behavior ##
Reconnection ...
## Actual Behaviour ##
no connection

andrepuschmann, 2022-06-15:
Hey, answer to this is simple: it's not supported to connect the UE again as both eNB and UE RF are not free running.

closed 2022-06-15 by andrepuschmann

dziugas-wg2, 2023-06-18:
I spent some time debugging inability for srsue to reconnect to srsenb until I came across this issue. At the network socket level I see that both upon an initial and subsequent launch of srsue, two TCP connections are established between srsue and srsenb and in the opposite direction (using a configuration of a single `rx_port` and `tx_port`). However, only upon initial TCP session establishment, ZMQ data frames are exchanges in both directions. @andrepuschmann Could you elaborate a bit more on what you meant by the fact that eNB and UE RF are not free running and do you think it could be possible to get around the limitation and resume transmission the data sample in a form of ZMQ data frames?