srsran/srsRAN_4G #893: Unable to get agpl_next branch running

sfguerrero, 2022-06-14:
## Issue Description ##
Running enb from the get go returns an error regarding SRB configuration. Recently I read that some of my troubles could be fixed by switching to this branch. By installing the same way as before on master (clone, checkout, make, make install, configs installation) I noticed that a srsgnb folder appeared, which doesnt seem to get installed on usr/local/bin even though the flag for that is set as ON on Cmakelist. 

## Setup Details ##
I7 10700, Ubuntu 20.04, USRP b205mini-i, Samsung S21 Ultra 5G

## Expected Behavior ##
gnb starting 

## Actual Behaviour ##
Reading configuration file enb_USRP.conf...

Built in Release mode using commit 6a3b9257e on branch agpl_next.

Optional 5G SRB configuration is not supported yet.
Please specify 5G SRB1 and SRB2 configuration.
Error parsing RB configuration
Error processing arguments.

## Steps to reproduce the problem ##
SDR connected, run sudo ./srsenb enb_USRP.conf , get error shown above.

## Additional Information ##
gnb was starting before changing branch
[rbconf.txt](https://github.com/srsr
[enbconf.txt](https://github.com/srsran/srsRAN/files/8904282/enbconf.txt)
an/srsRAN/files/8904275/rbconf.txt)
usr/local/bin folder after installing new branch:
[image]

TallGuy74, 2022-06-15:
Update your rb.conf from the default/example configuration. I believe removing ~/.config/srsran/rb.conf and running `srsran_install_configs.sh user` will fix it for you.

TallGuy74, 2022-06-15:
Also, make sure to run the same version of srsran on both UE and gNB.

andrepuschmann, 2022-06-15:
On top of what @TallGuy74 already suggested: In general always make sure to update/compare your configs with the example ones provided by a specific release. Although we try to minimize the number of changes they are unavoidable when new features are added.

TallGuy74, 2022-06-15:
@andrepuschmann Would it be an option to create a document which contains the (breaking) changes to configuration files? I saw the commit with the srb changes to rb.conf, so I knew about it when I ran into it. Documenting things like this will make it easier for people upgrading.

sfguerrero, 2022-06-15:
Thanks, It worked properly thanks to Tallguy74's suggestion. Im closing the issue

closed 2022-06-15 by sfguerrero