srsran/srsRAN_4G #894: 5G SA UE 5 MHz Bandwidth?

Clay-Uhing-APL, 2022-06-16:
## Issue Description ##
The appnote specifies that the 5G SA UE is limited to 10 MHz. Is it only able to support 10 MHz, or can it also support 5 MHz?

## Setup Details ##
Changed working 10 MHz 5G SA gNB Amarisoft configuration to 5 MHz (To do this, I also had to change the n_rb_coreset0 to 24 and the n_symb_coreset0 to 2).

## Expected Behavior ##
srsUE connects to 5 MHz 5G SA gNB

## Actual Behaviour ##
running srsue ue.conf hangs on "Attaching UE..."
srsue.log has line
RRC-NR [E] Proc "Cell Selection" - Unsupported SSB offset 0
(Amarisoft shows that the SSB still has the same SSB ARFCN of 368410)

## Steps to reproduce the problem ##
Run 5 MHz 5G SA gNB and try to attach srs 5G SA UE

tecandrew, 2022-06-17:
wondering and seeing this error as well. Currently testing with a UHD B210 running for 5 Mhz. I had some luck running 10 MHz, but it struggles to keep up.

rfquickjr, 2022-07-05:
I found the error message in https://github.com/srsran/srsRAN/blob/master/srsue/src/stack/rrc_nr/rrc_nr_procedures.cc. The code is certainly interesting (this is the current master branch).

  // Check unsupported settings
  if (mib.cell_barred) {
    Error("Cell barred");
    return proc_outcome_t::error;
  }
  if (mib.scs_common != srsran_subcarrier_spacing_15kHz) {
    Error("Unsupported SCS %s", srsran_subcarrier_spacing_to_str(mib.scs_common));
    return proc_outcome_t::error;
  }
  // TODO: calculate SSB offset
  if (mib.ssb_offset != 6) {
    Error("Unsupported SSB offset %d", mib.ssb_offset);
    return proc_outcome_t::error;
  }

The first error is the correct behavior for a barred cell. The second says (surprisingly) that srsUE doesn’t support 30 kHz common scs. The third says only kssb = 6 will work.

With 100 kHz channel spacing, I find no reference frequency that produces kssb = 6 with 15 kHz SSB scs. With 5 MHz bandwidth 30 kHz SSB scs would give only 11 PRBs, not enough to hold SSB (needs 20). Bands with 15 kHz channel raster don't seem to support 5 MHz bandwidth. Conclusion: 5 MHz bandwidth can’t work with the current configuration of srsUE.