srsran/srsRAN_4G #898: Compile errors when building from source on Fedora 36

Ottega, 2022-06-19:
## Issue Description ##
Trying to install srsRAN from source on Fedora 36 but I run into errors

## Setup Details ##
Fedora 36
gcc version Red Hat 12.1.1-1
cmake version 3.22.2

SoapySDR + PlutoSRD

## Steps to reproduce the problem ##
I'm following the [build instructions](https://docs.srsran.com/en/latest/general/source/1_installation.html)
I have installed the required libraries, `cmake` runs fine but when I run `make` this error pops up
```
[  0%] Building CXX object lib/src/asn1/CMakeFiles/srsran_asn1.dir/liblte_common.cc.o
[  0%] Building CXX object lib/src/asn1/CMakeFiles/srsran_asn1.dir/liblte_mme.cc.o
[  0%] Building CXX object lib/src/asn1/CMakeFiles/srsran_asn1.dir/gtpc.cc.o
[  0%] Linking CXX static library libsrsran_asn1.a
[  0%] Built target srsran_asn1
[  0%] Building CXX object lib/src/support/CMakeFiles/support.dir/emergency_handlers.cc.o
In file included from /home/ottega/src/srsRAN/lib/include/srsran/srslog/bundled/fmt/format.h:44,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/bundled/fmt/ostream.h:13,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/bundled/fmt/printf.h:14,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/detail/log_backend.h:25,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/log_channel.h:25,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/logger.h:25,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/srslog.h:26,
                 from /home/ottega/src/srsRAN/lib/include/srsran/support/srsran_assert.h:26,
                 from /home/ottega/src/srsRAN/lib/src/support/emergency_handlers.cc:24:
/home/ottega/src/srsRAN/lib/include/srsran/srslog/bundled/fmt/core.h:1307:44: error: field ‘value’ has incomplete type ‘std::array<char, 256>’
 1307 |     std::array<char, max_pool_string_size> value;
      |                                            ^~~~~
In file included from /usr/include/c++/12/bits/unique_ptr.h:36,
                 from /usr/include/c++/12/memory:76,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/detail/support/any.h:25,
                 from /home/ottega/src/srsRAN/lib/include/srsran/srslog/srslog.h:25:
/usr/include/c++/12/tuple:1595:45: note: declaration of ‘struct std::array<char, 256>’
 1595 |   template<typename _Tp, size_t _Nm> struct array;
      |                                             ^~~~~
make[2]: *** [lib/src/support/CMakeFiles/support.dir/build.make:76: lib/src/support/CMakeFiles/support.dir/emergency_handlers.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7962: lib/src/support/CMakeFiles/support.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
``` 
Any idea how to resolve that issue? 

Best,
Ottega

robertfalkenberg, 2022-06-22:
Hi Ottega,
the issue is related to GCC 12.1.
Please try to apply the patches (0001..0005) included in https://aur.archlinux.org/packages/srsran-git

robertfalkenberg, 2022-06-22:
You could also try this branch, which already includes the GCC 12.1 patches: https://github.com/srsran/srsRAN/tree/agpl_next

Ottega, 2022-06-22:
After changing the branch it compiled without issues. Thank you.

closed 2022-06-22 by Ottega