srsran/srsRAN_4G #900: COTS UE cannot find LTE network with USRP x310

niloysh, 2022-06-21:
## Issue Description ##
COTS UE fails to find the LTE network. 

## Setup Details ##

**Operating System**: Ubuntu 18.04.6 LTS x86_64
**Kernel**: 5.4.0-120-lowlatency
**CPU**: Intel i7-6770HQ (4) @ 2.600GHz
**RAM**: 15898MiB
**RF Front End**: USRP x310 with UBX 160MHz daughterboard
**UHD Version**: UHD_4.1.0.5-0-g6bd0be9c
**srsRAN Version**: 22.04

We are following the latest srsRAN documentation as given [here.](https://docs.srsran.com/en/latest/getting_started.html)
We changed the MNC and MCC values as mentioned [here](https://docs.srsran.com/en/latest/usermanuals/source/srsenb/source/3_enb_trouble.html#cots-ue-issues) and also tried different `dl_earfcn` values including the default value of `3350`.

## Expected Behavior ##
srsRAN LTE network should show up when scanning using different COTS UE.

## Actual Behaviour ##
We do not see the network, tried with Samsung A52, Google Pixel 4a 5G, Iphone Pro Max 13.

## Steps to reproduce the problem ##

We included the following line in the `enb.conf` file as per the srsRAN docs.
`device_args=type=x300,sampling_rate=11.52e6,lo_freq_offset_hz=11.52e6`.

Rest are the default configuration given, with the exception of changing MCC and MNC.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

**epc output**
```
Built in RelWithDebInfo mode using commit f2dff0b7a on branch HEAD.

---  Software Radio Systems EPC  ---

Couldn't open , trying /home/n6saha/.config/srsran/epc.conf
Reading configuration file /home/n6saha/.config/srsran/epc.conf...
Couldn't open user_db.csv, trying /home/n6saha/.config/srsran/user_db.csv
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf302, MNC: 0xf720
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:302, MNC:720
S1 Setup Request - TAC 7, B-PLMN 0x37202
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
SCTP Association Shutdown. Association: 70
Deleting eNB context. eNB Id: 0x19b
Releasing UEs context
No UEs to be released
Stopping ..
```

**enb output**
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_soapy.so libsrsran_rf_zmq.so
Inactive RF plugins:
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /home/n6saha/.config/srsran/enb.conf
Reading configuration file /home/n6saha/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /home/n6saha/.config/srsran/sib.conf
Couldn't open rr.conf, trying /home/n6saha/.config/srsran/rr.conf
Couldn't open rb.conf, trying /home/n6saha/.config/srsran/rb.conf

Built in RelWithDebInfo mode using commit f2dff0b7a on branch HEAD.

Opening 1 channels in RF device=default with args=type=x300,sampling_rate=11.52e6,lo_freq_offset_hz=11.52e6
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.1.0.5-0-g6bd0be9c
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
Supported RF device list: UHD soapy zmq file
Trying to open RF device 'UHD'
Opening USRP channels=1, args: type=x300,lo_freq_offset_hz=11.52e6,master_clock_rate=184.32e6
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
Stopping ..
```

robertfalkenberg, 2022-06-22:
Have you checked, whether the RF signal can be received by the UE?

> If signals are too weak or malformed then a UE may not successfully receive them and will not attempt to attach. The [gr-fosphor tool](https://github.com/osmocom/gr-fosphor) is a very useful SDR spectrum analyzer which can be used to check the properties of transmitted RF signals.

Other tools, such as [gqrx](https://gqrx.dk/) or [sdr++](https://github.com/AlexandreRouma/SDRPlusPlus) can be used as well to verify a proper emission of the signal.

robertfalkenberg, 2022-06-23:
Thank you. Please also share your config files.

niloysh, 2022-06-23:
Hi Robert,
Please find attached the config files.
[enb.txt](https://github.com/srsran/srsRAN/files/8963747/enb.txt)
[epc.txt](https://github.com/srsran/srsRAN/files/8963748/epc.txt)
[ue.txt](https://github.com/srsran/srsRAN/files/8963749/ue.txt)

Thanks

robertfalkenberg, 2022-06-23:
What is the MCC/MNC part of the IMSI of the SIM card you use in the COTS devices?

robertfalkenberg, 2022-06-23:
And: What kind of SIM card do you use in the COTS devices?

niloysh, 2022-06-23:
The MCC/MNC is the same in both the IMSI and config files, i.e., 302 and 720.

The SIM we are using is a testing SIM provided by the operator - we have access to the SIM parameters that we put in the `user_db.csv` file, including encryption, OPC keys, etc.

Apart from the testing SIM, we also have other commercial SIMs from this operator (same MCC/MNC) which we tried. I understand that these would not authenticate to the EPC, but is it possible to see the srsRAN network in the mobile operators list with these SIMs?

robertfalkenberg, 2022-06-23:
Can the UE see other public networks (including public 302/720) during network search?

niloysh, 2022-06-23:
Yes, it does. It picks up the public 302/720 as well as a few other 302/MNC networks.

niloysh, 2022-06-23:
Assuming this is a problem with the SIM, what is the expected behaviour with commercial SIMs provided the MCC/MNC of the SIM are the same as in the config file. Can we expect the network to show up during network search? Or does the SIM params need to be present in the `user_db.csv` file for the network to show up at all?

robertfalkenberg, 2022-06-23:
> Yes, it does. It picks up the public 302/720 as well as a few other 302/MNC networks.
In this case you will have to use a SIM card with a different IMSI with MCC/MNC part that is not used in the proximity.

Possibly, the UE finds the public 302/720 network first and ignores further findings of the same MCC/MNC combination.
Moreover, when powering up, the UE will most probably first search the frequency of the last successful attach.

niloysh, 2022-06-27:
Thanks for the help - this solved the issue.
In case anyone else has the same problems, here are the steps to address this.

- Some UE's cannot see testing MCC/MNC e.g., 999/70. Thus, make the MCC/MNC value a valid one, preferabley one that works in your country. **Important**: Make sure the MNC value is different than the public ones which are used in the proximity.
- Play around with the EARFCN values. We found that `3050` worked well on most of the COTS UEs we had, while the default value of `3350` did not show up.

closed 2022-06-27 by niloysh