srsran/srsRAN_4G #904: COTS UE can't see the 5G SA signal

yc541, 2022-06-24:
## Issue Description ##
COTS 5G UE can't see the 5G SA gNB signal

## Setup Details ##
I'm trying to run 5G SA with Open5GS (v2.4.5), SRSRAN is on an i7 laptop and Open5GS is on a VM. Radio is X310 and UHD version is 4.1. COTS UEs are a Oneplus Nord 5G and a Samsung A52. All these have no problems when running LTE and 5G NSA. The UEs are about 2 meters from the antennas.

## Expected Behavior ##
COTS UEs detect the 5G SA network

## Actual Behaviour ##
COTS UEs can't find the 5G SA network in cell search

## Steps to reproduce the problem ##
The conf files are attached

[amf.txt](https://github.com/srsran/srsRAN/files/8977257/amf.txt)
[enb.txt](https://github.com/srsran/srsRAN/files/8977258/enb.txt)
[rr.txt](https://github.com/srsran/srsRAN/files/8977261/rr.txt)
[upf.txt](https://github.com/srsran/srsRAN/files/8977262/upf.txt)

After running the gNB the NG connection is successful and the AMF is connected to the gNB, but neither the Nord 5G nor the Samsung can see the network. When configuring SRS to LTE or 5G NSA modes both UEs attach and have Internet. I have also rooted the Samsung and forcing 5G NR with NSG but had no luck. Tried putting 5G SA on N3 and N7, also tried swapping the radio with a B205 mini but the results was the same. The results with master branch and apgl_next are the same.

## Additional Information ##
Are there anything else I can do or are there any problems with my configurations? Any help is appreciated!

andrepuschmann, 2022-06-30:
It's likely a setting in the SIM or phone carrier settings. Search the list or Github for similar issues that already have been discussed. We work on a section in the docs for that.

helloTkk, 2022-09-11:
Hi @yc541,

I'm also using OnePlue Nord 5G and I encountered the same problem as you, did you figure it out?

Any help I would appreciate!

mayukhVT, 2022-09-20:
Hi, 

I am facing same issue with Samsung A90. Is there any update on this?

andrepuschmann, 2022-09-21:
I really suggest everyone to strictly follow [this](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html#preparing-cots-ue) guide. We've had issues with stock phones before but that worked for all of them, at least OnePlus models. I am closing the issue as the guide contains all information we have.

closed 2022-09-21 by andrepuschmann

helloTkk, 2022-10-11:
Hi,

I will try to re-program my USIM to enable those 5G related fields, if I can solve the problem, I will let you know. If you have any solutions, could you please also inform me?

Thanks

你通常不会收到来自 ***@***.*** 的电子邮件。了解这一点为什么很重要<https://aka.ms/LearnAboutSenderIdentification>

ntutanivia, 2023-02-20:
Hi @yc541 @andrepuschmann ,
I'm a student from Taiwan, currently working on 5G SA with Open5GS with Device b210. The problem I face is that I fail to connect to NG connection when I sudo srsenb (I've change the conf), it seems that there are some conflict on open5gs-amf.conf and enb.conf. Can someone show their enb.conf and amf.conf & mme.conf.
Thank's for your reply !

caleb18719516, 2023-07-11:
HI @ntutanivia  ,
I could help you about your problem.