srsran/srsRAN_4G #905: COST UE DL NAS Connection Reject

beruysaliz, 2022-06-28:
## Issue Description ##
I set up COST UE connection I am not sure if it works, but in srsenb terminal I see connected message, but in srsepc there is message "DL NAS : Sending PDN connectivity reject". I don't know what that means. 
### **srsepc**
[image]
### **srsenb**
[image]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
UBUNTU 20.04, USRP B210, UHD version 3.15.0, USB Dongle as UE, sysmoUSIM-SJS1 

## Expected Behavior ##
want to see output as in the COST UE tutorial

## Actual Behaviour ##
DL NAS : Sending PDN connectivity reject

## Steps to reproduce the problem ##
config files I created as in the tutorial of course configured IMSI,Key values on the USIM being used.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

beruysaliz, 2022-07-01:
I try again to connect ue and this is the epc logs.
[epc.log](https://github.com/srsran/srsRAN/files/9027865/epc.log)

pricalinat, 2023-02-19:
I have encountered the same problem, has this problem been solved?