srsran/srsRAN_4G #907: srsenb crashes when using config contains gui = true

muaddib1984, 2022-06-30:
## Issue Description ##
when ```gui = true ``` in the enb.conf file as soon as the srsue is launched the gNB crashes

## Setup Details ##
Ubuntu21.04
ZMQ sources
Followed the docs guide for 5G SA Step-By-Step
Installed Open5G Core and configured per instructions (matched local addresses)
srsue and srsenb are installed system-wide and the install_configs script was run with 'service' argument. conf files are in /etc/srsran
srsue and srsenb run correctly without the GUI enabled

## Expected Behavior ##
GUI window to open

## Actual Behaviour ##
When UE Connects to gNB the srsenb application crashes with the following error:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
--- command='srsenb' version=22.04.0 signal=6 date='30/06/2022 02:50:45' ---
	srsenb(+0x386f57) [0x560d9954af57]
	/lib/x86_64-linux-gnu/libc.so.6(+0x41040) [0x7f54fd0de040]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f54fd0ddfbb]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x116) [0x7f54fd0c3864]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xa1a31) [0x7f54fd493a31]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xad4fc) [0x7f54fd49f4fc]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xad567) [0x7f54fd49f567]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xad809) [0x7f54fd49f809]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xa4452) [0x7f54fd496452]
	srsenb(+0xcdd98) [0x560d99291d98]
	srsenb(+0xcddc2) [0x560d99291dc2]
	srsenb(+0x9e166) [0x560d99262166]
	/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xd5) [0x7f54fd0c5565]
	srsenb(+0xa0b3e) [0x560d99264b3e]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---
/home/user/srsran/lib/include/srsran/adt/pool/obj_pool.h:73: Assertion Failure: Not all objects have been deallocated (107 < 128)
--- command='srsenb' version=22.04.0 signal=6 date='30/06/2022 02:50:45' ---
	srsenb(+0x386f57) [0x560d9954af57]
	/lib/x86_64-linux-gnu/libc.so.6(+0x41040) [0x7f54fd0de040]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f54fd0ddfbb]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x116) [0x7f54fd0c3864]
	srsenb(+0xc0b23) [0x560d99284b23]
	srsenb(+0x31e6d8) [0x560d994e26d8]
	srsenb(+0x30485e) [0x560d994c885e]
	/lib/x86_64-linux-gnu/libc.so.6(+0x44147) [0x7f54fd0e1147]
	/lib/x86_64-linux-gnu/libc.so.6(on_exit+0) [0x7f54fd0e12f0]
	srsenb(+0x386f8d) [0x560d9954af8d]
	/lib/x86_64-linux-gnu/libc.so.6(+0x41040) [0x7f54fd0de040]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f54fd0ddfbb]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x116) [0x7f54fd0c3864]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xa1a31) [0x7f54fd493a31]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xad4fc) [0x7f54fd49f4fc]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xad567) [0x7f54fd49f567]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xad809) [0x7f54fd49f809]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xa4452) [0x7f54fd496452]
	srsenb(+0xcdd98) [0x560d99291d98]
	srsenb(+0xcddc2) [0x560d99291dc2]
	srsenb(+0x9e166) [0x560d99262166]
	/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xd5) [0x7f54fd0c5565]
	srsenb(+0xa0b3e) [0x560d99264b3e]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---
--- command='srsenb' version=22.04.0 signal=11 date='30/06/2022 02:50:45' ---
	srsenb(+0x386f57) [0x560d9954af57]
	/lib/x86_64-linux-gnu/libc.so.6(+0x41040) [0x7f54fd0de040]
	srsenb(+0xe0cda) [0x560d992a4cda]
	srsenb(+0xaaf7d) [0x560d9926ef7d]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x9450) [0x7f54fd9b6450]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f54fd1b4d53]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---
--- command='srsenb' version=22.04.0 signal=11 date='30/06/2022 02:50:45' ---
	srsenb(+0x386f57) [0x560d9954af57]
	/lib/x86_64-linux-gnu/libc.so.6(+0x41040) [0x7f54fd0de040]
	/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0x50) [0x7f54fd0e1800]
	/lib/x86_64-linux-gnu/libboost_program_options.so.1.74.0(+0x16ca7) [0x7f54fd953ca7]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---
```

## Steps to reproduce the problem ##
Install srsGUI
Install srsRAN
enable gui in config
```
[gui]
enable = true
```

andrepuschmann, 2022-06-30:
Thanks for reporting. We have a look on this shortly.

muaddib1984, 2022-07-15:
Hi @andrepuschmann, had any time to investigate this issue?