srsran/srsRAN_4G #910: 5G SA COTS UE Disconnecting after a few minutes.

cristina-posada, 2022-06-30:
## Issue Description ##
Following the 5G SA COTS UE [Application Note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html), after running srsENB with sudo ./srsenb enb.conf the UE successfully connects to the open5gs net and gets an assigned IP and access to the internet.
But after a few seconds or minutes the UE disconnects from the network. 

## Setup Details ##

- Operation System: Ubuntu 20.04
- srsRAN (srsRAN-agpl_next)
- RF: USRP B210 (UHD driver)
- Asus ZenBook intel Core i7
- UE: Galaxy S21 FE 5G Sysmocom USIM
- Third Party 5G Core: Open5GS

## Expected Behavior ##
The UE should be connected to the enb indefinitely

## Actual Behaviour ##
After a few seconds or minutes the UE disconnects from the network, the ping from PC to UE stop having a reply and the UE stops having acces to the internet. 

srsenb Output:

```
==== eNodeB started ===
Type <t> to view trace
Setting manual TX/RX offset to 100 samples
Setting frequency: DL=1847.5 Mhz, UL=1752.5 MHz for cc_idx=0 nof_prb=52
RACH:  slot=6251, cc=0, preamble=1, offset=3, temp_crnti=0x4601
RACH:  slot=5291, cc=0, preamble=2, offset=1477, temp_crnti=0x4602
Disconnecting rnti=0x4602.
RACH:  slot=5451, cc=0, preamble=6, offset=1477, temp_crnti=0x4603
Disconnecting rnti=0x4603.
RACH:  slot=5611, cc=0, preamble=4, offset=1477, temp_crnti=0x4604
Disconnecting rnti=0x4604.
RACH:  slot=5771, cc=0, preamble=1, offset=1477, temp_crnti=0x4605
Disconnecting rnti=0x4605.
RACH:  slot=5931, cc=0, preamble=7, offset=1477, temp_crnti=0x4606
Disconnecting rnti=0x4606.
RACH:  slot=6091, cc=0, preamble=4, offset=1477, temp_crnti=0x4607
Disconnecting rnti=0x4607.
RACH:  slot=6251, cc=0, preamble=2, offset=1475, temp_crnti=0x4608
Disconnecting rnti=0x4608.
RACH:  slot=6731, cc=0, preamble=6, offset=3, temp_crnti=0x4609
User 0x4609 requesting RRC Reestablishment as 0x4601. Cause: otherFailure
Disconnecting rnti=0x4601.
RACH:  slot=6891, cc=0, preamble=3, offset=3, temp_crnti=0x460a
Disconnecting rnti=0x460a.
RACH:  slot=7371, cc=0, preamble=4, offset=3, temp_crnti=0x460b
User 0x460b requesting RRC Reestablishment as 0x4609. Cause: reconfigurationFailure
Disconnecting rnti=0x4609.
RACH:  slot=7531, cc=0, preamble=4, offset=3, temp_crnti=0x460c
Disconnecting rnti=0x460c.
RACH:  slot=7691, cc=0, preamble=1, offset=3, temp_crnti=0x460d
Disconnecting rnti=0x460d.
RACH:  slot=7851, cc=0, preamble=6, offset=3, temp_crnti=0x460e
Disconnecting rnti=0x460e.
RACH:  slot=8011, cc=0, preamble=2, offset=5, temp_crnti=0x460f
Disconnecting rnti=0x460f.
RACH:  slot=8171, cc=0, preamble=2, offset=3, temp_crnti=0x4610
Disconnecting rnti=0x4610.
RACH:  slot=8331, cc=0, preamble=2, offset=3, temp_crnti=0x4611
Disconnecting rnti=0x4611.
RACH:  slot=8491, cc=0, preamble=0, offset=3, temp_crnti=0x4612
Disconnecting rnti=0x4612.
RACH:  slot=8651, cc=0, preamble=1, offset=3, temp_crnti=0x4613
Disconnecting rnti=0x4613.
RACH:  slot=9131, cc=0, preamble=5, offset=5, temp_crnti=0x4614
User 0x4614 requesting RRC Reestablishment as 0x460b. Cause: otherFailure
Disconnecting rnti=0x460b.
RACH:  slot=9291, cc=0, preamble=3, offset=3, temp_crnti=0x4615
Disconnecting rnti=0x4615.
RACH:  slot=9451, cc=0, preamble=4, offset=3, temp_crnti=0x4616
Disconnecting rnti=0x4616.
RACH:  slot=9611, cc=0, preamble=4, offset=3, temp_crnti=0x4617
Disconnecting rnti=0x4617.
RACH:  slot=9771, cc=0, preamble=2, offset=3, temp_crnti=0x4618
Disconnecting rnti=0x4618.
RACH:  slot=9931, cc=0, preamble=2, offset=3, temp_crnti=0x4619
Disconnecting rnti=0x4619.
RACH:  slot=10091, cc=0, preamble=5, offset=3, temp_crnti=0x461a
Disconnecting rnti=0x461a.
RACH:  slot=11, cc=0, preamble=7, offset=3, temp_crnti=0x461b
Disconnecting rnti=0x461b.
RACH:  slot=171, cc=0, preamble=4, offset=3, temp_crnti=0x461c
Disconnecting rnti=0x461c.
RACH:  slot=491, cc=0, preamble=5, offset=3, temp_crnti=0x461d
User 0x461d requesting RRC Reestablishment as 0x4614. Cause: otherFailure
Disconnecting rnti=0x4614.
RACH:  slot=651, cc=0, preamble=4, offset=3, temp_crnti=0x461e
Disconnecting rnti=0x461e.
RACH:  slot=811, cc=0, preamble=1, offset=3, temp_crnti=0x461f
Disconnecting rnti=0x461f.
RACH:  slot=971, cc=0, preamble=7, offset=3, temp_crnti=0x4620
Disconnecting rnti=0x4620.
RACH:  slot=1131, cc=0, preamble=1, offset=3, temp_crnti=0x4621
Disconnecting rnti=0x4621.
RACH:  slot=1291, cc=0, preamble=1, offset=3, temp_crnti=0x4622
Disconnecting rnti=0x4622.
RACH:  slot=1451, cc=0, preamble=5, offset=3, temp_crnti=0x4623
Disconnecting rnti=0x4623.
RACH:  slot=1611, cc=0, preamble=3, offset=3, temp_crnti=0x4624
Disconnecting rnti=0x4624.
RACH:  slot=1771, cc=0, preamble=0, offset=3, temp_crnti=0x4625
Disconnecting rnti=0x4625.
RACH:  slot=2091, cc=0, preamble=6, offset=3, temp_crnti=0x4626
User 0x4626 requesting RRC Reestablishment as 0x461d. Cause: otherFailure
Disconnecting rnti=0x461d.
RACH:  slot=2251, cc=0, preamble=4, offset=3, temp_crnti=0x4627
Disconnecting rnti=0x4627.
RACH:  slot=2411, cc=0, preamble=7, offset=3, temp_crnti=0x4628
Disconnecting rnti=0x4628.
RACH:  slot=2571, cc=0, preamble=1, offset=3, temp_crnti=0x4629
Disconnecting rnti=0x4629.
RACH:  slot=2731, cc=0, preamble=4, offset=5, temp_crnti=0x462a
Disconnecting rnti=0x462a.
RACH:  slot=2891, cc=0, preamble=3, offset=5, temp_crnti=0x462b
Disconnecting rnti=0x462b.
RACH:  slot=3051, cc=0, preamble=1, offset=5, temp_crnti=0x462c
Disconnecting rnti=0x462c.
RACH:  slot=3211, cc=0, preamble=5, offset=5, temp_crnti=0x462d
Disconnecting rnti=0x462d.
RACH:  slot=3371, cc=0, preamble=3, offset=5, temp_crnti=0x462e
Disconnecting rnti=0x462e.

```
After failure, If I **activate airplane mode in the UE and deactivate it again, the UE reconnects without any problems with the network and stays connected for a few minutes**, until it disconnects again and so on.

Sometimes it is connected for 20 minutes and others for just a few seconds...

I use a large RRC inactivity timer in the enb.conf `rrc_inactivity_timer = 1073741823`

## Steps to reproduce the problem ##
To reproduce this issue follow the guide https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html
Copy the config files, execute srsenb sudo ./srsenb enb.conf and try to connect the UE to the open5GS net.

srsRAN Config files:
[enb.conf.txt](https://github.com/srsran/srsRAN/files/9018929/enb.conf.txt)
[rr.conf.txt](https://github.com/srsran/srsRAN/files/9018932/rr.conf.txt)
[rb.conf.txt](https://github.com/srsran/srsRAN/files/9018933/rb.conf.txt)
[sib.conf.txt](https://github.com/srsran/srsRAN/files/9018934/sib.conf.txt)

Open5GS config files:
[amf.yaml.txt](https://github.com/srsran/srsRAN/files/9018606/amf.yaml.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN/files/9018610/upf.yaml.txt)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

avalori1, 2023-01-19:
Hi @cristina-posada, did you manage to clarify this?
Thanks

cristina-posada, 2023-03-03:
Hi @avalori1 
No, not yet. I have installed the latest versions of srsRAN and open5gs recently this week and the problem still exists for me.

igoforth, 2023-04-26:
Hi @cristina-posada,

I'm trying to figure out why I'm having similar issues.

Was your Sysmocom SJA2 purchased before or after September 2021 (is it SJA2v1 or SJA2v2)?

Thank you

frankist, 2023-04-27:
Hi

Did you have the same issue with the new srsRAN repository https://github.com/srsran/srsRAN_Project? If so, can you open an issue there?

Thanks

cristina-posada, 2023-04-27:
Hi @igoforth, @frankist. In my case, I don't have the same issue with the new repository https://github.com/srsran/srsRAN_Project. I tested it a few weeks ago and the UE was connected for hours. 

I don't know when the card I'm using was purchased, I don't know if it's version SJA2v1 or SJA2v2 but I'll find out and let you know! 

Thanks

igoforth, 2023-04-27:
> Hi
> 
> Did you have the same issue with the new srsRAN repository https://github.com/srsran/srsRAN_Project? If so, can you open an issue there?
> 
> Thanks

@frankist @cristina-posada Thank you for the response. 

In the case of my UE with srsRAN Project, it disconnects after 2-5 minutes. I have a feeling that's because Project's gNB does not yet support RRC Mobility, Paging and Reestablishment. I'm looking for idle state transition capabilities.

Thanks