srsran/srsRAN_4G #911: Error "internal is not a valid time source" when trying to start srsenb with Ettus USRP B100

smx12, 2022-07-01:
## Issue Description ##
When trying to start srsenb using Ettus B100 I get the following error:
[ERROR] [UHD RF] AssertionError: assertion failed:
  internal is not a valid time source.
  possible values are: [none, external, _external_].

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
PC Intel i3, linux ubuntu 22.04, Ettus B100, BladeRF A4, 

## Expected Behavior ##
srsenb should start properly

## Actual Behaviour ##
Fails with the following error:
[ERROR] [UHD RF] AssertionError: assertion failed:
  internal is not a valid time source.
  possible values are: [none, external, _external_].

## Steps to reproduce the problem ##
sudo ./srsenb/src/srsenb --rf.device_name=uhd --rf.device_args="recv_frame_size=9232,send_frames=9232"

## Additional Information ##
Console messages from srsenb

Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=uhd with args=recv_frame_size=9232,send_frames=9232
Supported RF device list: UHD soapy bladeRF zmq file
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
[INFO] [LOGGING] Fastpath logging disables at runtime.
Opening USRP channels=1, args: recv_frame_size=9232,send_frames=9232
[INFO] [UHD RF] RF UHD Generic instances constructed
/home/user/github/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc:932: uhd_init failed, freeing...
[ERROR] [UHD RF] AssertionError: assertion failed:
  internal is not a valid time source.
  possible values are: [none, external, _external_].
  
Error initializing radio.

egortaran, 2022-08-01:
I have the same issues

andrepuschmann, 2022-08-01:
The B100 isn't supported by srsRAN. It doesn't support the sample rates required and also is USB2 only.

closed 2022-08-01 by andrepuschmann