srsran/srsRAN_4G #918: 5G SA COTS UE

jancoej, 2022-07-04:
## Issue Description ##
The UE does not connect to the 5G network.

## Setup Details ##
Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf

Built in Release mode using commit 6a3b9257e on branch agpl_next.

Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD soapy bladeRF zmq file
Trying to open RF device 'UHD'
NG connection successful
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.HEAD-0-gdbaf4132
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=0 nof_prb=52
t
Enter t to stop trace.
RF status: O=0, U=1, L=2
^CStopping ..
Saving MAC PCAP (DLT=149) to /tmp/enb_mac.pcap
------------------------------------

root@planet:~/.config/srsran# uhd_find_devices 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.HEAD-0-gdbaf4132
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 3241A04
    name: MyB210
    product: B210
    type: b200

root@planet:~/.config/srsran# 
-------------------------------------------------

oot@planet:~/.config/srsran# uhd_usrp_probe 
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_3.15.0.HEAD-0-gdbaf4132
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B210
|   |   serial: 3241A04
|   |   name: MyB210
|   |   product: 2
|   |   revision: 4
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________

-----------------------------------

[enb.conf.txt](https://github.com/srsran/srsRAN/files/9039588/enb.conf.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN/files/9039601/upf.yaml.txt)
[amf.yaml.txt](https://github.com/srsran/srsRAN/files/9039602/amf.yaml.txt)
[amf.log.txt](https://github.com/srsran/srsRAN/files/9039608/amf.log.txt)
[upf.log.txt](https://github.com/srsran/srsRAN/files/9039609/upf.log.txt)
[enb.conf.txt](https://github.com/srsran/srsRAN/files/9039613/enb.conf.txt)

## Expected Behavior ##
UE attached to the 5G cots ue

## Actual Behaviour ##
UE is not attaching to the network

## Steps to reproduce the problem ##
[enb.conf.txt](https://github.com/srsran/srsRAN/files/9039588/enb.conf.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN/files/9039601/upf.yaml.txt)
[amf.yaml.txt](https://github.com/srsran/srsRAN/files/9039602/amf.yaml.txt)
[rr.conf.txt](https://github.com/srsran/srsRAN/files/9039715/rr.conf.txt)

## Additional Information ##
Phones tested 
OPPO Reno 6 5G
GALAXY A905G
PIXEL 6

brendan-mcauliffe, 2022-07-26:
Hi, I suggest moving this issue to the [mailing list](https://lists.srsran.com/mailman/listinfo/srsran-users) and also taking another look at our [5G SA COTS UE app note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html). We have updated the documentation recently to provide more info on ways to assist the phone in seeing and connecting to the srsRAN network. 

As for the lates and underflows, you will need to get rid of these or the UE will not be able to sync properly. I suggest you take a look at [this section](https://docs.srsran.com/en/latest/usermanuals/source/srsue/source/2_ue_getstarted.html?highlight=performance#operating-system-setup) of the documentation. Setting your CPU governor to performance mode should help with this. You can also try installing a low latency kernel. 

Closing this issue.

closed 2022-07-26 by brendan-mcauliffe

mode51software, 2022-08-24:
Hi @brendan-mcauliffe in this case the user has an on-board GPSDO but the default appears to be "internal":

https://github.com/srsran/srsRAN/blob/release_22_04_1/lib/src/phy/rf/rf_uhd_imp.cc#L631

So I think as well as the CPU note the clock sync is just as significant:

`
device_args = clock=gpsdo
`
I have a b205mini without the on-board GPSDO as it's an expensive optional extra, costing as much as the b205 itself (so others may do the same). I think clock sync is even more critical for 5G NR sync than for LTE. I have a Leo Bodnar external one but it took me a while to figure out that the param is "clock=external" whereas the USRP param is "clock_source=external". I see the COTS NSA doc mentions this but not the COTS SA one. I can (just about) see the green LEDs inside the b205mini come on when it syncs. 

I use:
`
device_args = clock=external
`

I note I also allow for 3 seconds rather than 300ms to get the lock:

https://github.com/srsran/srsRAN/blob/release_22_04_1/lib/src/phy/rf/rf_uhd_imp.cc#L830

I think there should be more detail about the need for the reference clock here and the use of the on-board one, with the caveat that I haven't tried the board mounted ones (TCXO and OCXO) so don't know the difference at this point:

https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html#reference-clock

In my case I'm using an external Leo Bodnar for clock sync and the Pixel 6 can see it and register (briefly), though the (rooted, NR locked) OnePlus can't see it at all. The Oppo can see it but not register. Do the developers use a board mounted GPSDO?