srsran/srsRAN_4G #919: File-based RF plugin for virtual radio

j0lama, 2022-07-04:
## Issue Description ##
I've recently discovered that srsRAN has something called RF plugins that allow you to switch between RF implementations (UHD, ZMQ, file-based) from the configuration file. I'm particularly interested in the file-based RF virtual radio because it is the simplest implementation, and it is a good example of building a new RF plugin. However, I cannot make it work. It looks like the UE receives something from the eNB but nothing else. If I inspect the files that are used as uplink and downlink channels, only the downlink file has some data (I guess those bytes are IQ Samples).

## Setup Details ##
I'm using the compile UE/eNB from commit f2dff0b7a on branch master. The only difference between the configuration files that I'm using and the example configuration files are the *device_name* and *device_args* fields.
These are the UE and eNB configuration files:
#### UE
```conf
#####################################################################
#                   srsUE configuration file
#####################################################################

#####################################################################
# RF configuration
#
# freq_offset: Uplink and Downlink optional frequency offset (in Hz)
# tx_gain: Transmit gain (dB).
# rx_gain: Optional receive gain (dB). If disabled, AGC if enabled
# srate: Optional fixed sampling rate (Hz), corresponding to cell bandwidth. Must be set for 5G-SA.
#
# nof_antennas:       Number of antennas per carrier (all carriers have the same number of antennas)
# device_name:        Device driver family. Supported options: "auto" (uses first found), "UHD" or "bladeRF"
# device_args:        Arguments for the device driver. Options are "auto" or any string.
#                     Default for UHD: "recv_frame_size=9232,send_frame_size=9232"
#                     Default for bladeRF: ""
# device_args_2:      Arguments for the RF device driver 2.
# device_args_3:      Arguments for the RF device driver 3.
# time_adv_nsamples:  Transmission time advance (in number of samples) to compensate for RF delay
#                     from antenna to timestamp insertion.
#                     Default "auto". B210 USRP: 100 samples, bladeRF: 27.
# continuous_tx:      Transmit samples continuously to the radio or on bursts (auto/yes/no).
#                     Default is auto (yes for UHD, no for rest)
#####################################################################
[rf]
freq_offset = 0
tx_gain = 40
rx_gain = 40
#srate = 11.52e6

#nof_antennas = 1

# For best performance in 2x2 MIMO and >= 15 MHz use the following device_args settings:
#     USRP B210: num_recv_frames=64,num_send_frames=64

# For best performance when BW<5 MHz (25 PRB), use the following device_args settings:
#     USRP B210: send_frame_size=512,recv_frame_size=512

#device_args = auto
#time_adv_nsamples = auto
#continuous_tx     = auto

# Example for ZMQ-based operation with TCP transport for I/Q samples
device_name = file
device_args = rx_file=/downlink,tx_file=/uplink,base_srate=23.04e6

#####################################################################
# EUTRA RAT configuration
#
# dl_earfcn:   Downlink EARFCN list.
#
# Optional parameters:
# dl_freq:            Override DL frequency corresponding to dl_earfcn
# ul_freq:            Override UL frequency corresponding to dl_earfcn
# nof_carriers:       Number of carriers
#####################################################################
[rat.eutra]
dl_earfcn = 3350
#nof_carriers = 1

#####################################################################
[... 891 lines cut ...]
Supported RF device list: zmq file
CHx base_srate=23.04e6
CH0 rx_file=/uplink
CH0 tx_file=/downlink
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
Error - buffer pool is empty
Error - buffer pool is empty
Error - buffer pool is empty
Error - buffer pool is empty
Error - buffer pool is empty
Error - buffer pool is empty
Error - buffer pool is empty
Error - buffer pool is empty
Error - buffer pool is empty
...
```

## Additional Information ##
My ultimate goal is to build an RF plugin compatible with FlexRAN (IQ Samples encapsulated in eCPRI messages over UDP). Is that even possible with srsUE?

mariatsampazi, 2022-12-09:
Hello @j0lama , have you figured out how to use file mode instead of zmq and usrps?

j0lama, 2022-12-09:
Hi @mariatsampazi, it turns out that the file mode was just a prototype that was initially used for some basic testing.

mariatsampazi, 2022-12-10:
Thank you @j0lama for your answer! What I would like to clarify is the content of the file. Is it IQ samples?

j0lama, 2022-12-10:
Yes, the ZMQ channel carries IQ samples