srsran/srsRAN_4G #920: USRP stops broadcasting signal when adding NR Cell

mrouili, 2022-07-04:
## Issue Description ##
We are using the 5G NSA deployment, the problem is that the network is undetectable by UE's, when we remove the NR cell from the rr.conf file and basically run an LTE deployment, everything works fine and it does show up so we believe that the issue is coming from the NR cell, the ENB signal also disappears when using 5G NSA. 

## Setup Details ##
Operating System: Ubuntu 18.04.6 LTS x86_64
Kernel: 5.4.0-120-lowlatency
CPU: Intel i7-6770HQ (4) @ 2.600GHz
RAM: 15898MiB
RF Front End: USRP x310 with UBX 160MHz daughterboard
UHD Version: UHD_4.1.0.5-0-g6bd0be9c
srsRAN Version: 22.04

We are following the latest srsRAN documentation as given [here.](https://docs.srsran.com/en/latest/getting_started.html)
We changed the MNC and MCC values as mentioned [here](https://docs.srsran.com/en/latest/usermanuals/source/srsenb/source/3_enb_trouble.html#cots-ue-issues) and also tried different dl_earfcn values including the default value of 3350.

## Expected Behavior ##
srsRAN 5G NSA network should show up when scanning using different COTS UE.

## Actual Behaviour ##
We do not see the network, tried with Google Pixel 4a 5G, Iphone Pro Max 13.

## Steps to reproduce the problem ##
We included the following line in the enb.conf file as per the srsRAN docs.
device_args=type=x300,sampling_rate=11.52e6,lo_freq_offset_hz=11.52e6.
We changed the mtu on the interface that connects to the USRP to 4000 , which is the maximum we can do using our PC (default is 1500).
We also changed the dl_earfcn to 3050 (default is 3350) as this is supported by our UE's
Rest are the default configuration given, with the exception of changing MCC and MNC.
We also tried with different bands in the nr cell, but it made no difference. changing to TDD also made no difference
## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
EPC output

```
---  Software Radio Systems EPC  ---

Couldn't open , trying /home/n6saha/.config/srsran/epc.conf
Reading configuration file /home/n6saha/.config/srsran/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf302, MNC: 0xf760
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
```
ENB output
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_soapy.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /home/n6saha/.config/srsran/enb.conf
Reading configuration file /home/n6saha/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /home/n6saha/.config/srsran/sib.conf
Couldn't open rr.conf, trying /home/n6saha/.config/srsran/rr.conf
Couldn't open rb.conf, trying /home/n6saha/.config/srsran/rb.conf

Built in RelWithDebInfo mode using commit f2dff0b7a on branch HEAD.

Opening 2 channels in RF device=default with args=type=x300,sampling_rate=11.52e6,lo_freq_offset_hz=23.04e6
Supported RF device list: UHD soapy zmq file
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.1.0.5-0-g6bd0be9c
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=x300,lo_freq_offset_hz=23.04e6,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 3972 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=2650.0 Mhz, UL=2530.0 MHz for cc_idx=0 nof_prb=50
Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=1 nof_prb=52
```

## Log File 
[enb (1).log](https://github.com/srsran/srsRAN/files/9041696/enb.1.log)

## LTE signal is broadcasted when we don't have NRcell
[image]

## LTE signal disappears (just noise) when we have NRcell
[image]

## ENB output
We get some warnings when we use NRcell, which is not there when using LTE
[image]

andrepuschmann, 2022-07-05:
Hey, can you post the output of `uhd_usrp_probe`? Also can you check the Tx LEDs on both ports of the X310?

mrouili, 2022-07-05:
> Hey, can you post the output of `uhd_usrp_probe`? Also can you check the Tx LEDs on both ports of the X310?

Hello, here is the output of `uhd_usrp_probe` 

```
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.1.0.5-0-g6bd0be9c
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 3972 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
  _____________________________________________________
 /
|       Device: X-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: X310
|   |   revision: 11
|   |   revision_compat: 7
|   |   product: 30818
|   |   mac-addr0: 00:80:2f:34:56:48
|   |   mac-addr1: 00:80:2f:34:56:49
|   |   gateway: 192.168.10.1
|   |   ip-addr0: 192.168.10.2
|   |   subnet0: 255.255.255.0
|   |   ip-addr1: 192.168.20.2
|   |   subnet1: 255.255.255.0
|   |   ip-addr2: 192.168.30.2
|   |   subnet2: 255.255.255.0
|   |   ip-addr3: 192.168.40.2
|   |   subnet3: 255.255.255.0
|   |   serial: 324662C
|   |   FW Version: 6.0
|   |   FPGA Version: 38.0
|   |   FPGA git hash: 26793b8
|   |   
|   |   Time sources:  internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|     _____________________________________________________
|    /
|   |       RFNoC blocks on this device:
|   |   
|   |   * 0/DDC#0
|   |   * 0/DDC#1
|   |   * 0/DUC#0
|   |   * 0/DUC#1
|   |   * 0/Radio#0
|   |   * 0/Radio#1
|   |   * 0/Replay#0
|     _____________________________________________________
|    /
|   |       Static connections on this device:
|   |   
|   |   * 0/SEP#0:0==>0/DUC#0:0
|   |   * 0/DUC#0:0==>0/Radio#0:0
|   |   * 0/Radio#0:0==>0/DDC#0:0
|   |   * 0/DDC#0:0==>0/SEP#0:0
|   |   * 0/Radio#0:1==>0/DDC#0:1
|   |   * 0/DDC#0:1==>0/SEP#1:0
|   |   * 0/SEP#2:0==>0/DUC#1:0
|   |   * 0/DUC#1:0==>0/Radio#1:0
|   |   * 0/Radio#1:0==>0/DDC#1:0
|   |   * 0/DDC#1:0==>0/SEP#2:0
|   |   * 0/Radio#1:1==>0/DDC#1:1
|   |   * 0/DDC#1:1==>0/SEP#3:0
|   |   * 0/SEP#4:0==>0/Replay#0:0
|   |   * 0/Replay#0:0==>0/SEP#4:0
|   |   * 0/SEP#5:0==>0/Replay#0:1
|   |   * 0/Replay#0:1==>0/SEP#5:0
|     _____________________________________________________
|    /
|   |       TX Dboard: 0/Radio#0
|   |   ID: UBX-160 v2 (0x007d)
|   |   Serial: 323CE03
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: UBX TX
|   |   |   Antennas: TX/RX, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: QI
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: 0/Radio#0
|   |   ID: UBX-160 v2 (0x007e)
|   |   Serial: 323CE03
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: UBX RX
|   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       TX Dboard: 0/Radio#1
|   |   ID: UBX-160 v2 (0x007d)
|   |   Serial: 323CE34
|   |     _____________________________________________________
|   |    /
|   |   |       TX Frontend: 0
|   |   |   Name: UBX TX
|   |   |   Antennas: TX/RX, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: QI
|   |   |   Uses LO offset: No
|     _____________________________________________________
|    /
|   |       RX Dboard: 0/Radio#1
|   |   ID: UBX-160 v2 (0x007e)
|   |   Serial: 323CE34
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: UBX RX
|   |   |   Antennas: TX/RX, RX2, CAL
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range PGA0: 0.0 to 31.5 step 0.5 dB
|   |   |   Bandwidth range: 160000000.0 to 160000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
```
This is when the usrp is idle, since when the usrp is running as rf for the enb, the command doesn't detect the device.

The TX/RX LED's on both ports keep blinking in red when the 5G NSA is deployed, while the RX2 ports keep blinking in green.

mrouili, 2022-07-06:
> If the leds are blinking you have an real-time issue with your PC and it can't hold the load. you need to get rid of the blinking and achieve a continues light. otherwise you have holes in your transmission and depending on your sampling you might not even see your transmission like you discribed

Thank you for your reply, I am not sure how I can fix this issue, also the PC we are using meets the minimum requirements mentioned in the guide (minimum 6th gen i7 or similar) which is what this PC has. Any ideas on what I can try to fix this issue ?
thanks !

brendan-mcauliffe, 2022-07-26:
You can try setting the CPU governor of the PC to performance mode. You can do this with: 

`echo "performance" | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`

You can also try to install a low latency kernel.

cristina-posada, 2022-08-24:
Hello!! I'm having the same problem configuring the 5G NSA COTS UE

If I remove the nr_cell section from the rr.conf file, the UE can attach the LTE network and connects just fine. But if I add the nr_cell to the rr.conf The UE is not even able to see the network in the list of network operators.

I'm using the USRP B210 and last srsRAN version
Ubuntu 20.04.4 LTS 1th Gen Intel® Core™ i7-1165G7 @ 2.80GHz × 8 

- I already have the CPU governor of the PC at performance mode
- The TX/RX LED's on both ports are **not** blinking

[enb.conf.txt](https://github.com/srsran/srsRAN/files/9415773/enb.conf.txt)
[epc.conf.txt](https://github.com/srsran/srsRAN/files/9415774/epc.conf.txt)
[rb.conf.txt](https://github.com/srsran/srsRAN/files/9415775/rb.conf.txt)
[rr.conf.txt](https://github.com/srsran/srsRAN/files/9415776/rr.conf.txt)
[sib.conf.txt](https://github.com/srsran/srsRAN/files/9415777/sib.conf.txt)

**srsepc output**
```
Built in Release mode using commit ce8a3cae1 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01
S1 Setup Request - TAC 7, B-PLMN 0xf110
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response

```

**srsenb output**
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_blade.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...

Built in Release mode using commit ce8a3cae1 on branch master.

Opening 2 channels in RF device=uhd with args=clock=gpsdo
Supported RF device list: UHD bladeRF zmq file
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.0-0ubuntu1~focal1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=2, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO , Firmware Rev 0.929a
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting USRP time to 1661342201.000000s
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[... 98 lines cut ...]
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: B
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B210 TX dual DAC
|   |   |   |   Gain Elements: None

```

cristina-posada, 2022-08-31:
> Hello!! I'm having the same problem configuring the 5G NSA COTS UE
> 
> If I remove the nr_cell section from the rr.conf file, the UE can attach the LTE network and connects just fine. But if I add the nr_cell to the rr.conf The UE is not even able to see the network in the list of network operators.
> 
> I'm using the USRP B210 and last srsRAN version Ubuntu 20.04.4 LTS 1th Gen Intel® Core™ i7-1165G7 @ 2.80GHz × 8
> 
> * I already have the CPU governor of the PC at performance mode
> * The TX/RX LED's on both ports are **not** blinking
> 
> [enb.conf.txt](https://github.com/srsran/srsRAN/files/9415773/enb.conf.txt) [epc.conf.txt](https://github.com/srsran/srsRAN/files/9415774/epc.conf.txt) [rb.conf.txt](https://github.com/srsran/srsRAN/files/9415775/rb.conf.txt) [rr.conf.txt](https://github.com/srsran/srsRAN/files/9415776/rr.conf.txt) [sib.conf.txt](https://github.com/srsran/srsRAN/files/9415777/sib.conf.txt)
> 
> **srsepc output**
> 
> ```
> Built in Release mode using commit ce8a3cae1 on branch master.
> 
> 
> ---  Software Radio Systems EPC  ---
> 
> Reading configuration file epc.conf...
> HSS Initialized.
> MME S11 Initialized
> MME GTP-C Initialized
> MME Initialized. MCC: 0xf001, MNC: 0xff01
> SPGW GTP-U Initialized.
> SPGW S11 Initialized.
> SP-GW Initialized.
> Received S1 Setup Request.
> S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
> S1 Setup Request - MCC:001, MNC:01
> S1 Setup Request - TAC 7, B-PLMN 0xf110
> S1 Setup Request - Paging DRX v128
> Sending S1 Setup Response
> ```
> 
> **srsenb output**
> 
> ```
> Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_blade.so libsrsran_rf_zmq.so
> Inactive RF plugins: 
> ---  Software Radio Systems LTE eNodeB  ---
> 
> Reading configuration file enb.conf...
> 
> Built in Release mode using commit ce8a3cae1 on branch master.
> 
> Opening 2 channels in RF device=uhd with args=clock=gpsdo
> Supported RF device list: UHD bladeRF zmq file
> [INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.0-0ubuntu1~focal1
> [INFO] [LOGGING] Fastpath logging disabled at runtime.
> Opening USRP channels=2, args: type=b200,master_clock_rate=23.04e6
> [INFO] [UHD RF] RF UHD Generic instance constructed
> [INFO] [B200] Detected Device: B210
> [INFO] [B200] Operating over USB 3.
> [INFO] [B200] Detecting internal GPSDO.... 
> [INFO] [GPS] Found an internal GPSDO: GPSTCXO , Firmware Rev 0.929a
> [INFO] [B200] Initialize CODEC control...
> [INFO] [B200] Initialize Radio control...
> [INFO] [B200] Performing register loopback test... 
> [INFO] [B200] Register loopback test passed
> [INFO] [B200] Performing register loopback test... 
> [INFO] [B200] Register loopback test passed
> [INFO] [B200] Asking for clock rate 23.040000 MHz... 
> [INFO] [B200] Actually got clock rate 23.040000 MHz.
> Setting USRP time to 1661342201.000000s
> [INFO] [MULTI_USRP]     1) catch time transition at pps edge
> [INFO] [MULTI_USRP]     2) set times next pps (synchronously)
> 
> ==== eNodeB started ===
> Type <t> to view trace
[... 98 lines cut ...]
> |   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
> |   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
> |   |   |   |   Connection Type: IQ
> |   |   |   |   Uses LO offset: No
> |   |   |     _____________________________________________________
> |   |   |    /
> |   |   |   |       TX Frontend: B
> |   |   |   |   Name: FE-TX1
> |   |   |   |   Antennas: TX/RX
> |   |   |   |   Sensors: temp, lo_locked
> |   |   |   |   Freq range: 50.000 to 6000.000 MHz
> |   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
> |   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
> |   |   |   |   Connection Type: IQ
> |   |   |   |   Uses LO offset: No
> |   |   |     _____________________________________________________
> |   |   |    /
> |   |   |   |       TX Codec: A
> |   |   |   |   Name: B210 TX dual DAC
> |   |   |   |   Gain Elements: None
> ```
> 
> **logs:** [enb.log](https://github.com/srsran/srsRAN/files/9416054/enb.log)

Ok, I **can't run a 5G NSA network with just one USRP B210** because I should use an SDR that supports two independent channels, like USRP N310 or USRP X310.

andrepuschmann, 2022-09-07:
@mrouili Were you able to solve the issue meanwhile?

@cristina-posada The B210 does NOT support NSA mode.

mrouili, 2022-09-23:
@andrepuschmann We have new and powerful computers (i9 10980XE cpus), we are still facing issues with 5G NSA mode and SA releases, for NSA we are able to start the network and even connect on our Samsung S20 Ultra 5G UE, when we connect we see the 5G symbol on the UE however the enb trace shows only LTE traffic and no NR, so what we are suspecting is that the UE is connecting to the LTE cell but for some reason is not switching to the NR cell after that, we have further confirmed this by running a simple speedtest on the UE where we got the exact same results we got with the LTE only release, any help or ideas would be appreciated, thanks.

spaceshub, 2024-12-20:
can someone sahre configs for srsle to make 4G working on B210 . i'm getting same problem 

`[INFO] [LOGGING] Fastpath logging disabled at runtime.
NG connection successful
Opening USRP channels=1, args: num_recv_frames=64,num_send_frames=64,type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1842.5 Mhz, DL_SSB=1843.25 Mhz (SSB-ARFCN=368650), UL=1747.5 MHz for cc_idx=0 nof_prb=52
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 1.46571 >= 0. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 4.70659 >= 4.16392. Starting new burst...`

trying to run 4g only or NR nodes, same outcome

i have this DSR connected via cable to USB Hub Usb3 -> USBC and connected to my macbook pro
i tried different hardware arguments.
in some cases i get green lamp blinking only
in some cases reg and green blinks for 3-4 seconds and than turn off

Thanks!