srsran/srsRAN_4G #924: srsue synchronization issues with SA Amarisoft Callbox

kbaccar, 2022-07-08:
## Issue Description ##
I have been trying to make a 5G SA connection between srsue and the amarisoft callbox following the tutorial on the wiki page. Im using an SDR B210 and without a GPSDO the connection seems completely random as sometimes it stops at the RA procedure or at most I manage to just get an RRC connection for it to get released afterward with a scheduling request error. After installing a GPSDO (tcxo from ettus) on the SDR, setting clock=external would give me completely nothing and the SDR fails to send any message to the callbox. I only managed to get a  PDU session with clock=gpsdo but the connection is still nonetheless unstable as i lose synchronization immediately after trying an iperf test.

## Setup Details ##
srsue machine: ubuntu 20.04., srsue 22.4, UHD 3.15.
Amarisoft callbox: lteots-linux-2022-03-19, usrp B210 connected on sdr1 on rx1 and tx1 with -30 db attenuators.

[enb.txt](https://github.com/srsran/srsRAN/files/9072757/enb.txt)
[ue.txt](https://github.com/srsran/srsRAN/files/9072769/ue.txt)

andrepuschmann, 2022-07-11:
The lates your seeing (`RF status: O=0, U=5, L=54`) causes the UE to loose sync and it then can't resync again. It's a known limitation I am afraid.

TallGuy74, 2022-08-30:
@kbaccar Are you running iperf3 on the client with zerocopy (-Z) enabled? It makes a huge difference on the CPU use of iperf, and may be enough to get rid of the late packets. Also, make sure you have the CPU governor to performance with `echo performance | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`.

IndrawanBudiP, 2024-03-13:
> The lates your seeing (`RF status: O=0, U=5, L=54`) causes the UE to loose sync and it then can't resync again. It's a known limitation I am afraid.

Hi, if something happen like that. So what should I do to make it sync again