srsran/srsRAN_4G #929: How to connect 2 UEs to an eNB with ZMQ?

zzylol, 2022-07-15:
## Issue Description ##
How to connect 2 UEs to an eNB with ZMQ?

## Setup Details ##

I have 4 vms and their roles and ip addresses are:
epc -- 10.0.0.5
enb -- 10.0.0.6
ue1 -- 10.0.0.7
ue2 -- 10.0.0.8

I can connect 1 UE to the enb with this configuration:
[enb]
enb_id = 0x19B
mcc = 208
mnc = 93
mme_addr = 10.0.0.5
gtp_bind_addr = 10.0.0.6
s1c_bind_addr = 10.0.0.6
s1c_bind_port = 0
n_prb = 50
[enb_files]
sib_config = sib.conf
rr_config  = rr.conf
rb_config = rb.conf
[rf]
#dl_earfcn = 3350
tx_gain = 80
rx_gain = 40
device_name = zmq
device_args = fail_on_disconnect=true,tx_port=tcp://10.0.0.6:2000,rx_port=tcp://10.0.0.7:2001,id=enb,base_srate=23.04e6

I modified the enb.conf to 
`device_args = fail_on_disconnect=true,tx_port=tcp://10.0.0.6:2000,rx_port=tcp://10.0.0.[7-8]:2001,id=enb,base_srate=23.04e6`
But it can let neither UE to connect to the eNB. The UE stuck at attaching phase. 

## Expected Behavior ##
2 UEs can be attached to 1 eNB.

WingPig99, 2022-07-17:
@zzylol This https://github.com/srsran/srsRAN/issues/818 should give you some insights.

zzylol, 2022-07-18:
Hi @WingPig99, thanks a lot for the reference.

closed 2022-07-18 by zzylol