srsran/srsRAN_4G #930: How to use GNU-Radio Companion Integration with ZMQ

zzylol, 2022-07-19:
## Issue Description ##
I followed #818's instruction to set up 2 UEs to connect to one eNB, but I cannot make either of the UE to connect and GRC shows nothing. I am wondering how to integrate srsRAN with ZMQ with GRC. What are the exact steps to run? 
Before I started epc, enb, 2 UEs in order, and then clicked "Execute" button in GRC. my eNB can connect to EPC, but 2 UEs stuck at "Attaching UE...".

UE1:
```
~/srsRAN/build/srsue/src$ sudo ./srsue ue1.conf --gw.netns=ue1
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue1.conf...

Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.*:2000,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.*:2000
CH0 tx_port=tcp://127.0.0.1:2001
Waiting PHY to initialize ... done!
Attaching UE...

```
UE2: 
```
~/srsRAN/build/srsue/src$ sudo ./srsue ue2.conf --gw.netns=ue2
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue2.conf...

Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2011,rx_port=tcp://127.*:2010,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.*:2010
CH0 tx_port=tcp://127.0.0.1:2011
Waiting PHY to initialize ... done!
Attaching UE...
```
EPC:
```
~/srsRAN/build/srsepc/src$ sudo ./srsepc epc.conf 

Built in Release mode using commit f2dff0b7a on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf208, MNC: 0xff93
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:208, MNC:93
S1 Setup Request - TAC 1, B-PLMN 0x2f839
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
```

eNB:
```
~/srsRAN/build/srsenb/src$ sudo ./srsenb enb.conf 
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...

Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://127.0.0.1:2101,rx_port=tcp://127.*:2100,id=enb,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.*:2100
CH0 tx_port=tcp://127.0.0.1:2101
CH0 fail_on_disconnect=true

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Current sample rate is 11.52 MHz with a base rate of 23.04 MHz (x2 decimation)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50

```
## Setup Details ##
I am running in an ubuntu 20.04 virtual machine. 
my configuration files are:
[2ue.txt](https://github.com/srsran/srsRAN/files/9137395/2ue.txt) as .grc file.
[enb.txt](https://github.com/srsran/srsRAN/files/9137417/enb.txt) as .conf file.
[epc.txt](https://github.com/srsran/srsRAN/files/9137418/epc.txt) as .conf file.
[ue1.txt](https://github.com/srsran/srsRAN/files/9137419/ue1.txt) as .conf file.
[ue2.txt](https://github.com/srsran/srsRAN/files/9137420/ue2.txt) as .conf file.

LongRiver-hash, 2022-07-19:
I also encountered the same problem. Connecting one UE is also the same. Does it need a network cable connection or a WiFi LAN connection?

LongRiver-hash, 2022-07-19:
I find it and solved.see https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html
you need the command
sudo ip netns add ue1
sudo ip netns list

zzylol, 2022-07-19:
> I find it and solved.see https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html you need the command sudo ip netns add ue1 sudo ip netns list

I already start my 2 UEs in 2 netnses. I am curious how many UEs are you connecting? Thanks.

zzylol, 2022-07-19:
I solved this problem. The run steps are described as above. But for me, I cannot use ip address like * or 127.*. I changed to exact ip addresses for 2 UEs to connect to the eNB.

closed 2022-07-19 by zzylol

raul-parada, 2023-10-19:
Hi @zzylol, the link is broken. Please, could you tell me the steps you have followed? Did you try with this [environment](https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html)?