srsran/srsRAN_4G #931: UE's can't connect to LimeSDR on a bandwhith 1,4Mhz

zikkuratti, 2022-07-19:
Faced with the problem that the LimeSDR USB doesn't start registration procedure then n_prb=6 (Band 8) on prb = 50 it's work well but to me is principal  got any useful advice that to config to solve my problem, also using SrsRan 20.04 because last is drop after 5 call's trying different UE's PC's etc. The only i point i saw is to config in sib.conf  line  prach_freq_offset = 4;
 to zero. How about you, comrades?

zikkuratti, 2022-07-27:
Last main 22.04 work with 1,4Mhz well, but couldn't make more then 4 calls, dear maintainers could u provide me lines, functions of code wich i can fix? or i go to re:fuck it all)

zikkuratti, 2022-07-29:
Files affected by changes in the n_prb key variable affecting the band width when migrating from version 21.04 to 22.04

1)
https://github.com/srsran/srsRAN/blob/release_22_04/lib/src/phy/ch_estimation/chest_ul.c

2)
https://github.com/srsran/srsRAN/blob/release_22_04/srsenb/hdr/stack/mac/sched_phy_ch/sched_dci.h
#include "../sched_lte_common.h"
#include "srsenb/hdr/stack/mac/sched_phy_ch/sched_phy_resource.h"

3)
replaced by static constants in fresh code
https://github.com/srsran/srsRAN/blob/release_22_04/lib/src/phy/ch_estimation/refsignal_ul.c

four)
rewritten includes and part of drb code to rb
https://github.com/srsran/srsRAN/blob/release_22_04/srsenb/hdr/enb.h
#include "srsenb/hdr/stack/mac/sched_interface.h"
#include "srsgnb/hdr/stack/gnb_stack_nr.h"
#include "srsran/interfaces/enb_time_interface.h"
#include "srsran/interfaces/enb_x2_interfaces.h"

5)
almost all changed something about ports and antennas
https://github.com/srsran/srsRAN/blob/release_22_04/lib/src/phy/phch/ra_dl_nr.c

6)
added to config nr_pdsch_mcs=28 and much more commented out
https://github.com/srsran/srsRAN/blob/release_22_04/srsenb/enb.conf.example

7)
more than a thousand lines of changes supposedly added 5G but not only
https://github.com/srsran/srsRAN/blob/release_22_04/srsenb/src/enb_cfg_parser.cc
#include "srsgnb/hdr/stack/rrc/rrc_nr_config_utils.h"
#include "srsran/asn1/rrc_utils.h"
#include "srsran/common/band_helper.h"
#include "srsran/rrc/rrc_common.h"

eight)
memory handling slightly updated
https://github.com/srsran/srsRAN/blob/release_22_04/srsenb/hdr/stack/mac/sched_grid.h
#include "sched_common.h"
#include "sched_interface.h"

9)
work with signal modulation has been altered, technologies may have been added
https://github.com/srsran/srsRAN/blob/release_22_04/lib/src/phy/phch/ra_nr.c
#include "srsran/phy/fec/cbsegm.h"
#include <math.h> /* floor */

ten)
some memory points have been changed and the inclusion has been patched to the lte marker
https://github.com/srsran/srsRAN/blob/release_22_04/srsenb/src/stack/mac/sched_phy_ch/sched_dci.cc
#include "srsenb/hdr/stack/mac/sched_lte_common.h"

eleven)
200+ changes to the main launch file, mostly from a fresh config, but also logic and includes added
https://github.com/srsran/srsRAN/blob/release_22_04/srsenb/src/main.cc
#include <sys/mman.h>
#include "srsran/common/tsan_options.h"
#include "srsran/support/emergency_handlers.h"
#include "srsran/support/signal_handler.h"

Dear maintainers got some advices?

egortaran, 2022-08-02:
Which version of SoapySDR and LimeSuite are you using?

zikkuratti, 2022-08-12:
it's nevermind bro. 

Also i catch some info from test. 
Error. Detected 2 CQI != 15 in eNB logs. We should finish with all qci=15.

11:16:45.250556 [PHY2   ] [I] [  361] PUCCH: cc=0; rnti=0x46, f=2, n_pucch=0, dmrs_corr=-nan, snr=-inf dB, corr=0.000, cqi=0 (cc=0), ta=0.0 us
11:16:58.562601 [PHY0   ] [I] [  361] PUCCH: cc=0; rnti=0x46, f=2, n_pucch=0, dmrs_corr=-nan, snr=-inf dB, corr=0.000, cqi=0 (cc=0), ta=0.0 us

And looks like it has some reasons here but not shure at all: /lib/src/mac/mac_sch_pdu_nr.cc