srsran/srsRAN_4G #934: 5G SA COTS UE can not found network operators

SophieXin9636, 2022-07-21:
## Issue Description ##
I followed up this [note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html).
The srsenb and Open5GS work correctly, but UE can not found network operators. 

## Setup Details ##
* Operation System: Ubuntu 18.04
* srsRAN 22.04
* RF: USRP B210 (UHD driver)
* CPU: i7-7700
* UE: Galaxy S21 5G 
* Card type: Sysmocom USIM (sysmoUSIM-SJS1)
* Third Party 5G Core: Open5GS

## Expected Behavior ##
srsenb would receive "User 0x?? connected"

## Actual Behaviour ##
UE can not found network operators.
UE do not have 5G signal.

## Steps to reproduce the problem ##
1. Reset IP Tables and Restart open5gs daemons follow this [note](https://open5gs.org/open5gs/docs/guide/01-quickstart/)
[image]

2. Set the UE's APN in the Open5GS WebUI
[image]

3. UE APN setting
Setting -> Access Point Names -> Add
* Name: internet
* APN: internet
* APN protocol: IPv4
[image]

4. Force NR only and enable Data roaming
[image]

5. Run `sudo srsenb enb.conf`
[image]

6. Click Network operators -> Select automatically
No result (& no signal)
[image]

## Additional Information ##

bedrankara, 2022-07-26:
Hey,

Samsung phone using the Exynos chipsets are very picking when it comes to cell selection. 
The phone will find the cell only if you use an external clock with the B210. This is my experience for LTE at least.

SophieXin9636, 2022-07-27:
Hi @bedrankara 

Thanks your comment, but the Samsung phone in my region is using Qualcomm chipsets (Snapdragon 888).
I have no idea whether it's chip related.

strongcourage, 2022-08-22:
Hi @SophieXin9636,
Did your phone successfully connect to the network? I faced the same problem using a Samsung SM-G981B (chip Exynos990) and am still looking for the solution.

Cheers,

iman432, 2022-08-25:
Hi,

Samsung phones (also Apple) have configuration for different PLMNs and should allow 5G for specific PLMN, if you can change your PLMN to an commercial PLMN maybe helps, meanwhile you can dial *#0011# for checking phone detect any signal(your earfcn) or not.

strongcourage, 2022-08-29:
Thanks @iman432. Actually I tried to use a commercial PLMN in a new subscriber's IMSI on Open5GS web UI, restarted SRSRAN and Open5GS with new config files and reprogrammed my SIM card. However, my phone still can't find the network :(

roema, 2022-09-03:
Hi!
I have the same problem, testes with S21 Ultra, Xiamo Note 11 Pro 5G, S22+.
We use the B200-mini. No UE detect 5G SA cell. LTE work very well.

We also have tested with commercial PLMN.

iman432, 2022-09-06:
Hi,
in my case S21 FE is detecting 5G SA signal and even by private PLMN :

[image]

roema, 2022-09-06:
@iman432 thanks for Info.
Wich hardware are you suing?
Do you use external ref clock?

iman432, 2022-09-06:
> @iman432 thanks for Info. Wich hardware are you suing? Do you use external ref clock?

I'm using USRP B200 and not any external clock.

roema, 2022-09-06:
@iman432 Thanks!

Can you share you configs ?

roema, 2022-09-06:
@iman432 when i configure ARFCN 368410 with band 3, i get an error, 

DL ARFCN (368410) is not valid for the specified band (3)

Wich version are you suing?

iman432, 2022-09-06:
> @iman432 Thanks!
> 
> Can you share you configs ?

its my configuration: 

[enb.zip](https://github.com/srsran/srsRAN/files/9498572/enb.zip)

 and version is 22.04

roema, 2022-09-07:
@iman432 Thanks!! I will test.

roema, 2022-09-07:
@iman432 Sorry, but your configuration are for LTE.
n_prb = 100 not valid for 5G, only 10MHz supported.
And in your rr.conf, lte cells active.

iman432, 2022-09-07:
> @iman432 Sorry, but your configuration are for LTE. n_prb = 100 not valid for 5G, only 10MHz supported. And in your rr.conf, lte cells active.

Sorry, below is for 5G and working for me.
[gNB.zip](https://github.com/srsran/srsRAN/files/9505470/gNB.zip)

roema, 2022-09-07:
@iman432 Thank you very much!!

roema, 2022-10-04:
Sorry for the late answer.
I have tested, but it doesnt work for me.
Xiamo Note 11 Pro 5G will not find the 5G cell.

hyunseoha, 2022-11-15:
is there any positive update from this issue ?
i am using the same note to construct my own 5g network with oneplus 5g mobile phone.
it has exactly the same problem. it does not find any 5g cell, but working well with LTE.

Thanks !

fllay, 2022-11-17:
I also has  the same problem. I am using Samsung A33 5G. I asked a person of the 5G operator here. He told me that the phone is locked to the PLMN here. I think I have to follow the recommendation here 

"Many 5G handsets may contain a carrier policy file that may limit 5G capabilities of the phone based on the PLMN of the USIM (first 6 digits of IMSI). Carrier policy files typically don’t include test network PLMNs, so setting a test PLMN may result in 5G being disabled. If possible, using a shielded box and configuring the network with a commercial carrier PLMN may avoid policy file issues."

I use SIM card from here 

https://shop.sysmocom.de/sysmoISIM-SJA2-SIM-USIM-ISIM-Card-10-pack-with-ADM-keys/sysmoISIM-SJA2-10p-adm

I have no idea how to change  PLMN. I use pysim to change IMSI but it seems to invalidate the SIM card. Do I have to change Ki and Opc? Anyone has any idea? I am not a SIM card expert and I dont want tp be. 

For LTE, using the same SIM card, It works just fine. I only swap comment on cell configuration in `rr.conf`.  

I think the problem is the UE only and I dont think I can afford more phone to test. The only UE I can think of now is  ZTE MU5002 since It is not officially sell here. It is imported from China and people here can use it to connect to any operator. I dont think it is any restriction on this ZTE.

hyunseoha, 2022-11-17:
> 
I thought PLMN is a combination of MNC and MCC. you can change them in sysmoISIM configuration. See the tutorial in the website of the seller.

fllay, 2022-11-17:
Yes it is. I read the doc in the website and I also tried to change it. However, if we look at the command to change MNC and MCC 

```
$ ./pySim-prog.py -p 0 -t sysmoUSIM-SJS1 -a 58001006  -x 901 -y 71 -i 901700000010659 -s 8988211000000110000 -o 398153093661279FB1FC74BE07059FEF -k 1D8B2562B992549F20D0F42113EAA6FA
 Insert card now (or CTRL-C to cancel)
 Generated card parameters :
  > Name    : Magic
  > SMSP    : e1ffffffffffffffffffffffff0581005155f5ffffffffffff000000
  > ICCID   : 8988211000000110000
  > MCC/MNC : 901/71
  > IMSI    : 901700000010659
  > Ki      : 1D8B2562B992549F20D0F42113EAA6FA
  > OPC     : 398153093661279FB1FC74BE07059FEF
  > ACC     : None
 Programming ...
 Done !
```

You will see that  MCC/MNC : 801/71 is changed but IMSI    : 901990000000018 is still the same. If we take a look at APN setting. You will notice that MCC/MNC does not change. I guess it is read from IMSI.  So I change the IMSI 

```
$ ./pySim-prog.py -p 0 -x 801 -y 71 -i 801710000000018 -s 8988211000000110000 -o 398153093661279FB1FC74BE07059FEF -k 1D8B2562B992549F20D0F42113EAA6FA
```

I can be changed but now the SIM card will be invalid. It does not even work with LTE network anymore. It will get connect and disconnect from the LTE network and for 5G, it does not show anything. Network search show nothing as well. My question is how to change  MCC/MNC without making the SIM card invalid.

helloTkk, 2023-01-13:
> is there any positive update from this issue ? i am using the same note to construct my own 5g network with oneplus 5g mobile phone. it has exactly the same problem. it does not find any 5g cell, but working well with LTE.
> 
> Thanks !

Hi, I'm also using Oneplus Nord 5G as listed in the note and I'm facing the same issue. Did you figure it out? My email is tukai2259@gmail.com , maybe we can further discuss and try to solve it.