srsran/srsRAN_4G #937: Cannot enable Sounding reference signal on srsue in release 22.04

Luke20000429, 2022-07-22:
## Issue Description ##
Hi, I tried to customize RRC message to let srsue send sounding reference signal. I added flags in `rrc_utils.cc` to see if my customized configs are applied in PHY setup and it seems that `ul_cfg.srs.configured` is true and `ul_cfg.srs` looks correct. However, srsue still cannot pass `srs_tx_enabled(&cfg->ul_cfg.srs, tti)` in function `add_srs` and fails to attach to eNB. 

## Setup Details ##
We are running on 2 USRP B210 on two hosts, with ubuntu low-latency kernel.

I saw there is a `srs_setup` sample in `ue_rr_cfg.cc` disabled by `#if SRS_ENABLED`, but that is called when only a new cell is added. Therefore, I modified `rrc::ue::send_connection_setup()` by adding 
```
  bool srs_present = parent->cfg.sibs[1].sib2().rr_cfg_common.srs_ul_cfg_common.type(); 

  auto* phy_cfg = &rr_cfg.phys_cfg_ded;
  if (srs_present) {
    fprintf(stderr, "[M: %s] establish configuration srs\n", __func__);
    phy_cfg->srs_ul_cfg_ded_present = true;
    auto& srs_setup = phy_cfg->srs_ul_cfg_ded.set_setup();
    srs_setup.srs_bw.value = srs_ul_cfg_ded_c::setup_s_::srs_bw_opts::bw0;
    srs_setup.srs_hop_bw.value = srs_ul_cfg_ded_c::setup_s_::srs_hop_bw_opts::hbw0;
    srs_setup.cyclic_shift.value = srs_ul_cfg_ded_c::setup_s_::cyclic_shift_opts::cs0;
    srs_setup.freq_domain_position = 0;
    srs_setup.srs_cfg_idx = 167;
    srs_setup.tx_comb = 0;
    srs_setup.dur = true;
    phy_cfg->srs_ul_cfg_ded_v1020.set_present();
    phy_cfg->srs_ul_cfg_ded_v1020.get()->srs_ant_port_r10.value = srs_ant_port_opts::an1;
    phy_cfg->srs_ul_cfg_ded_aperiodic_r10.set_present();
    phy_cfg->srs_ul_cfg_ded_aperiodic_r10.get()->set(setup_opts::release);
  }

 // Configure PHY layer
  apply_setup_phy_config_dedicated(rr_cfg.phys_cfg_ded); // It assumes SCell has not been set before

```

And in the sib2 parser `parse_sib2(std::string filename, sib_type2_s* data)`
```
  parser::section srs_ul_cnfg("srs_ul_cnfg");
  rr_config.add_subsection(&srs_ul_cnfg);
  auto &srs_setup = rr_cfg_common->srs_ul_cfg_common.set_setup();

  srs_ul_cnfg.add_field(
    make_asn1_enum_number_parser("bw_cnfg", &srs_setup.srs_bw_cfg)
  );
  srs_ul_cnfg.add_field(
    make_asn1_enum_number_parser("subfr_cnfg", &srs_setup.srs_sf_cfg)
  );
  srs_ul_cnfg.add_field(
    new parser::field<bool>
    ("ack_nack_simul_tx", &srs_setup.ack_nack_srs_simul_tx)
  );
  srs_ul_cnfg.add_field(
    new parser::field<bool>
    ("max_up_pts_present", &srs_setup.srs_max_up_pts_present)
  );

  // PUCCH configuration
  parser::section pucch_cnfg("pucch_cnfg");
```
After `parse_sib2`, run ` sib2->rr_cfg_common.srs_ul_cfg_common.set(srs_ul_cfg_common_c::types::setup); // release->setup`

I also add `srs_ul_cnfg` section in `sib.conf` and I've checked the parameters are read correctly.

## Expected Behavior ##
I need `srs_tx_enabled(&cfg->ul_cfg.srs, tti)` in function `add_srs` returns true so that the srsue send the SRS and I can extract that on eNB, and obviously UE must be correctly attached to eNB. 

## Actual Behaviour ##
I added some flags to see what happens.
Output of srsenb
```
---  Software Radio Systems LTE eNodeB [Customized] ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor
WARNING: Could not verify cpu4 scaling governor
WARNING: Could not verify cpu5 scaling governor
WARNING: Could not verify cpu6 scaling governor
WARNING: Could not verify cpu7 scaling governor

Built in Release mode using commit 474c43b73 on branch srs_ul_enb.

[M: parse_sib2] parse SRS_UL_CNFG
Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD file
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz...
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
[M: set_phy_cfg_t_common_srs] common enabled srs!
RACH:  tti=3061, cc=0, pci=1, preamble=44, offset=1, temp_crnti=0x46
[M: send_connection_setup] establish configuration srs
[M: set_phy_cfg_t_dedicated_cfg] dedicated enable srs!
[M: set_phy_cfg_t_dedicated_cfg] srs.configured = 1!
[M: set_phy_cfg_t_dedicated_cfg] srs.srs_cfg_idx = 167!
[M: set_phy_cfg_t_common_srs] common enabled srs!
RACH:  tti=7701, cc=0, pci=1, preamble=30, offset=1, temp_crnti=0x47
[M: send_connection_setup] establish configuration srs
[M: set_phy_cfg_t_dedicated_cfg] dedicated enable srs!
[M: set_phy_cfg_t_dedicated_cfg] srs.configured = 1!
[M: set_phy_cfg_t_dedicated_cfg] srs.srs_cfg_idx = 167!
Disconnecting rnti=0x46.
[M: set_phy_cfg_t_common_srs] common enabled srs!
RACH:  tti=951, cc=0, pci=1, preamble=6, offset=1, temp_crnti=0x48
[M: send_connection_setup] establish configuration srs
[M: set_phy_cfg_t_dedicated_cfg] dedicated enable srs!
[M: set_phy_cfg_t_dedicated_cfg] srs.configured = 1!
[M: set_phy_cfg_t_dedicated_cfg] srs.srs_cfg_idx = 167!
Disconnecting rnti=0x47.
SCTP Association Shutdown. Association: 0
Restarting S1 connection
connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
```
Output of srsue
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins:
Couldn't open , trying /root/.config/srsran/ue.conf
Couldn't open /root/.config/srsran/ue.conf either, trying /etc/srsran/ue.conf
Reading configuration file /etc/srsran/ue.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor

Built in Release mode using commit f1ade9811 on branch master.

srsLog error - Unable to create log file "/tmp/ue.log": Permission denied
Opening 1 channels in RF device=default with args=default
Supported RF device list: UHD zmq file
Trying to open RF device 'UHD'
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz...
[INFO] [B200] Actually got clock rate 23.040000 MHz.
RF device 'UHD' successfully opened
Add customer modification
Waiting PHY to initialize ... done!
Attaching UE...
.
Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.1 KHz
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=49, tti=7861, ra-rnti=0x2
[M: pusch_encode] called!
[M: add_srs] called!
[M: pucch_encode] called!
[M: add_srs] called!
RRC Connected
Random Access Complete.     c-rnti=0x46, ta=1
[M: pucch_encode] called!
[M: add_srs] called!
[M: pusch_encode] called!
[M: add_srs] called!
....... (similar outputs)
[M: pucch_encode] called!
[M: add_srs] called!
[M: pucch_encode] called!
[M: add_srs] called!
Warning: Detected Radio-Link Failure
[M: pucch_encode] called!
[M: add_srs] called!
[M: pusch_encode] called!
[M: add_srs] called!
RRC IDLE
```
It looks like the RRC message is customized incorrectly, so I am wondering if there is any required parameter modification is left.

Luke20000429, 2022-07-24:
[pcaps.zip](https://github.com/srsran/srsRAN/files/9176913/pcaps.zip)
Here are the packet capture logs of enb and ue, I enabled WRITE SIB.

debashish216, 2022-08-17:
Hi @Luke20000429 , I am curious to ask you whether you found any workaround to fix this issue? Are you able to decode the sounding reference signal at srseNB? Thanks.

Luke20000429, 2022-08-17:
Hi @debashish216, I've already resolved the attachment failure and decode the SRS by modifying configurations. The SRS is not transmitted even when `srs_cfg.configured = true` is because `srs_cfg_idx` is not coupled with subframe configuration. However, I still cannot understand the actual cause of the attachment failure. Although it didn't appear after updating my configuration, my UE cannot reconnect BS if the BS reboots.

debashish216, 2023-01-08:
Dear Luke, Thank you very much for your previous answer. 
Do you think above configurations will work for a COTS UE? 
Have you tried to configure Sounding RS so that the COTS UE sends it to the eNodeB ? If yes, could you kindly guide me on what type of changes are required for the COTS UE, please.  Thank you.

debashish216, 2023-01-21:
Hi Luke, I think the "RRCConnectionReconfiguration" message from eNB to UE does NOT reconfigure the sounding reference signal. Even if it is present in "RRCConnectionSetup" from eNB to UE, the same is absent in RRCConnectionReconfiguration IE. Do you think it could be an issue due to which the UE does not send sounding RS. Thanks. Please refer the attached image from pcap.
[image]