srsran/srsRAN_4G #939: turn ON USE_LTE_RATES options UE fail to discover srsran eNodeB

samssm, 2022-07-25:
## Issue Description ##
[Describe the issue in detail]

turn ON USE_LTE_RATES options UE(iphone6 plus & iphone13 pro & Mi 10) fail to discover srsran eNodeB(although measure look fine in Keysight N9020A MXA Signal Analyzer), but UE(iphone6 plus & iphone13 pro & Mi 10) work fine when turn OFF USE_LTE_RATES in CMakefilelist。Anynone give a clue?

PS: Use ettus B210, default 23.04Msps have change to 30.72Msps in lib/src/phy/rf/rf_uhd_impl.c. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

samssm, 2022-07-25:
PS: all test case is 5MHz SISO

samssm, 2022-07-25:
adj tx_adv_auto nsamples, this issue can be fixed, close this issue.

andrepuschmann, 2022-07-25:
Excellent. Thanks for the feedback

closed 2022-07-25 by andrepuschmann